esp8266/Arduino #4010: Question: getting update progress

mcmega, 2017-12-21:
Hi guys!
How can I get the firmware upgrade progress?

ESP8285
1M/128K
80MHz

```cpp
void updateESP() {
  flagUpdateESP = false;
  sendUART("SYS", "updateESP", "Start");            // Сообщаем о начале прошивки ESP
  if (_updateFW(true)) {                            // Обновляем FS
    saveConfig();                                   // Сохраняем текущие настройки в новую FS
    if (_updateFW(false)) {                         // Обновляем прошивку модуля
      sendUART("SYS", "updateESP", "End");          // Прошивка ESP завершена
    }
  }
}

bool _updateFW(bool updTarget) {
  HTTPClient httpClient;
  httpClient.useHTTP10(true);
  httpClient.setTimeout(5000);
  char url[65];
  sprintf(url, "http://%s/%s-%d-%s.bin", OTA_HOST, DEVICE_MODEL, latVer, updTarget ? "ffs" : "esp");
  httpClient.begin(url);
  if (httpClient.GET() != HTTP_CODE_OK) {
    sendUART("ER!", "updateESP", "Unable to Fetch");
    return false;
  }
  if (!Update.setMD5(updTarget ? ffsMD5 : espMD5)) {
    sendUART("ER!", "updateESP", "MD5");
    Update.printError(Serial);
    return false;
  }
  if (!Update.begin(httpClient.getSize(), updTarget ? U_SPIFFS : U_FLASH)) {
    sendUART("ER!", "updateESP", "Begin");
    Update.printError(Serial);
    return false;
  }
  if (!Update.writeStream(*httpClient.getStreamPtr())) {
    sendUART("ER!", "updateESP", "Write Stream");
    Update.printError(Serial);
    return false;
  }
  if (!Update.end()) {
    sendUART("ER!", "updateESP", "End");
    Update.printError(Serial);
    return false;
  }
  httpClient.end();
  return true;
}
```
The update is going well, but I want to know how much is completed.
Help.

devyte, 2017-12-21:
@mcmega I believe the updated class has callbacks, and one is for progress. I haven't used them myself, but "use the source, Luke!".
Closing per issue policy doc.

closed 2017-12-21 by devyte

mcmega, 2017-12-23:
It works, maybe someone will need it!
Updating the file system and sketch from the remote server with checking md5 and returning the progress in percent.
void updateESP() {
  sendWS(NULL, "infoMessage", "FFS Update...");
  if (_updateFW(true)) {
    delay(250);
    sendWS(NULL, "infoMessage", "ESP Update...");
    if (_updateFW(false)) {
      sendWS(NULL, "infoMessage", "All Updates Completed");
      delay(250);
    }
  }
}

bool _updateFW(bool updTarget) {
  udp.close();
  HTTPClient httpClient;
  httpClient.useHTTP10(true);
  httpClient.setTimeout(5000);
  char url[65];
  sprintf(url, "http://%s/%s-%d-%s.bin", OTA_HOST, DEVICE_MODEL, latVer, updTarget ? "ffs" : "esp");
  httpClient.begin(url);
  if (httpClient.GET() != HTTP_CODE_OK) {
    sendWS(NULL, "infoMessage", "Unable to fetch, please reboot the device and try again");
    return false;
  }
  int httpSize = httpClient.getSize();
  if (!Update.setMD5(updTarget ? ffsMD5 : espMD5)) {
    sendWS(NULL, "infoMessage", "Invalid MD5, please reboot the device and try again");
    return false;
  }
  if (!Update.begin(httpSize, updTarget ? U_SPIFFS : U_FLASH)) {
    sendWS(NULL, "infoMessage", "Incorrect startup conditions, please reboot the device and try again");
    return false;
  }
  uint8_t buff[1024] = { 0 };
  size_t sizePack;
  WiFiClient * stream = httpClient.getStreamPtr();
  while (httpClient.connected() && (httpSize > 0 || httpSize == -1)) {
    sizePack = stream->available();
    if (sizePack) {
      int c = stream->readBytes(buff, ((sizePack > sizeof(buff)) ? sizeof(buff) : sizePack));
      Update.write(buff, c);
      if (httpSize > 0)
        httpSize -= c;
    }
    if (progress != int(Update.progress() * 100 / httpClient.getSize())) {
      progress = int(Update.progress() * 100 / httpClient.getSize());
      updTarget ? sendWS(NULL, "updPrgFFS", String(progress)) : sendWS(NULL, "updPrgESP", String(progress));
    }
  }
  if (!Update.end()) {
    sendWS(NULL, "infoMessage", "Invalid completion conditions, please reboot the device and try again");
    return false;
  }
  httpClient.end();
  return true;
}

Disane87, 2018-01-05:
Its would be really great to have a callback function for the progress, instead of hacky workarounds :/
In my project, I really need to show the progress, but I can't get it working:

How can I get the progress working with my code?
```cpp
void checkForUpdates() {
  display_text("Checking for updates");
  // display.drawProgressBar(4, 40, 120, 8, 10);
  // display.display();

  // String fwVersionURL = String(fwUrlBase)+"/"+String(DeviceName)+".version";

  Debug.println("[checkForUpdates]: Checking for firmware updates.");
  Debug.println("[checkForUpdates]: Firmware version URL: "+String(fwUrlBase));

  t_httpUpdate_return ret = ESPhttpUpdate.update(fwUrlBase, String(FW_VERSION));
  switch(ret) {
      case HTTP_UPDATE_FAILED:
          Debug.println("[checkForUpdates]: Update failed: "+String(ESPhttpUpdate.getLastErrorString()));
          break;
      case HTTP_UPDATE_NO_UPDATES:
          Debug.println("[checkForUpdates]: No Update needed");
          break;
      case HTTP_UPDATE_OK:
          Debug.println("[checkForUpdates]: Update ok."); // may not called we reboot the ESP
          break;
  }

```

pidloop, 2018-02-25:
Yes, I too would like to get progress when using ESPhttpUpdate.update(). Thanks!

1TekPro, 2019-02-06:
I managed to get a progress bar working on httpupdate but not on this library. Please help us here.

d-a-v, 2019-02-07:
duplicate with opened #5715

franzuu, 2020-02-23:
Hi! I see this progress callback function is merged, but how to use it? Are there any examples?