esp8266/Arduino #4012: W5500 Board with Wemos D1 mini pro, Ethernet Webserver example gets no IP address

weigu1, 2017-12-21:
#### Hardware
Hardware:		Wemos D1 mini pro, W5500 board
Core Version:      	2.3, 2.4-rc1,2.4-rc2

### Description
The board  (http://www.haoyuelectronics.com/Attachment/W5500-Ethernet/W5500-Ethernet-2.jpg) works without issue with different teensys and arduinos.

When I try the Ethernet (8266) Webserver example with Wemos D1 mini pro, I get always "server is at 255.255.255.255". 
Wemos D1 Mini Pro Pins: D6 MISO 12,  D7 MOSI 13,  D5 SCK 14, D8 SS 15.
Tried with different speeds (in W5100.h) and other SS Pins (D1 (5) and D3 (0)) with no better result ("server is at 0.111.111.111"). 

BTW to chose another SS Pin I had to add:
#define SS 5
to W5100.h
in W5100.cpp we have:
 #if defined(ARDUINO_ARCH_AVR) || defined(ESP8266)
  SPI.begin();
  initSS();
so initSS() is called in W5100.h wich needs SS:
inline static void initSS()    { pinMode(SS, OUTPUT); };

or an I missing something?

Thanks for Help

Guy

devyte, 2017-12-21:
@weigu1 support for w5500 is currently not implemented in the core. There are currently no developers available for it, so feel free to open a PR with an implementation proposal (you could base it off other repos or off the current w5100 implementation). If you aren't willing to implement or port yourself, please close this issue.

weigu1, 2017-12-22:
@devyte 
Thanks for the reply.

weigu1, 2017-12-30:
Hi 
I bought W5100 boards. Again these boards work flawless with Teensy/Arduino
The Ethernet example is not working with Wemos D1 mini pro
I get no IP address.
CLK => D5
MISO => D6
MOSI => D7
SPI Bus SS (CS)is D8
Thanks for Help

devyte, 2017-12-31:
Please open a new issue for the 5100. Be aware that:
-no  developer of those currently active has that hardware, at least per my understanding, so the issue is unlikely to get fixed soon
-the wemos D1 mini pro was added recently, due to very recent support for 16MB flash in the SDK. It is experimental at best, and is known to have some odd issues. investigation hasn't started yet, so I suggest testing the 5100 with a wemos D1 mini r2 first (4MB) or similar in the meantime.

closed 2017-12-31 by devyte

weigu1, 2017-12-31:
Thank you for the clarifications :)