esp8266/Arduino #4013: form URL encoding fails

s0170071, 2017-12-21:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini
Core Version:      	2.3.0 / 2.4.0rc2

### Description

the URL encoding for post forms is broken for post forms with the default enctype

Behaviour with core version 2.3.0   method='post' .. OK
Behaviour with core version 2.3.0  method='post'enctype='multipart/form-data'' .. OK
Behaviour with core version 2.4.0rc2   method='post' .. OK
Behaviour with core version 2.4.0rc2   method='post'enctype='multipart/form-data'' .. FAIL

Fail means that you have to URLencode the string in the form yourself in order for it to get received as intended  by the ESP. This means that it probably gets URLdecoded although it was not ENcoded in the first place.

Using the test sketch:
1. fill in ssid and password.
2. run and open the serial monitor. Open the IP displayed here in the browser.
3. open a.htm for method='post' form, b.htm for a method='post'enctype='multipart/form-data'' form
4. write something with a "%" in the field and hit enter. Watch the serial window. 

 

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	 512k
Upload Using:		 SERIAL
Reset Method:		power cycle

### Sketch
```

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

const char* webpage ="<!DOCTYPE html><html lang='en'><head></head><body><form name='frmselect' method='post'><table><TR><TD colspan='2'><h2>Form Test</h2><TR><TD>Line 1:<TD><input type='text' name='abc' maxlength=64 value='%ip%'><TR><TD><TD><input type='submit' value='Submit'><input type='hidden' name='edit' value='1'><input type='hidden' name='page' value='1'></table></form></body></html>";
const char* webpage2 ="<!DOCTYPE html><html lang='en'><head></head><body><form name='frmselect' method='post'enctype='multipart/form-data'><table><TR><TD colspan='2'><h2>Form Test</h2><TR><TD>Line 1:<TD><input type='text' name='abc' maxlength=64 value='%ip%'><TR><TD><TD><input type='submit' value='Submit'><input type='hidden' name='edit' value='1'><input type='hidden' name='page' value='1'></table></form></body></html>";
const char* ssid = "...";
const char* password = "...";

ESP8266WebServer server(80);

void handlea() {
  server.send(200, "text/html", webpage);
}
void handleb() {
  server.send(200, "text/html", webpage2);
}
 

void setup(void){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
 

  server.on("/a.htm", handlea);
  server.on("/b.htm", handleb);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
  String message = "";
  Serial.println(">" + server.arg("abc") +"<");
  delay(1000);
}
```

kapyaar, 2017-12-26:
I had a similar issue, and posted this. 

https://github.com/esp8266/Arduino/issues/3791

Based on the reply by devyte, I used the Parsing.cpp PR code and that resolved the issue. Can you check that out?

devyte, 2017-12-28:
@s0170071 does PR #3313 fix this?

s0170071, 2017-12-28:
Give me some days. Maybe I should also try the head revision. just in case the issue is already fixed otherwise. 
But just to clarify: it's a bug, not a feature, right?

Am 28. Dezember 2017 04:27:33 MEZ schrieb Develo <notifications@github.com>:
>@s0170071 does PR #3313 fix this?

devyte, 2017-12-28:
@s0170071 I understand it's a bug in how the decoding is done.
Please test with latest git plus the referenced PR.

s0170071, 2017-12-30:
Just confirmed. PR #3313 fixes this, current latest git (head revision) doesn't.

closed 2017-12-30 by s0170071