esp8266/Arduino #4014: document release spinning process, so it may be delegated

herrold, 2017-12-21:
This is a process bug, and not a 'sources' bug.  As such the usual questionaire does not apply

reference 
    https://github.com/esp8266/Arduino/issues/3991

We face load from people (reasonably) pulling a release tarball, and attempting to install it.  This load should go away, by not leaving known stale content as the seeming current content.

my local personal process is to build a daily includes:

...
YND-`date +%Y_m_d`
cd vcs/git/esp8266/Arduino
git pull --no-ff
cd ..
mkdir Arduino-0.${YMD}
rsync -a ${VERBOSE} Arduino/.  Arduino-0.${YMD}/.
#   get rid of VCS cruft
rm -rf Arduino-0.${YMD}/.git*
#   new since I looked last ... CI as well
rm -rf Arduino-0.${YMD}/.travis.yml
tar zcf Arduino-0.${YMD}.tgz ./Arduino-0.${YMD}/
...

and then handing that daily tarball over to my later build processes.  I intentionally 'prefix' a daily with the leading "0." so I can spot them more easily.  Some processes, pre release 1.x, use 
    0.0. blah
 but this one is past that

igrr, 2017-12-26:
Releasing a new version works more or less like this:

1. Contributor pushes a new tag, e.g. '2.4.0', to Github
2. CI builds the boards manager package, creates a Release on Github, uploads the created package (.zip + .json files) into the release.

I disagree with your point about "not leaving known stale content as the seeming current content" though. Although release 2.3.0 is very old, using it has one benefit over using git version: much simpler setup process. Especially on Windows, which doesn't come with git by default. For that reason, i think the existing boards manager package should be kept as a setup option, even if it is outdated.

earlephilhower, 2019-07-26:
We've got this taken care of now.  It was so easy that even I could do it. :)

closed 2019-07-26 by earlephilhower