esp8266/Arduino #4017: Getting Soft WFT Resets all the time

krmax44, 2017-12-23:
### Basic Infos

I keep getting Soft WFT Resets while trying to send RF commands...

#### Hardware
Hardware:			ESP 8266 NodeMCU v3 dev kit (LoLin)

### Description

Problem description

### Settings in IDE

Module:  			Nodemcu v3
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		onboard SERIAL

### Sketch

```cpp
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);

  Serial.println("starting");
 
  mySwitch.enableTransmit(0);
  mySwitch.setPulseLength(102);
  mySwitch.setProtocol(3);
  mySwitch.setRepeatTransmit(150);
  Serial.println("started");
}

void loop() {
  Serial.println("bf sending");
  mySwitch.send(4967468, 24);
  Serial.println("sent");
  delay(5000);
}
```

### Debug Messages

Output of the serial monitor:

```
⸮⸮ѥ⸮⸮5
booted
bf sending

Soft WDT reset

ctx: cont 
sp: 3ffef330 end: 3ffef580 offset: 01b0

>>>stack>>>
3ffef4e0:  0000001c 00000000 3ffee528 40100574  
3ffef4f0:  40106cb0 3ffee318 00000000 4020244c  
3ffef500:  00000000 00000009 3ffe8451 40201d84  
3ffef510:  3fff0609 0000000a 3ffee528 40202075  
3ffef520:  3ffe84d8 00000045 ffffffff 40201e3e  
3ffef530:  004bcc2c 3ffee32a 00000017 40202090  
3ffef540:  feefeffe feefeffe 3ffee528 3ffee54c  
3ffef550:  3fffdad0 00000000 3ffee528 40201c7a  
3ffef560:  feefeffe 00000000 3ffee544 40202244  
3ffef570:  feefeffe feefeffe 3ffee560 40100114  
<<<stack<<<
⸮⸮⸮⸮H!⸮ȩ
```

coelner, 2017-12-23:
https://github.com/me-no-dev/EspExceptionDecoder

5chufti, 2017-12-24:
since you do not write which version of rcswitch you are using and the wdt-reset obviously occurs while being in this library, one can only assume that this version of rcswitch is not esp compatible. That is no problem with the arduino esp core but with the lib you chose.

earlephilhower, 2019-01-16:
Closing the problem, if any, is in a 3rd party library which probably isn't ever calling yield().

closed 2019-01-16 by earlephilhower