esp8266/Arduino #4018: flush() is a TX related function

jandrassy, 2017-12-24:
WiFiClient.flush should send all data from tx buffer. it is not true that in Arduino world it is a rx buffer thing. the Client class is inherited from Stream class and the [flush method is from Stream](https://www.arduino.cc/reference/en/language/functions/communication/stream/streamflush/). and in Stream implementations the flush is a tx thing. And the reference for the Client class implementation is [Ethernet library](https://www.arduino.cc/en/Reference/ClientFlush).

d-a-v, 2017-12-25:
Duplicate of issue #1002.
This has been very recently fixed by #3967 and is currently in git master (not 2.4.0-rc2).

devyte, 2017-12-25:
Closing as duplicate of #1002 .

closed 2017-12-25 by devyte

d-a-v, 2017-12-25:
@jandrassy  (from #3967)
> and moving flush() from Stream to Print?

The ::flush() beeing a tx not rx stuff was #1002's point and fix is merged.
::flush() in arduino is declared in Print:: which is inherited by Stream::, and our flush() is declared only in Stream::. 

So you were suggesting for moving it to Print:: ?
Can you please elaborate ?

jandrassy, 2017-12-26:
flush() accompanies write functions in Print. so yes it should be in Print. Stream adds read functions. flush() is important for effective implementations of write() with buffering. most Arduino user programmers do not use flush(). Therefore in my implementations of Print (mostly derived from Stream or Client) I add flush into available() if rx buffer is empty and tx buffer is not sent. It is much better the [sending by one byte over some protocol with frame bytes overhead.](https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/Story)

d-a-v, 2017-12-26:
Since current implementation does not exactly follows arduino's core library, this is a fair proposal.
Have you considered making yourself a pull request for this ?

jandrassy, 2017-12-26:
no, because it would take to much time for me to fork the project and set it up for testing

d-a-v, 2017-12-26:
I hope you'll have time to try #4029

PR #4029 (2017-12-29): follow arduino API: pure virtual Stream::flush() moved to empty virtual Print::flush()
re fix #4018
::flush() declaration are now exactly the same as arduino's
[flush added](https://github.com/arduino/Arduino/blob/cd9a15ebd1647336200b9328bc0912495e3a7195/hardware/arduino/avr/cores/arduino/Print.h#L90)
[flush removed](https://github.com/arduino/Arduino/blob/cd9a15ebd1647336200b9328bc0912495e3a7195/hardware/arduino/avr/cores/arduino/Stream.h#L62)
commit 0b2df351 (2017-12-29): follow arduino API: pure virtual Stream::flush() moved to empty virtual Print::flush() (#4029)

re fix #4018