esp8266/Arduino #4022: DHT temperature reading unstable with 2.4.0.RC2

robcazzaro, 2017-12-25:
I have been running the same code on a series of ESP8266 for some time using 2.3.0: read temp and humidity from a DHT22, upload to a Thingsboard server using mqtt libraries. My code uses SPIFFS to store data, and a web server to modify parameters while running, plus OTA updates

I am using the Adafruit DHT libraries with a single fix, as per https://github.com/adafruit/DHT-sensor-library/issues/94. That code has been stable and I have tens of thousands of data points with stable readings

I tried compiling the same code with 2.4.0-rc2, no other change, and on a generic ESP12 module (1M, with 128k SPIFFS), and the temperature readings are "noisy", up to 4 degrees C variability when usually on a stable environment the noise is 0.1C

I'm enclosing a screenshot that shows the temperature readings with 2.4.0-rc2 (spiky), and the exact same devices after OTA_updating back to 2.3.0. The blue device is a Wemos D1, and doesn't seem to be impacted by 2.4.0-rc2, unlike the 3 generic ESP12 (all from different eBay sellers, bought over time)

#### Hardware
Hardware:			Generic ESP-12, 1M, 128k SPIFFS
Core Version:      	2.4.0-rc2
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		OTA & SERIAL, no difference
Reset Method:		ck 

### Sketch (relevant part) only

#include "DHT.h"
//>#include <PubSubClient.h>
#include <MQTTClient.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>       
#include <ESP8266WiFiMulti.h>
#include <ArduinoJson.h>
#include "FS.h"

// DHT
#define DHTPIN 12
#define DHTTYPE DHT22

DHT dht(DHTPIN, DHTTYPE);

void getAndSendTemperatureAndHumidityData()
{
  digitalWrite(LED_PIN, HIGH);  // flash LED if cannot read DHT sensor
  delay(50);
  digitalWrite(LED_PIN, LOW);  // turn on LED
  Serial.print("Collecting temperature data. ");

  // Reading temperature or humidity takes about 250 milliseconds!
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();

  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t)) 
  {
    Serial.println("Failed to read from DHT sensor!");
    lastSentTime = lastSentTime + 1000;  // hack: retry in 1 sec
    return;
  }

  h = h + hOffset;
  t = t + tOffset;
 
  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");

  String temperature = String(t);
  String humidity = String(h);

  Serial.print("Sending temperature and humidity : [");
  Serial.print(temperature); 
  Serial.print(",");
  Serial.print(humidity);
  Serial.print("]   -> ");

  // Prepare a JSON payload string
  String payload = "{";
  payload += "\"temperature\":"; payload += temperature; payload += ",";
  payload += "\"humidity\":"; payload += humidity;
  payload += "}";

  // Send payload
  char attributes[100];
  payload.toCharArray(attributes, 100);

  Serial.print("Sending telemetry(");
  Serial.print(mqttClient.publish("v1/devices/me/telemetry", attributes));
  Serial.print("): ");
  Serial.println(attributes);
  digitalWrite(LED_PIN, HIGH);
  lastSentTime = millis();
}

void setup()
{
  Serial.begin(9600);
  pinMode(LED_PIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  dht.begin();
  delay(10);
}

[image]

5chufti, 2017-12-25:
Did you consider EMI? 2.3.0 / 2.4.0-rc have different wifi behaviour.
Try putting wifi to at least modemsleep during measurment.
The flat blue line of WemosD1 may be a prove that different placement reduces EMI susceptibility.

robcazzaro, 2017-12-25:
No, I haven't tried yet, the thought didn't cross my mind. Just to make sure I understand, are you suggesting to do something like:

WiFi.forceSleepBegin();
delay(10);
//read DHT22
WiFi.forceSleepWake()
delay(10);

Please confirm, and I will try and see if that's the issue in this case. Disconnecting WiFi would not be an acceptable solution for me, since it would terminate WiFi and mqtt connections every 3-5 seconds (i.e. for me 2.3.0 would be better). But happy to help narrow this down and see if it's a 2.4 issue or simply a poorly designed ESP12 module interacting with the DHT22

5chufti, 2017-12-25:
yes, either this 
```
  WiFi.setSleepMode(WIFI_MODEM_SLEEP);
  WiFi.forceSleepBegin();
  delay(10);
```
or making sure that the DHT22 and its supply is sufficiently shielded / decoupled.

robcazzaro, 2017-12-26:
Thanks... when I execute those 3 lines, my device crashes right after calling WiFi.forceSleepBegin() with the below. What am I doing wrong? 

I also tried recompiling the test code with 2.3.0 instead of 2.4.0-RC2, and I do not get the crash at all. Code works as expected, ESP12 sleeps, wakes up and reconnects. I really seem to have a lot of issues with RC2 on all my ESP12 (I downloaded from the official distribution point using https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json in the preferences dialog)

//read DHT22
  WiFi.forceSleepWake();
  delay(100);
  WiFi.mode(WIFI_STA); 
  delay(100); 
  wifiReconnect(); // (usual loop waiting for connection   
  delay(100);

Soft WDT reset

ctx: cont 
sp: 3fff2300 end: 3fff25c0 offset: 01b0

>>>stack>>>
3fff24b0:  0001360d 00000600 3fff11a8 40237ab0  
3fff24c0:  0fffffff 4020dabe 00000000 4020daa9  
3fff24d0:  3fff1134 3fff11a8 3fff1228 4020b727  
3fff24e0:  3ffe9c14 3fff04ac 3fff14d0 402168c9  
3fff24f0:  3ffe91f0 00000000 00000000 00000000  
3fff2500:  3ffea1a0 1401a8c0 3fff14d0 40216af0  
3fff2510:  3fff1168 3fff04ac 3fff14d0 4020a506  
3fff2520:  3ffe914c 3fff14d0 3fff14d0 40216a7c  
3fff2530:  00000000 00000000 3fff0564 4020f606  
3fff2540:  3fff1168 3fff14d0 3fff11a8 00000003  
3fff2550:  40104c70 00c1eb8c 3fff12e8 00000000  
3fff2560:  3ffefea0 3fff12e8 3fff15a0 3fff12e8  
3fff2570:  3fffdad0 3fff1590 402180fc 3fff15a0  
3fff2580:  40201f32 000003e8 000003e8 3fff1590  
3fff2590:  3fffdad0 3fff04ac 3fff1228 4020b9b0  
3fff25a0:  3fffdad0 00000000 3fff1588 40218148  
3fff25b0:  feefeffe feefeffe 3fff15a0 4010070c  
<<<stack<<<

5chufti, 2017-12-26:
hmmm, the test code you posted doesn't show any wifi handling.
try to really reduce the code to a bare minimum example (w/o all the ota, webserver, etc includes)  that somebody can use to reproduce your problem. It looks like some "not relevant" part of the code tries to access wifi (waits for) when it is turned off.

robcazzaro, 2017-12-26:
I have tried changing the power supply for the impacted modules, and while there still are differences between 2.3.0 and 2.4.0-rc2 those are much less pronounced. So it looks like you are right, and EMI plays a role. As such, I don't think it's worth keeping this issue open, because would be extremely hard to repro my situation exactly (and I can work around it)

On the other hand, the exact same code (sleep) works on 2.3 and crashes with a stack dump on 2.4. Yes, my code is pretty complex (that's why I only posted a snippet), but the section where I was turning off wifi is very simple and no other wifi activity takes place (in my code, that is):

Serial.println("Ready to sleep");
delay(100);
  WiFi.setSleepMode(WIFI_MODEM_SLEEP);
  delay(100);
  Serial.println("Ready to force");
  WiFi.forceSleepBegin();
  delay(100);
  
  Serial.print("Collecting temperature data. ");

I never see the "Collecting temperature data" message, the crash happens right after I see "ready to force" on the serial monitor.

I'll close this one and see if I can repro the crash (if so, I'll open a new issue). It's possible that OTA does something in the delay loop (I know that the webserver and mqtt do not, based on tests I did with long delay()), do you know if OTA handling changed between 2.3 and 2.4?

devyte, 2017-12-26:
@robcazzaro a whole lot has changed since 2.3.0, including a new SDK version with changes to how wifi is calibrated on startup, and a lot has changed since 2.4.0-rc2, including a new lwip. Please test with latest git built with lwip2 (option in Arduino IDE menu) rather than rc2.
Closing this due to power supply.

closed 2017-12-26 by devyte