esp8266/Arduino #4026: RFC - automatically load CA Certs from SPIFFS or PROGMEM for WiFiClientSecure/ESP8266HTTPClient

liebman, 2017-12-25:
After the discussion on https://github.com/esp8266/Arduino/pull/3933 I started looking at making it easier to handle CA Certificates.  
My thinking is to add a new function to the axtls library that would return the name of the certificate that needs to be loaded using ca_cert_dn from the last cert in the chain. Then use that to load data either from SPIFFS or PROGMEM.  
Has anyone looked at this or something similar?  Thoughts?

igrr, 2017-12-26:
I'm working on https://github.com/igrr/axtls-8266/issues/44, which has similar goals, but uses authority key identifier instead of the CA certificate name.

liebman, 2018-08-06:
With the integration of bearssl I think this can be closed.

closed 2018-08-06 by liebman