esp8266/Arduino #4030: ESP8266 VERY HIGH current consumption about 160mA even when in deep sleep!

eldonjakee08, 2017-12-27:
Hi guys! I've recently bought a nodemcu board from banggood and I've noticed that when in deep sleep it still has a very high current consumption about 130mA and it sets the chips quite warm too. Also I've noticed that whenever I power the nodeMCU with just the battery the esp8266 runs the sketch fine. But whenever I connect a multimeter in series with the battery to Vin of the nodeMCU (to measure the current consumption) it won't run the sketch and reads a very high current consumption even when in deep sleep, current consumption is at 130mA. I thought I had a broken multimeter at first so I just bought a new multimeter and did the same test but still the same results the sketch just wont run when a multimeter is in series with the battery. I also tried disconnecting the regulator I disconnected its ground pin but still the same results. Any ideas as to what might be causing this? .... Thanks in advance!! :D (Photos shown underneath)

[image]
**Figure 1.1 The nodeMCU board**

and here is the test sketch that I uploaded into the board.

`extern "C" {
#include "user_interface.h" // this is for the RTC memory read/write functions
}

// defines pins numbers for ultrasonic
const int trigPin = 5; //D1
const int echoPin = 4; //D2

int sensor_read;
long duration;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
  pinMode(echoPin, INPUT); // Sets the echoPin as an Input
  system_rtc_mem_read(64, &sensor_read, 4); //reads previous sensor reading stored in RTC memory
   Serial.println("");
  Serial.print("Past Sensor Reading: ");
  Serial.println(sensor_read);
  
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  sensor_read= duration*0.034/2;
  Serial.print("Actual Read: ");
  Serial.println(sensor_read);
  system_rtc_mem_write(64, &sensor_read, 4); //writes present sensor reading stored in RTC memory
  delay(10);
  system_rtc_mem_read(64, &sensor_read, 4); //reads present sensor reading stored in RTC memory
  Serial.print("Present Sensor Reading: ");
  Serial.println(sensor_read);

  Serial.println("Going Back to Sleep!");
  ESP.deepSleep(5e6); //sleeps for 5 seconds
  delay(100);
  //just a test to see if it did sleep properly, if it did not sleep it would print this out to serial.
  Serial.print("olkpko");

}

void loop() {
  // put your main code here, to run repeatedly:

}`

[image]
**Figure 1.2 Here I'm measuring the current consumption of the board currently consuming 130mA**

[image]
**Figure 1.3 measuring current shown at a different angle**

[image]
**Figure1.4 The sketch would run just good if I just connect the board directly to the battery, without the multimeter in between them.**

[image]
**Figure 1.5 Here I disconnected the GND pin of the regulator to see if there would be any changes to the current consumption but still shows the same high curretnt.**

liebman, 2017-12-27:
I see other items connected in the circuit as well...  Have you tried with them removed?

This is probably not the right place to ask for help with your particular project. This is an issue tracker. Please refer to a community forum for help.

devyte, 2017-12-27:
@eldonjakee08 sounds like an electrical issue, and not a code issue. I suggest:
-Erasing the entire flash with esptool
-doing a clean install to make sure you are building with the correct core
-eliminate the protoboard and test deepsleep with just the nodemcu without any other hardware

This is not the right place to ask what's wrong with your particular project. Please refer to a community forum for help.
Closing per issue policy doc

closed 2017-12-27 by devyte

eldonjakee08, 2017-12-27:
oh I'm so sorry!! and thanks so much for the suggestions! :D

torntrousers, 2017-12-27:
@eldonjakee08 I've found that it often doesn't work using cheap multimeters with just a 200ma range connected in series with the ESP during power on. I think its because the ESP has quite a high burst of current at power on an with the 200ma range the multimeter doesn't let enough current be drawn and so the ESP crashes, and the 130ma current you see is what the ESP draws in its crashed state. One way I've got around this is to short the multimeter probes together for power on and then once the ESP is up and running you can break the short and the ESP keeps running and the multimeter can measure the current.

eldonjakee08, 2017-12-27:
right right!!!!! thank you so much for the enlightenment!!! :D :D

eldonjakee08, 2017-12-27:
thank you so much! it works now!!!! :D :D