esp8266/Arduino #4032: High current during wakup from LIGHT_SLEEP_T pulse?

sheffieldnick, 2017-12-27:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.4.0-rc2

### Description

I'm triggering a wakeup from forced light sleep mode using:

`gpio_pin_wakeup_enable(14, GPIO_PIN_INTR_HILEVEL);`
`wifi_fpm_set_wakeup_cb(fpm_wakup_cb_func1);`

This works, but I'm getting a high current spike during the wakeup process?

1. ESP is running with RF disabled: 15mA
(had previously called ESP.deepSleep(usecs, WAKE_RF_DISABLED) )
Pin 14 is LOW

2. ESP enters light sleep: 0.4mA

3. Pin 14 goes HIGH
**ESP goes to 72mA for as long as pin 14 remains high** - WHY?!
Wakeup callback not called yet

4. Pin 14 goes LOW
ESP is still running at 72mA, instead of the 15mA (RF disabled) I expected
Wakeup callback is called

5. ESP returns to light sleep: 0.4mA

So there are 2 issues here I think:

- When the ESP wakes from light sleep on a GPIO level, it runs at high current while waiting for the GPIO to change level. Shouldn't this polling be done in some low power state?

- When the ESP wakes from light sleep it doesn't preserve the modem state of when it entered light sleep, always powering up the modem?

Thanks

igrr, 2017-12-27:
I suggest reporting this at https://github.com/espressif/ESP8266_NONOS_SDK/issues, since the functions you are calling are implemented inside the closed source SDK, and the issues you have described can not be dealt with within this project.

devyte, 2017-12-27:
Per previous comment, closing as off-topic for this repo.

closed 2017-12-27 by devyte