esp8266/Arduino #4037: Strange output in  Serie Monitor

Marcianikus, 2017-12-27:
#### Hardware
Hardware:	ESP-12
Core Version:     2.4.0-rc2 / 2.3 

### Description
Strange output Monitor Serie :

Problem description:
Output Monitor Serie :

IN1: 
AIN2: 
are/esp8266/2.4.0-rc2/cores/esp8266/core_esp8266_main.cpp
ireccion: 

### Settings in IDE
Module:  			Wemos D1
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

#include <Arduino.h>

int PWMA = 14;        // control de velocidad
int AIN1 = 12;        // dirección
int AIN2 = 13;        // dirección
#define PWMRANGE 255;

void mover (int velocidad, int direccion){

  int inPin1 = 0;     
  int inPin2 = 1;   

  if(direccion == 1)
    {
      inPin1 = 1;
      inPin2 = 0;
    }

  digitalWrite(AIN1, inPin1);
  digitalWrite(AIN2, inPin2);
  analogWrite(PWMA, velocidad);
  
  Serial.println("AIN1: " + inPin1  );
  Serial.println("AIN2: " + inPin2 );
  Serial.println("Velocidad: " + velocidad );
  Serial.println("Direccion: " + direccion );
}

/////////////////////////////////////////////////////////////////////////////////

void setup(){

  pinMode(14, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);

  //inicializa el puerto serie
  Serial.begin(115200);
  delay(10);
}
/////////////////////////////////////////////////////////////////////////////////
void loop(){
  mover(255, 1);          
  delay(1000);               
  mover(128, 0);       
  delay(1000);
}
/////////////////////////////////////////////////////////////////////////////////

### Debug Messages

```
messages here
```

devyte, 2017-12-28:
@Marcianikus 
This is wrong:
Serial.println("AIN1: " + inPin1 );
"AIN1:" is of type const char *, which means a pointer to the first char. Adding inPin1 to the pointer makes it advance, in this case inPin1 is 0, so no change, inPin2 is 1, so the pointer advaces 1 char, hence you see "IN1:" printed out.
Later on you are adding velocided, which is 255 or 128, which makes the pointer advance to some stupid memory address that has whatever, so you garbage.
Do this:
Serial.println(String("AIN1: ") + inPin1 );
The String object has operator+() defined so that you get what you want: concatenation.
Closing as user error.

closed 2017-12-28 by devyte

Marcianikus, 2017-12-28:
Thanks a lot!!