esp8266/Arduino #4040: more linker scripts for 4MB flash chips

everslick, 2017-12-28:
For 4MB flashes there are only two configurations supported: 1MB SPIFFS and 3MB SPIFFS.

When the binary size of 500KB is reached one is forced to switched from the bigger SPIFFS to the smaller SPIFFS configuration to still be able to use OTA updates. A few more options would allow for a finer grained binary size / spiffs size balance. at least a 2m2m config could be added.

d-a-v, 2017-12-28:
@everslick, check [this #3722 comment](https://github.com/esp8266/Arduino/pull/3722#issuecomment-351703309)

everslick, 2017-12-31:
thanks, so i wait until boards.txt.py is merged. but still, boards.txt.py does only create a hard coded list of memory layouts, doesn't it? but i see that a 4m2m config is already planned for inclusion. in the long run it would probably be best to be able to create arbitrary configs by specifying the flash size and the desired size of SPIFFS. (just thinking aloud here)

d-a-v, 2017-12-31:
Yes, it currently generates a hard coded list for boards and matching ldscripts files.
It probably is easy to calculate everything from simple inputs.
Once merged it will be opened to enhancements via pull requests.

earlephilhower, 2019-01-22:
I think this is already done, with @d-a-v's changes (i.e. 4m3m, 4m2m, 4m1m options, etc.).  If not please could you reopen and explain what you're looking for?

closed 2019-01-22 by earlephilhower