esp8266/Arduino #4042: Error compiling Wifi sketch

rioan28, 2017-12-29:
Hello all,
I have been trying for few good days now to figure out through online searches why I am getting this compile error for the line "Serial.println(WiFi.getNetworkInfo());" I am using in my sketch. I have also tried results = WiFi.getNetworkInfo(); but getting the same error.
I am using Arduino 1.8.3 with ESP8266 ver 2.3.0

Any help would be greatly appreciated.
Thank you.

The error is:
**C:\Users\Tino\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI.ino: In function 'void setup()':

Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI:83: error: no matching function for call to 'ESP8266WiFiClass::getNetworkInfo()'

   Serial.println(WiFi.getNetworkInfo());

                                      ^

C:\Users\Tino\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI.ino:83:38: note: candidate is:

In file included from C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:36:0,

                 from C:\Users\Tino\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI\Ioan_ESP8266_JustThermometer2_with_Keypad_with_WIFI.ino:2:

C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiScan.h:43:14: note: bool ESP8266WiFiScanClass::getNetworkInfo(uint8_t, String&, uint8_t&, int32_t&, uint8_t*&, int32_t&, bool&)

         bool getNetworkInfo(uint8_t networkItem, String &ssid, uint8_t &encryptionType, int32_t &RSSI, uint8_t* &BSSID, int32_t &channel, bool &isHidden);

              ^

C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiScan.h:43:14: note:   candidate expects 7 arguments, 0 provided** 

### Basic Infos

#### Hardware
Hardware:			Nodemcu 1.0 ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Nodemcu 1.0 ESP-12E
Flash Size: 		4MB
CPU Frequency:	80Mhz

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Bugzee, 2017-12-29:
Hi, 

The function expects some arguments which you have failed to provide.
You can see from the error message itself. **(candidate expects 7 arguments, 0 provided)**

> C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiScan.h:43:14: note: candidate expects 7 arguments, 0 provided

Thanks.

rioan28, 2017-12-29:
I have looked at how others are using this function and none are suggesting to pass any arguments. I am suspecting that I am not using an updated library.

Bugzee, 2017-12-29:
Version 2.3.0 is the latest stable release, so you are using the updated stable version. Even in the latest code available [here ](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp) needs the arguments.

rioan28, 2017-12-29:
I stand corrected.  Thank you very much. I got it working.

5chufti, 2017-12-29:
so please close the issue

closed 2017-12-29 by rioan28