esp8266/Arduino #4045: Eduroam - 802.1x

martinius96, 2017-12-29:
Module:  			        NodeMCU v3 Lolin
Flash Size: 		        4MB/1MB
CPU Frequency:		80Mhz
#### Hardware
Hardware:			ESP-12E
Core Version:      	        latest
Hello there, I have problem and I am looking for solution for problem connecting to Eduroam network at my University. 
Our Eduroam network is using WPA/WPA2 Enterprise with Protected EAP (PEAP) and MsCHAPv2. Is there any option how to join this network? We also have certificates in PEM. and DER. format. I have read some things and I got informations like that:

With SDK 1.5 I can only connect to 802.1x EAP networks
With SDK 2 I can connect to EAP + PEAP networks. Are there some sketches in esp8266 on github, what I can use for join to that network?! 
Momentally i got SDK 1.5 and I am using Arduino core.
Thanks a lot. Cheers, Martin

devyte, 2017-12-30:
@martinius96 Which version of this core are you using? latest stable is 2.3.0, it links against SDK 1.5.something. Latest master of this core links against SDK 2.1.0, plus some extras. I suggest trying that master (install latest git, instructions are in readthedocs).
As a reference, #2595, although the comments about editing the outer identity should no longer be valid (I understand that SDK 2.1.0 is supposed to provide an api for that).
Having said the above, this is not the right place to ask for help. This is an issue tracker. Please refer to a community forum for further assistance.
Closing per issue policy.

closed 2017-12-30 by devyte

CristovaoNunes, 2018-09-19:
Hi were you able to connect to eduroam?

martinius96, 2018-09-19:
No, because there aren't methods implemented to esp8266 core. I have tried ESP32 and it worked on first time ;) on Eduroam network. You can connect with ESP8266 to WPA/WPA2 Enterprise network but only one standard.. I think EAP-TLS only... But, most WPA/WPA2 Enterprise networks using PEAP+ MsCHAPv2 (Eduroam) or EAP-TTLS.

sarkrui, 2019-03-29:
> No, because there aren't methods implemented to esp8266 core. I have tried ESP32 and it worked on first time ;) on Eduroam network. You can connect with ESP8266 to WPA/WPA2 Enterprise network but [...]

#include <ESP8266WiFi.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

// SSID to connect to
static const char* ssid = "";
// Username for authentification
static const char* username = "";
// Password for authentication
static const char* password = "";

char buff[20];
String ip;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  delay(500);

  // WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
    wifi_set_opmode(STATION_MODE);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect();
  // WPA2 Connection ends here

  // Wait for connection AND IP address from DHCP
  Serial.println();
  Serial.println("Waiting for connection and IP Address from DHCP");
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  IPAddress myAddr = WiFi.localIP();
  sprintf(buff, "%d.%d.%d.%d", myAddr[0], myAddr[1], myAddr[2], myAddr[3]);
  ip = String(buff);
  Serial.println(ip);
 
}

void loop() {
}
**************
I am using a NodeMCU ESP-12E Module. I got connected to the eduroam at the TU/e with the above codes though, it wasn't stable at all. The connection failed after serval HTTP requests.

martinius96, 2019-03-29:
Hello, which version of ESP8266 Arduino core are you using? I want to test it at Monday :-) 
Really nice snippet of code for that purposes, hope it will work at my university too.

sarkrui, 2019-03-29:
> which version of ESP8266 Arduino core are you using?
ESP8266 Board 2.5.0, and good luck! It's not stable probably because it's still in development. When it failes to access WPA2-Enterprise, it constantly reboots.

martinius96, 2019-03-31:
Hello, there.. I have tested it and it is working! I wasn't testing on eduroam but on similar network under 802.1x with same PEAP + MsCHAPv2 auth. methods.
[image]
I have added also custom MAC address function, because connection there in LAN is based on MAC address. I wanted to know amouth of memory that board will use during authentication, but it is not so much, i am so intereted :-) I have make also new repo where I will add connection examples.. like i have for ESP32. 
Thanks a lot once more! :)

martinius96, 2019-03-31:
It looks like there is problem with malloc: https://github.com/martinius96/ESP8266-eduroam/blob/master/verbose_dump.txt
Maybe @igrr will be interested too.

sarkrui, 2019-04-03:
FyYI, TU Eindhoven uses EAP-PEAP (MSCHAPv2). @martinius96 Did you try to make serval HTTP requests in you uni network?  I did manage to connect to WPA-Enterprise though, ESP-12E automatically restarted and returned RST 2, mode 2,6 error.

SalmonLoki, 2019-05-30:
> 
> 
> FyYI, TU Eindhoven uses EAP-PEAP (MSCHAPv2). @martinius96 Did you try to make serval HTTP requests in you uni network? I did manage to connect to WPA-Enterprise though, ESP-12E automatically restarted and returned RST 4, mode 2,6 error.

I have the same eeror now. Have you solved it?

martinius96, 2019-05-30:
I have same problem, connection is sucessful, but after few seconds it will restart automatically. 
I cannot tell you from my position what is problem.

sarkrui, 2019-05-30:
> > FyYI, TU Eindhoven uses EAP-PEAP (MSCHAPv2). @martinius96 Did you try to make serval HTTP requests in you uni network? I did manage to connect to WPA-Enterprise though, ESP-12E automatically restarted and returned RST 4, mode 2,6 error.
> 
> I have the same problem now. Have you solved it?

Same, that forced me to switch to ESP32. FYI, at the TU Eindhoven, they are now constructing a new WPA Personal AP for IoT projects in particular. You may try asking if you have the same at your uni.

martinius96, 2020-05-08:
something new? :)

ro371, 2024-10-08:
> #include <ESP8266WiFi.h>
>
> extern "C" { #include "user_interface.h" #include "wpa2_enterprise.h" }
>
> // SSID to connect to static const char* ssid = ""; // Username for authentification static const [...]

I am not able to reproduce the connection using v2.5.0. Have you found out another way to use PEAP?

mhightower83, 2024-10-26:
@ro371 
Adding `enable_wifi_enterprise_patch()` to `setup()` should resolve reconnect memory leaks and provide a larger system stack for WiFi Enterprise system calls.
I tested with:
*  SDK v3.05 
* Relevant network hardware used: ASUS RT-N15 and FreeRadius on pfsense with the package "freeradius3" installed. 
* `Minimum TLS version`: was set to v1.1.
* My notes, indicate TLS v1.2  is not supported by the SDK.  - I reconfirmed it will fail with Minimum TLS set to v1.2.

For those who are familiar with FreeRadius logs (I am not - anymore)
A successful log entry looks like this:
```
Oct 25 18:53:34	radiusd	15401	(34) Login OK: [Bob/<via Auth-Type = mschap>] (from client asus-1 port 0 via TLS tunnel)
Oct 25 18:53:34	radiusd	15401	(35) Login OK: [Bob/<via Auth-Type = eap>] (from client asus-1 port 11 cli b4e62d6a3a9b)
```

Changes applied to https://github.com/esp8266/Arduino/issues/4045#issuecomment-478119686 example
```cpp
#include <ESP8266WiFi.h>
#include <coredecls.h>              // enable_wifi_enterprise_patch()

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

// SSID to connect to
static const char* ssid = "";
// Username for authentification
static const char* username = "";
// Password for authentication
static const char* password = "";

char buff[20];
String ip;

void setup() {
  enable_wifi_enterprise_patch();

  Serial.begin(115200);

  delay(500);

  // WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
  wifi_set_opmode(STATION_MODE);
  struct station_config wifi_config;
  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  wifi_station_set_config(&wifi_config);
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));
  wifi_station_connect();
  // WPA2 Connection ends here

  // Wait for connection AND IP address from DHCP
  Serial.println();
  Serial.println("Waiting for connection and IP Address from DHCP");
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  IPAddress myAddr = WiFi.localIP();
  sprintf(buff, "%d.%d.%d.%d", myAddr[0], myAddr[1], myAddr[2], myAddr[3]);
  ip = String(buff);
  Serial.println(ip);

}

void loop() {
}
```
Further dialog may belong in Discussions or if you can create a minimum sketch to recreate a problem, open a new issue.