esp8266/Arduino #4055: Exception 29 - crash when using the Ticker timer

rioan28, 2017-12-31:
Hello all,
I have been able to run my sketch with no problem until I attempted to instead of running the main part of it inside the Loop function, to put it inside a Ticker timer. Now I get resets every single time and not sure what is wrong.
I have posted below the code and the crash report along with the decoder output.

Any help would be greatly appreciated.

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      

### Description
ESP-12 crashes with the following error:

Connecting to rtino50801_2G
....
WiFi Connected
IP address: -----erased----
MAC Address: ----erased----
Signal: -51 dBm

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  402027a8 0000012a 00000000 60000600  
3fffff60:  3ffee434 3ffee5b4 3ffefedc 40202060  
3fffff70:  00670afb 3ffede60 3ffefedc 402198f5  
3fffff80:  4021993a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffede70 3fffdad0 3ffee5e0 402033b3  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld
⸮

The Decoder comes up with:

Decoding 8 results
0x402027a8: ESP8266WiFiSTAClass::RSSI() at C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 518
0x40202060: Temp_SignalStrength_Function() at C:\Users\Tino\Timer_Issue/Timer_Issue.ino line 102
0x402198f5: ets_timer_handler_isr at ?? line ?
0x4021993a: ets_timer_handler_isr at ?? line ?
0x402033b3: loop_task at C:\Users\Tino\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

### Settings in IDE

Module:  			Nodemcu ESP-12
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			
Flash Frequency:	
Upload Using:		
Reset Method:		

### Sketch
// Including the following libraries
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Wire.h>
#include <SSD1306.h>
#include <Ticker.h>

// WiFi Parameters
const char* ssid = "removed";
const char* password = "removed";

// Include custom images
#include "C:\Users\Tino\Documents\Arduino\libraries\images.h"

// Initialize the OLED display using Wire library
SSD1306  display(0x3c, D1, D4);  // SDA then SCLK

int TempStatus=0;
static int SignalStrength=0;

Ticker Temp_SignalStrength;

// ************************************* Setup ********************************************
void setup() {
  // Initializing serial port for debugging purposes
  Serial.begin(115200);
  delay(10);
  
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid,password);
  while(WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");  
  }
 
  Serial.println("");
  Serial.println("WiFi Connected");

  // Print the IP Address
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Print MAC address
  Serial.print("MAC Address: ");
  Serial.printf("%s\n", WiFi.BSSIDstr().c_str());

  delay(1000);

  Temp_SignalStrength.attach(2,Temp_SignalStrength_Function);
  
} // End of Setup function

// ************************************* END Setup ********************************************

// ***************************************************************************************************
// ************************************** Main Loop **************************************************
// ***************************************************************************************************
void loop() 
  {
    
  }

  // ***************************************************************************************************
  // ************************************** END Main Loop **********************************************
  // ***************************************************************************************************

 
void Temp_SignalStrength_Function()
{    
  
  
  SignalStrength = WiFi.RSSI();
  Serial.printf("Signal: %d dBm\n", SignalStrength );

  display.setTextAlignment(TEXT_ALIGN_RIGHT);
  display.clear();
  
  if ( SignalStrength >= -50 )
  {
    Serial.println("5 bars");
    display.drawXbm(95, 0, Wifi_5bar_width, Wifi_5bar_height, Wifi_5bar_bits);
   
  }
  // A lot of code removed here
  else
  {
    Serial.println("No signal");
  }    
}

String bssidToString(uint8_t *bssid) 
{
  char mac[20] = {0};
  sprintf(mac,"[%02X:%02X:%02X:%02X:%02X:%02X]", bssid[0],  bssid[1],  bssid[2], bssid[3], bssid[4], bssid[5]);
  return String(mac);
}

  

### Debug Messages

```
messages here
```

devyte, 2017-12-31:
@rioan28 without looking too deep into this, I think you're making the typical mistake of trying to do blocking operations from the Ticker callback. What you should do in the Ticker callback is set a flag and check it inside the loop to run your code, or schedule a function to be executed with the function scheduler interface (it will execute right after the loop).
Closing as user error.

closed 2017-12-31 by devyte

rioan28, 2017-12-31:
Hello devyte, I am sorry but I don't think I understand you explanation. Yes I am a newbie. 
I don't have anything running in my Loop(), the code gets executed when the timer fires.
Would you be so kind to detail your explanation.

Thank you.

devyte, 2017-12-31:
The display methods use Wire, which is blocking,  and you can't do that from within a Ticker callback. So, do one of:
-set a flag in the Ticker callback, check it in the loop, and when true clear it and execute your code.
-schedule a function call inside the Ticker callback. The function you schedule executes your code. Because it runs as part of the loop (right after), that function can be blocking. Issue #4048 has a scheduling example, although it's for a function that reschedules itself, which you don't want.

hreintke, 2017-12-31:
This is how I write my text in a Ticker callback to the SSD using a scheduled function.

```
DisplayControl::DisplayControl(SSD1306* inputDisplay)
:ssdDisplay(inputDisplay)
{
}

DisplayControl::~DisplayControl() {
}

void DisplayControl::showText(String reqString)
{
    schedule_function(std::bind(&DisplayControl::internalShowText,this,reqString));
}

void DisplayControl::internalShowText(String reqString)
{
	ssdDisplay->clear();
	ssdDisplay->flipScreenVertically();
    ssdDisplay->setFont(ArialMT_Plain_24);
    ssdDisplay->drawString(0, 26, reqString.c_str());
    ssdDisplay->display();
}
```

@devyte 
I have a local patch for Ticker that gives an option to run the callback in a scheduled function "automatic"
If you want I can make a PR for that

devyte, 2017-12-31:
Alright, please  make the PR and I'll take a look.

rioan28, 2017-12-31:
@hreintke  thank you very much.

igrr, 2017-12-31:
> run the callback in a scheduled function "automatic"

If it doesn't change the default behavior, that's a nice option to add.

rioan28, 2017-12-31:
@igrr I am very sorry but don't really understand what you mean.  I have programmed C many years ago and right now I am trying to catch up again.
Would you mind giving me a little more detail?
Thank you very much.

devyte, 2017-12-31:
@rioan28 that was meant for @hreintke , who is making a PR with an enhancement that would make your use case a bit easier.

rioan28, 2018-01-01:
@igrr  Thank you very much and Happy New Year!

rioan28, 2018-01-01:
@hreintke Happy New Year!

rioan28, 2018-01-01:
@devyte Happy New Year!