esp8266/Arduino #4059: ESP8266WebServer does't decode POST parameters

elefa, 2018-01-01:
### Basic Infos
ESP8266WebServer does't decode POST parameters from XMLHttpRequest.

#### Hardware
Hardware:			WeMos Mini 
Core Version:      	lastest from GitHub 

### Description
ESP8266WebServer does't decode POST parameters. 
I use lastest ESP8266WebServer (from yesterday)

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			? Arduino default
Flash Frequency:	? Arduino default
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch
#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer websrv(80);

void setup() {
	Serial.begin(115200);
	SPIFFS.begin();
	WiFi.begin("******","********-");
	Serial.print("Connecting");
	while (WiFi.status() != WL_CONNECTED) 	{
		delay(500);
		Serial.print(".");
	}
	Serial.println("");
	Serial.print("Connected with IP: ");
	Serial.println(WiFi.localIP());
	websrv.on("/", handleRoot);
	websrv.on("/saveconfig", handleSaveConfig);
	websrv.begin();
}

void loop() {
	websrv.handleClient();
}

void handleRoot() {
	File f = SPIFFS.open("/index.html", "r");
	size_t sent = websrv.streamFile(f, "text/html");
	Serial.println("File /index.html sent");
	f.close();
}

void handleSaveConfig() {
	Serial.print("Number of args: ");
	Serial.println(websrv.args());
	Serial.print("Argument(0): ");
	Serial.println(websrv.arg(0));
	websrv.send(200, "text/html", String(websrv.arg(0)));
}
### Debug Messages
pm open,type:2 0
New client
method: POST url: /saveconfig search: 
headerName: Host
headerValue: 192.168.254.23
headerName: Connection
headerValue: keep-alive
headerName: Content-Length
headerValue: 25
headerName: Origin
headerValue: http://192.168.254.23
headerName: User-Agent
headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.108 Safari/537.36
headerName: Content-Type
headerValue: text/plain;charset=UTF-8
headerName: Accept
headerValue: */*
headerName: Referer
headerValue: http://192.168.254.23/
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Accept-Language
headerValue: cs-CZ,cs;q=0.9,en;q=0.8
args: 
Plain: arg1=uno&arg2=two&arg3=77
Request: /saveconfig
 Arguments: 
Number of args: 1
Argument(0): arg1=uno&arg2=two&arg3=77

[files.zip](https://github.com/esp8266/Arduino/files/1595556/files.zip)

Firehed, 2018-01-01:
Not sure if it's related, but I just ran into a very similar issue (if not the same). My HTTP client was lowercasing the `Content-Length` header, and as a result this library would fail to parse the POST body correctly.

As a short-term work around I was able to put the POST body as query string parameters, but that's not a real solution.

devyte, 2018-01-01:
@scop could you please take a look at this?

caverna, 2018-01-02:
another tip would be include **HTTP_POST**:
```websrv.on("/saveconfig", HTTP_POST, handleSaveConfig);```

elefa, 2018-01-02:
I tried it. Unfortunately, no change.

scop, 2018-01-16:
> headerName: Content-Type
> headerValue: text/plain;charset=UTF-8

No decoding should occur for text/plain POSTs. Try setting Content-Type to application/x-www-form-urlencoded

elefa, 2018-01-16:
Yes, it works. 
Thank you for your advice

closed 2018-01-16 by elefa