esp8266/Arduino #4060: HTTPClient causes exception(28) in 2.4.0-rc2

jp112sdl, 2018-01-01:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0-rc2

### Description

HTTPClient causes exception(28)

### Settings in IDE

Module:  			Generic ESP8285 Module
Flash Size: 		1MB
Flash Mode:			DOUT
Upload Using:		OTA
Reset Method:		ck

### Sketch

```cpp
bool setStateCUxD(String id, String value) {
  if (id.indexOf(".null.") == -1 && String(GlobalConfig.ccuIP) != "0.0.0.0") {
    if (WiFi.status() == WL_CONNECTED) {
      HTTPClient http;
      http.setTimeout(1500);
      id.replace(" ", "%20");
      String url = "http://" + String(GlobalConfig.ccuIP) + ":8181/cuxd.exe?ret=dom.GetObject(%22" + id + "%22).State(" + value + ")";
      http.begin(url);
      int httpCode = http.GET();
      String payload = "";

      if (httpCode > 0) {
        payload = http.getString();
      }

      http.end();

      payload = payload.substring(payload.indexOf("<ret>"));
      payload = payload.substring(5, payload.indexOf("</ret>"));
      return (payload != "null");

    } else {
      if (!doWifiConnect())
        ESP.restart();
    }
  } else return true;
}
```

### Debug Messages
```
0x402443ad: tcp_pcb_purge at core/tcp.c line 1446
0x402446c7: tcp_pcb_remove at core/tcp.c line 1468 (discriminator 1)
0x402448a4: tcp_abandon at core/tcp.c line 372
0x40244902: tcp_abort at core/tcp.c line 406
0x4020137a: delay at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/core_esp8266_wiring.c line 54
0x40211984: ClientContext::abort() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
0x40211d02: ClientContext::connect(ip_addr*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 135
0x402117f8: WiFiClient::connect(char const*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
0x40212a07: TransportTraits::create() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40212e94: HTTPClient::connect() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4021ab27: String::changeBuffer(unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 518
0x40213664: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x401004d8: malloc at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1668
0x4010020c: _umm_free at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40213712: HTTPClient::GET() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4020b25c: setStateCUxD(String, String) at /Users/user/Documents/Arduino/ECHT-Einsatz/InUse/SonoffHMLOX/HomeMatic.ino line 11
```

In my code I'm calling a http.Get() every 30 seconds.
That works for a few hours. 
Then I get a few times a HTTP Return Code (-1) (```HTTPC_ERROR_CONNECTION_REFUSED```) and after that it seems that the WiFi connection is lost.
A new attempt to establish the Wifi connection fails.

With the latest stable Version 2.3.0 everything works fine.

devyte, 2018-01-01:
Please retest with latest git.

jp112sdl, 2018-01-01:
With the latest git the beaviour is another, when the HTTP Error -1 occurs. 
I enabled Debug for "WIFI" and got the following:

```
[hostByName] Host: 192.168.1.251 is a IP!
setStateCUxD url: http://192.168.1.252:8181/cuxd.exe?ret=dom.GetObject(%22Pow1EC%22).State(5.86)
[hostByName] Host: 192.168.1.252 is a IP!
[hostByName] Host: 192.168.1.251 is a IP!
HTTP Pow1EC failed with HTTP Error Code -1
[hostByName] Host: 192.168.1.251 is a IP!
result: 
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt 

connected with Apple Network, channel 5
dhcp client start...
wifi evt: 0
pm open,type:2 0
[hostByName] Host: 192.168.1.251 is a IP!
[hostByName] Host: 192.168.1.251 is a IP!
Connecting WiFi...
.sl
scandone
usl
.............................
```
-> WiFi Connection fails, so I do a ESP.restart();

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld
````

jp112sdl, 2018-01-01:
It seems that the condition `(WiFi.status() == WL_CONNECTED)` also is not fulfilled although the WIFI Debug message writes: `connected with Apple Network, channel 5`

In my code I do a WiFi re-connect, with a wait time of 15 seconds to connect:
``` 
WiFi.mode(WIFI_STA);
WiFi.begin(_ssid.c_str(), _psk.c_str());
    while (WiFi.status() != WL_CONNECTED) {
      waitCounter++;
      Serial.print(".");
      if (waitCounter == 30) {
        return false;
      }
      delay(500);
    }
```

The only way to reconnect to the WiFi network is restarting the device.

jp112sdl, 2018-01-02:
I have created a new small sample sketch:
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

char ssid[] = "none";
char key[] = "thisissecret";

unsigned long lastMillis = 0;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  if (doWifiConnect()) {
    Serial.println("WiFi connection successful!");
  } else ESP.restart();
}

void loop() {
  if (millis() - lastMillis > 5000) {
    lastMillis = millis();
    doHttpCall();
  }
}

void doHttpCall() {
  Serial.println("Current WiFi.status() = "+String(WiFi.status()));
  if (WiFi.status() == WL_CONNECTED) {
    HTTPClient http;
    http.setTimeout(1500);
    http.begin("http://192.168.1.1/index.htm");
    int httpCode = http.GET();
    if (httpCode > 0) {
      Serial.println("HTTP success");
      String payload = http.getString();
    }
    if (httpCode != 200) {
      Serial.println("HTTP fail " + String(httpCode));
    }
    http.end();
  } else {
    //WiFi.reconnect();
    //doWifiConnect();
  }
}

bool doWifiConnect() {
  Serial.println("Connecting WLAN...");
  //WiFi.disconnect(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, key);
  int waitCounter = 0;
  digitalWrite(LED_BUILTIN, LOW);
  while (WiFi.status() != WL_CONNECTED) {
    waitCounter++;
    if (waitCounter == 20) {
      digitalWrite(LED_BUILTIN, HIGH);
      return false;
    }
    delay(500);
  }
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.println("Wifi Connected");
  return true;
}
```

The Debug Output is:
```
Connecting WLAN...
wifi evt: 2
wifi evt: 0
wifi evt: 3
Wifi Connected
WiFi connection successful!
Current WiFi.status() = 3
[hostByName] Host: 192.168.1.1 is a IP!
HTTP success
Current WiFi.status() = 3
[hostByName] Host: 192.168.1.1 is a IP!
HTTP success
```
**--- Now here I've disconnected my access point ---**
```
pm open,type:2 0
Current WiFi.status() = 3
[hostByName] Host: 192.168.1.1 is a IP!
HTTP fail -1
bcn_timout,ap_probe_send_start
Current WiFi.status() = 3
[hostByName] Host: 192.168.1.1 is a IP!
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200
Fatal exception 28(LoadProhibitedCause):
epc1=0x40202db5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000054, depc=0x00000000
```

The decoded stack is:
```
0x40202dae: WiFiClient::connect(IPAddress, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x402049bd: String::operator=(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 545
0x40202974: WiFiClient::connect(char const*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x40203213: TransportTraits::create() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 53
0x402036a8: HTTPClient::connect() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 53
0x402047cb: String::changeBuffer(unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 545
0x40203ea0: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 53
0x401007ac: malloc at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1668
0x4010020c: _umm_free at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40203f4e: HTTPClient::GET() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 53
0x40202292: doHttpCall() at /Users/user/Documents/Arduino/test_wifi_httpclient/test_wifi_httpclient.ino line 30
0x40204480: Print::println(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/Print.cpp line 107
0x40202339: loop at /Users/user/Documents/Arduino/test_wifi_httpclient/test_wifi_httpclient.ino line 22
0x402052d0: loop_wrapper at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/core_esp8266_main.cpp line 57
0x40100a28: cont_norm at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/cont.S line 109
```

devyte, 2018-01-06:
@jp112sdl Please retest with the referenced PR.

jp112sdl, 2018-01-06:
@devyte 
Tested immediately. 
Looks very good! 
The exception does not occur anymore when I turn off my access point and power it on again. 
WiFi reconnects automatically and HTTPClient works fine after Wifi reconnect 👍 

I will let it run a few hours... sometimes the exception also occured after WPA Re-Keying interval.

My settings were:
[image]

closed 2018-08-01 by devyte