esp8266/Arduino #4061: Writing files with SPIFFS no longer working for ESP-01

funtastic99, 2018-01-01:
### Basic Infos
SPIFFS no longer works on a ESP-01.

#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0-rc2
                                also tried with stable

### Description

Problem description
SPIFFS worked for a long time but all of a sudden it stoped working on a ESP-01. 
Whenever i write a file its not existing or empty after next boot.
Using the same sketch on an Wemos D1 mini its works without problems. Actually im only facing this problem on a ESP-01.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		 1MB (64k SPIFFS), also tried with 512k (64k SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	     40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <FS.h>

void setup()
{
   Serial.begin(115200);

   if (!SPIFFS.begin()) Serial.println("error while mounting filesystem!");

   readFile();
   writeFile();
   readFile();

   Serial.println("done");
}

void readFile()
{
  Serial.println("reading");
  File f = SPIFFS.open("/myFile.txt", "r");
  if (!f) Serial.println("file not available");
  else if (f.available()<=0) Serial.println("file exists but available <0");
  else
  {
    String ssidString = f.readStringUntil('#');
    Serial.print("read from file: ");
    Serial.println(ssidString);
  }
  f.close();
}

void writeFile()
{
  Serial.println("writing");
  File f = SPIFFS.open("/myFile.txt", "w");
  if (!f) Serial.println("File creation failed");
  else
  {
    f.print("networkConfig");
    f.print("#");
    f.flush();
    f.close();
  }
}

void loop()
{
}

### Debug Messages

messages here

- Output of first run:
reading
file not available
writing
reading
read from file:          <--- here we see the problem already
done

- Output of second run:
eading
file not available
writing
reading
read from file: 
done

- Output of third run:
reading
read from file: 
writing
File creation failed    <---- check this!
reading
read from file: 
done

Any ideas?
thank you!

funtastic99, 2018-01-03:
I found out some additional information that might be helpful.
The working ESP-01 report as flash_id:
Manufacturer: e0
Device: 4014

and the new ones that are not working:
Manufacturer: 85
Device: 6014

May the new flash chip is not supported by the arduino esp plugin yet ?

ondabeach, 2018-01-05:
I'm having the same problem. 

Out of the first batch of 20 where the problem first showed up only 7 of them worked properly. All functions except for SPIFFS seem fine.

Out of the next batch of 20, 9 worked properly.

I just received another 2 batches of 20. One batch from a different seller on eBay and the other batch from a seller on AliExpress.

Out of the new batch from eBay only 2 work properly.

All 20 of the new batch from Aliexpress are faulty.

I checked the Chip ID's and I can confirm that Flash Chips with a manufacturer ID of 85 and Device ID of 6014 do not work. I can also confirm that the 2 that did work have the manufacturer ID of E0 and Chip ID of 4014.

I had a look at the flash chips with a magnifying glass and found the faulty ones were all branded FUYA and the 2 that have working SPIFFS are branded BergMicro.

I sure hope someone has a solution for this as it's a major problem for anyone who needs to store stuff like settings. EEPROM is an alternative but it's so clunky compared to SPIFFS and I really don't want to have to recode my sketches.

funtastic99, 2018-01-05:
Its definitely a problem with writing SPIFFS. Reading works because i uploaded a flash image of a functional device with SPIFFS data present. But for reading the flash as anyway only mapped to memory while for writing a correct initializiation and timing is needed. I wonder why the booter itself has no problems writing the flash but SPIFFS has. I would have expected that finally the same functions are used to write the flash but it seems that the spiffs implementation is doing something strang and chip dependend.

Pablo2048, 2018-01-05:
So it seems like problem in SPIFFS and/or write implementation. Right?

ondabeach, 2018-01-05:
I found that the 'Upload Sketch Data' function of the IDE works. A sketch can then read the data but a sketch can't write to SPIFFS.

This will impact a lot of people, I've bought esp-01 modules witj these PUYA branded flash chips from 3 different suppliers now. Out of the 80 modules only 18 have worked as they had BergMicro flash chips onboard.

weswitt, 2018-01-06:
I'm seeing a similar, but different problem. My sketch does not find any files in the fs after uploading the spiffs image. I're tried uploading with esptool and the Arduino IDE, but I get the same result. The processor is an ESP01 on a Sonoff device. I've tried both the 512k64k and 1m64k configurations and neither work.

Is the issue I'm seeing related to this report? Am I possibly doing something wrong? What is the best way to diagnose this? I've added some code to the setup function that lists all files and it finds none.

ondabeach, 2018-01-06:
@weswitt is this a new board purchased recently?
Can you check the writing on the Flash chip, is it PUYA brand? Look for a number like P25Q80?.
Or you can load and run the following code to get the chip data. You're looking for the manufacturer ID of 85. If you have that flash chip then yes, you have the same problem as us.

//*************************************************
#include "FS.h"

void setup() {
  Serial.begin(9600);

  SPIFFS.begin();
  FSInfo fs_info;
  SPIFFS.info(fs_info);

  float fileTotalKB = (float)fs_info.totalBytes / 1024.0; 
  float fileUsedKB = (float)fs_info.usedBytes / 1024.0; 

  float flashChipSize = (float)ESP.getFlashChipSize() / 1024.0 / 1024.0;
  float realFlashChipSize = (float)ESP.getFlashChipRealSize() / 1024.0 / 1024.0;
  float flashFreq = (float)ESP.getFlashChipSpeed() / 1000.0 / 1000.0;
  FlashMode_t ideMode = ESP.getFlashChipMode();

  Serial.printf("\n#####################\n");

  Serial.printf("__________________________\n\n");
  Serial.println("Firmware: ");
  Serial.printf("    Chip Id: %08X\n", ESP.getChipId());
  Serial.print("    Core version: "); Serial.println(ESP.getCoreVersion());
  Serial.print("    SDK version: "); Serial.println(ESP.getSdkVersion());
  Serial.print("    Boot version: "); Serial.println(ESP.getBootVersion());
  Serial.print("    Boot mode: "); Serial.println(ESP.getBootMode());

  Serial.printf("__________________________\n\n");

  Serial.println("Flash chip information: ");
  Serial.printf("    Flash chip Id: %08X (for example: Id=001640E0  Manuf=E0, Device=4016 (swap bytes))\n", ESP.getFlashChipId());
  Serial.printf("    Sketch thinks Flash RAM is size: ");  Serial.print(flashChipSize); Serial.println(" MB");
  Serial.print("    Actual size based on chip Id: "); Serial.print(realFlashChipSize); Serial.println(" MB ... given by (2^( \"Device\" - 1) / 8 / 1024");
  Serial.print("    Flash frequency: "); Serial.print(flashFreq); Serial.println(" MHz");
  Serial.printf("    Flash write mode: %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));

  Serial.printf("__________________________\n\n");

  Serial.println("File system (SPIFFS): ");
  Serial.print("    Total KB: "); Serial.print(fileTotalKB); Serial.println(" KB");
  Serial.print("    Used KB: "); Serial.print(fileUsedKB); Serial.println(" KB");
  Serial.printf("    Block size: %lu\n", fs_info.blockSize);
  Serial.printf("    Page size: %lu\n", fs_info.pageSize);
  Serial.printf("    Maximum open files: %lu\n", fs_info.maxOpenFiles);
  Serial.printf("    Maximum path length: %lu\n\n", fs_info.maxPathLength);

  Dir dir = SPIFFS.openDir("/");
  Serial.println("SPIFFS directory {/} :");
  while (dir.next()) {
    Serial.print("  "); Serial.println(dir.fileName());
    Serial.println("  "); Serial.println(dir.fileSize());
  }

  Serial.printf("__________________________\n\n");

  Serial.printf("CPU frequency: %u MHz\n\n", ESP.getCpuFreqMHz());
  Serial.print("#####################");

  // open file for reading
  File f = SPIFFS.open("/login.txt", "r");
  if (!f) {
      Serial.println("file open failed");
  }  Serial.println("====== Reading from SPIFFS file =======");
  // write 10 strings to file
  for (int i=1; i<=10; i++){
    String s=f.readStringUntil('\n');
    Serial.print(i);
    Serial.print(":");
    Serial.println(s);
  }
}

void loop() {
  delay(2);
}
//********************************************

devyte, 2018-01-06:
Same issue as #3616 , tracking there.
Closing this one.

weswitt, 2018-01-06:
@ondabeach thanks for the assistance. below is the diagnostic output from the sketch you provided:

#####################
__________________________

Firmware:
 Chip Id: 0042A910
 Core version: 2_4_0
 SDK version: 2.1.0(deb1901)
 Boot version: 31
 Boot mode: 1
__________________________

Flash chip information:
 Flash chip Id: 001440E0 (for example: Id=001640E0 Manuf=E0, Device=4016 (swap bytes))
 Sketch thinks Flash RAM is size: 0.50 MB
 Actual size based on chip Id: 1.00 MB ... given by (2^( "Device" - 1) / 8 / 1024
 Flash frequency: 40.00 MHz
 Flash write mode: DIO
__________________________

File system (SPIFFS):
 Total KB: 51.72 KB
 Used KB: 0.00 KB
 Block size: 4096
 Page size: 256
 Maximum open files: 5
 Maximum path length: 32

SPIFFS directory {/} :
__________________________

CPU frequency: 80 MHz

#####################file open failed
====== Reading from SPIFFS file =======
1:
2:
3:
4:
5:
6:
7:
8:
9:
10:

ondabeach, 2018-01-06:
@weswitt The last sketch was just to get the flash chip data, the printing of 1~10 at the bottom was left in by mistake.

The chip ID retrieved shows that you have a flash chip that should work, it does for the OP and myself. 
FYI, you have flash size set to 512k 64k but your board actually has 1MB. That isn't an issue as the actual size is bigger than your setting.

Try the following code, it should work on your board.

//**********************************************************
#include "FS.h"

String realSize = String(ESP.getFlashChipRealSize());
String ideSize = String(ESP.getFlashChipSize());
bool flashCorrectlyConfigured = realSize.equals(ideSize);

void setup() {
  Serial.begin(9600);

  if(flashCorrectlyConfigured){
    SPIFFS.begin();
    Serial.println("flash correctly configured, SPIFFS started, IDE size: " + ideSize + ", real size: " + realSize);
  }
  else Serial.println("flash incorrectly configured, SPIFFS cannot start, IDE size: " + ideSize + ", real size: " + realSize);
  
  Serial.println("\nVery basic Spiffs example, writing 10 lines to SPIFFS filesystem, and then read them back");
  SPIFFS.begin();

  if (!SPIFFS.exists("/formatComplete.txt")) {
    Serial.println("Please wait 30 secs for SPIFFS to be formatted");
    SPIFFS.format();
    //delay(30000);
    Serial.println("Spiffs formatted");
    
    File f = SPIFFS.open("/formatComplete.txt", "w");
    if (!f) {
        Serial.println("file open failed");
    } else {
        f.println("Format Complete");
    }
  } else {
    Serial.println("SPIFFS is formatted. Moving along...");
  }
}

void loop() {
  // open file for writing
  File f = SPIFFS.open("/f.txt", "w");
  if (!f) {
      Serial.println("file open failed");
  }
  Serial.println("====== Writing to SPIFFS file =========");
  // write 10 strings to file
  for (int i=1; i<=10; i++){
    f.print("Millis() : ");
    f.println(millis());
    Serial.println(millis());
  }

  f.close();

  // open file for reading
  f = SPIFFS.open("/f.txt", "r");
  if (!f) {
      Serial.println("file open failed");
  }  Serial.println("====== Reading from SPIFFS file =======");
  // write 10 strings to file
  for (int i=1; i<=10; i++){
    String s=f.readStringUntil('\n');
    Serial.print(i);
    Serial.print(":");
    Serial.println(s);
  }
  // wait a few seconds before doing it all over again
  delay(3000);
  
}
//*************************************************************************

weswitt, 2018-01-07:
Thanks @ondabeach. Running that code does confirm that writing to and reading from SPIFFS work fine on my device.

But this brings me back to my problem. What I'm seeing is that when I flash a SPIFFS to the device and then run a sketch the sketch sees zero files.

This happens if I use the Arduino IDE or esptool. This is the esptool command line that I'm using:

esptool.exe -vv -cd ck -cb 115200 -cp com3 -ca 0xEB000 -cf test-spiffs.bin

Does the tool used to flash the SPIFFS need to be up todate with the SPIFFS in the core? Maybe the tool is too old and is not writing a format that the core runtime understands?

ondabeach, 2018-01-07:
No problem @weswitt. 

I'm not sure if esptool needs to.be updated. Much quicker and simpler to use the IDE as you're already in there rather than using command line methods.

We know your hardware is ok and that sketch can write and read SPIFFS so to look at your uploading files to SPIFFS issue:

No idea of your experience level, I'm just a hacker myself so we'll start with the basics.

Use "Show Sketch Folder" under "Sketch" menu.
Create a folder there called "data".
Create/copy your files there. (For testing just create a text file there called "Test.txt" with contents as follows:
Line 1
Line 2
Line 3
Line 4
Line 5
Line 6
Line 7
Line 8
Line 9
Line 10

Make sure you carriage return after the "10" or that line won't be read.
Select "Sketch Data Upload" from Tools menu. (Make sure serial monitor is not open and board is in upload mode)

Here's a tidied up version of the Get Flash Data/SPIFFS Read sketch.

//*************************************************************
#include "FS.h"

void setup() {
  Serial.begin(9600);

  SPIFFS.begin();
  FSInfo fs_info;
  SPIFFS.info(fs_info);

  float fileTotalKB = (float)fs_info.totalBytes / 1024.0; 
  float fileUsedKB = (float)fs_info.usedBytes / 1024.0; 

  float flashChipSize = (float)ESP.getFlashChipSize() / 1024.0 / 1024.0;
  float realFlashChipSize = (float)ESP.getFlashChipRealSize() / 1024.0 / 1024.0;
  float flashFreq = (float)ESP.getFlashChipSpeed() / 1000.0 / 1000.0;
  FlashMode_t ideMode = ESP.getFlashChipMode();

  Serial.println("=========================================================="); 
  Serial.println("Firmware: ");
  Serial.printf("    Chip Id: %08X\n", ESP.getChipId());
  Serial.print("    Core version: "); Serial.println(ESP.getCoreVersion());
  Serial.print("    SDK version: "); Serial.println(ESP.getSdkVersion());
  Serial.print("    Boot version: "); Serial.println(ESP.getBootVersion());
  Serial.print("    Boot mode: "); Serial.println(ESP.getBootMode());

  Serial.printf("__________________________\n\n");

  Serial.println("Flash chip information: ");
  Serial.printf("    Flash chip Id: %08X (for example: Id=001640E0  Manuf=E0, Device=4016 (swap bytes))\n", ESP.getFlashChipId());
  Serial.printf("    Sketch thinks Flash RAM is size: ");  Serial.print(flashChipSize); Serial.println(" MB");
  Serial.print("    Actual size based on chip Id: "); Serial.print(realFlashChipSize); Serial.println(" MB ... given by (2^( \"Device\" - 1) / 8 / 1024");
  Serial.print("    Flash frequency: "); Serial.print(flashFreq); Serial.println(" MHz");
  Serial.printf("    Flash write mode: %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));
  Serial.printf("    CPU frequency: %u MHz\n\n", ESP.getCpuFreqMHz());
  Serial.printf("__________________________\n\n");

  Serial.println("File system (SPIFFS): ");
  Serial.print("    Total KB: "); Serial.print(fileTotalKB); Serial.println(" KB");
  Serial.print("    Used KB: "); Serial.print(fileUsedKB); Serial.println(" KB");
  Serial.printf("    Block size: %lu\n", fs_info.blockSize);
  Serial.printf("    Page size: %lu\n", fs_info.pageSize);
  Serial.printf("    Maximum open files: %lu\n", fs_info.maxOpenFiles);
  Serial.printf("    Maximum path length: %lu\n\n", fs_info.maxPathLength);
  Serial.printf("__________________________\n\n");

  
  Dir dir = SPIFFS.openDir("/");
  Serial.println("SPIFFS directory {/} :");
  while (dir.next()) {
    Serial.print("File Name    "); Serial.println(dir.fileName());
    Serial.print("File Length  "); Serial.println(dir.fileSize());
    Serial.print("====== First 10 lines of file ("); Serial.print(dir.fileName()); Serial.println(") =======");
    // open file for reading
    File f = SPIFFS.open("/Test.txt", "r");
    if (!f) {
        Serial.println("file open failed");
    }
    else{  
    // read 10 strings from file 
    for (int i=1; i<=10; i++){
      String s=f.readStringUntil('\n');
      Serial.print(i);
      Serial.print(" : ");
      Serial.println(s);
      }
    }
    Serial.println("==========================================================");   
  }

}

void loop() {
  delay(3000);
}
//****************************************************************

ondabeach, 2018-01-07:
Hi @igrr, here's the output with the PUYA chip:

phys_addr=962560
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 1048576
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Mounting FS...
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
33: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

ondabeach, 2018-01-07:
@igrr For comparison, this is the output from a BergMicro chip on one of my last 2 modules that can use SPIFFS (hint hint ;)

phys_addr=962560
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   001440E0
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 1048576
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Mounting FS...
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
33: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
File contents:
foobar

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=14
File contents:
foobar
foobar

---
SPIFFS_close: fd=1

igrr, 2018-01-07:
@ondabeach thanks for these logs! What happens if you run the test on the PUYA module, setting flash configuration in IDE to, say, 512K/64K?

ondabeach, 2018-01-07:
@igrr pretty much the same result Ivan. Thanks for your help with this issue. Whatever you need, just let me know.

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 524288
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Mounting FS...
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
33: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

igrr, 2018-01-07:
Hmm, that doesn't give any clue, to be honest. Okay, let's try another kind of a test:

https://gist.github.com/igrr/964b0d139ce63fcee07c8c1f1facb24d

Same as before, please have debug level set to CORE. No need to run on the "working" board, just on the PUYA one.

funtastic99, 2018-01-07:
Hi igrr,
i am also still facing this problem but as a workaround now switched to eeprom.
I also loaded your test sketch. Here is the output:

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 12
cnt 

connected with CAROLAN, channel 6
dhcp client start...
phys_addr=962560
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 1048576
Testing write...
Test data partition @ 0xeb000
dst=0 src=0 len=0
dst=0 src=0 len=4
dst=0 src=0 len=16
dst=0 src=0 len=64
dst=0 src=0 len=1
dst=0 src=1 len=1
dst=1 src=0 len=1
dst=1 src=1 len=1
dst=1 src=1 len=2
dst=1 src=1 len=3
dst=1 src=1 len=4
dst=1 src=1 len=5
dst=3 src=2 len=5
dst=4 src=4 len=60
dst=59 src=0 len=5
dst=60 src=0 len=4
dst=60 src=0 len=3
dst=60 src=0 len=2
dst=63 src=0 len=1
dst=64 src=0 len=0
dst=59 src=59 len=5
dst=60 src=60 len=4
dst=60 src=60 len=3
dst=60 src=60 len=2
dst=63 src=63 len=1
dst=64 src=64 len=0
Testing write/read...
Seed=0 Start=235 Count=16
235: erase write ip:192.168.178.107,mask:255.255.255.0,gw:192.168.178.1
read done
236: erase write read done
237: erase write read done
238: erase write read done
pm open,type:2 0
239: erase write read done
240: erase write read done
241: erase write read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write bcn_timout,ap_probe_send_start
read done
Seed=1 Start=235 Count=16
235: erase write ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
read done
236: erase write scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 12
cnt 

connected with CAROLAN, channel 6
dhcp client start...
read done
237: erase write read done
238: erase write read done
239: erase write read done
240: erase write read done
241: erase write pm open,type:2 0
read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write read done
Seed=42 Start=235 Count=16
235: erase write read done
236: erase write read done
237: erase write read done
238: erase write read done
239: erase write read done
240: erase write read done
241: erase write read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write read done
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Mounting FS...
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
36: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
File contents:
foobar

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=14
File contents:
foobar
foobar

---
SPIFFS_close: fd=1

igrr, 2018-01-07:
@funtastic99 Your output is what i would describe as "normal". Perhaps the issues you and @ondabeach are seeing are not the same after all...

funtastic99, 2018-01-07:
@igrr  The strange thing is that ondebeach an me both have spiffs problems with flash id Manufacturer: 85, Device: 6014. Maybe its causing different effects.
But still the initial testprogramm at the beginning of this ticket is not showing the expected output after writing. Any idea why ?
Thank!

ondabeach, 2018-01-07:
@igrr Ok, exact same settings as last test, same board, nothing moved or changed.

phys_addr=962560
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 1048576
Testing write...
Test data partition @ 0xeb000
dst=0 src=0 len=0
dst=0 src=0 len=4
dst=0 src=0 len=16
dst=0 src=0 len=64
dst=0 src=0 len=1
dst=0 src=1 len=1
dst=1 src=0 len=1
dst=1 src=1 len=1
dst=1 src=1 len=2
dst=1 src=1 len=3
dst=1 src=1 len=4
dst=1 src=1 len=5
dst=3 src=2 len=5
dst=4 src=4 len=60
dst=59 src=0 len=5
dst=60 src=0 len=4
dst=60 src=0 len=3
dst=60 src=0 len=2
dst=63 src=0 len=1
dst=64 src=0 len=0
dst=59 src=59 len=5
dst=60 src=60 len=4
dst=60 src=60 len=3
dst=60 src=60 len=2
dst=63 src=63 len=1
dst=64 src=64 len=0
Testing write/read...
Seed=0 Start=235 Count=16
235: erase write read done
236: erase write read done
237: erase write read done
238: erase write read done
239: erase write read done
240: erase write read done
241: erase write read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write read done
Seed=1 Start=235 Count=16
235: erase write read done
236: erase write read done
237: erase write read done
238: erase write read done
239: erase write read done
240: erase write read done
241: erase write read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write read done
Seed=42 Start=235 Count=16
235: erase write read done
236: erase write read done
237: erase write read done
238: erase write read done
239: erase write read done
240: erase write read done
241: erase write read done
242: erase write read done
243: erase write read done
244: erase write read done
245: erase write read done
246: erase write read done
247: erase write read done
248: erase write read done
249: erase write read done
250: erase write read done
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Mounting FS...
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
36: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

igrr, 2018-01-07:
Sorry, not getting any closer...

Please try replacing Esp.cpp in your esp8266 core directory with the one attached here, run same test.

[Esp.zip](https://github.com/esp8266/Arduino/files/1609884/Esp.zip)

ondabeach, 2018-01-07:
@igrr esp.cpp replaced, IDE restarted and all settings the same.

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 524288
Testing write...
Test data partition @ 0x6b000
dst=0 src=0 len=0
dst=0 src=0 len=4
dst=0 src=0 len=16
dst=0 src=0 len=64
dst=0 src=0 len=1
dst=0 src=1 len=1
dst=1 src=0 len=1
dst=1 src=1 len=1
dst=1 src=1 len=2
dst=1 src=1 len=3
dst=1 src=1 len=4
dst=1 src=1 len=5
dst=3 src=2 len=5
dst=4 src=4 len=60
dst=59 src=0 len=5
dst=60 src=0 len=4
dst=60 src=0 len=3
dst=60 src=0 len=2
dst=63 src=0 len=1
dst=64 src=0 len=0
dst=59 src=59 len=5
dst=60 src=60 len=4
dst=60 src=60 len=3
dst=60 src=60 len=2
dst=63 src=63 len=1
dst=64 src=64 len=0
Testing write/read...
Seed=0 Start=107 Count=16
107: erase write Flash verify error @0x6b000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
108: erase write Flash verify error @0x6c000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
109: erase write Flash verify error @0x6d000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
110: erase write Flash verify error @0x6e000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
111: erase write Flash verify error @0x6f000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
112: erase write Flash verify error @0x70000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
113: erase write Flash verify error @0x71000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
114: erase write Flash verify error @0x72000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
115: erase write Flash verify error @0x73000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
116: erase write Flash verify error @0x74000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
117: erase write Flash verify error @0x75000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
118: erase write Flash verify error @0x76000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
119: erase write Flash verify error @0x77000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
120: erase write Flash verify error @0x78000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
121: erase write Flash verify error @0x79000, orig=0xffffffff, data=0x00000000, read=0x00000000
read done
[... 74 lines cut ...]
Flash verify error @0x6b105, orig=0x00, data=0x00000000, read=0x00000000
Flash verify error @0x6b106, orig=0x00, data=0x00000000, read=0x00000000
Flash verify error @0x6b107, orig=0x00, data=0x00000000, read=0x00000000
Flash verify error @0x6b108, orig=0x00, data=0x00000000, read=0x00000000
Flash verify error @0x6b109, orig=0x00, data=0x00000000, read=0x00000000
Flash verify error @0x6b10a, orig=0x00, data=0x00000000, read=0x00000000
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

ondabeach, 2018-01-07:
@igrr I have a second machine as a backup. I made everything the same as far as I can see, including the new Esp.cpp. The only difference I'm aware of is that on the original machine that gave the Flash Verify errors in the output The esp8266com/esp8266 folder was updated to v2.4.0 using GitGui whereas on this machine it was done through the Boards Manager and the URL in IDE preferences.

I can't write to SPIFFS via a sketch on either machine but the esp8266 Sketch Data Upload plugin can.

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 524288
Testing write...
Test data partition @ 0x6b000
dst=0 src=0 len=0
dst=0 src=0 len=4
dst=0 src=0 len=16
dst=0 src=0 len=64
dst=0 src=0 len=1
dst=0 src=1 len=1
dst=1 src=0 len=1
dst=1 src=1 len=1
dst=1 src=1 len=2
dst=1 src=1 len=3
dst=1 src=1 len=4
dst=1 src=1 len=5
dst=3 src=2 len=5
dst=4 src=4 len=60
dst=59 src=0 len=5
dst=60 src=0 len=4
dst=60 src=0 len=3
dst=60 src=0 len=2
dst=63 src=0 len=1
dst=64 src=0 len=0
dst=59 src=59 len=5
dst=60 src=60 len=4
dst=60 src=60 len=3
dst=60 src=60 len=2
dst=63 src=63 len=1
dst=64 src=64 len=0
Testing write/read...
Seed=0 Start=107 Count=16
107: erase write read done
108: erase write read done
109: erase write read done
110: erase write read done
111: erase write read done
112: erase write read done
113: erase write read done
114: erase write read done
115: erase write read done
116: erase write read done
117: erase write read done
118: erase write read done
119: erase write read done
120: erase write read done
121: erase write read done
122: erase write read done
Seed=1 Start=107 Count=16
107: erase write read done
108: erase write read done
109: erase write read done
110: erase write read done
111: erase write read done
112: erase write read done
113: erase write read done
114: erase write read done
115: erase write read done
116: erase write read done
117: erase write read done
118: erase write read done
119: erase write read done
120: erase write read done
121: erase write read done
122: erase write read done
Seed=42 Start=107 Count=16
107: erase write read done
108: erase write read done
109: erase write read done
110: erase write read done
111: erase write read done
112: erase write read done
113: erase write read done
114: erase write read done
115: erase write read done
116: erase write read done
117: erase write read done
118: erase write read done
119: erase write read done
120: erase write read done
121: erase write read done
122: erase write read done
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Mounting FS...
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
36: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

igrr, 2018-01-07:
Thanks @ondabeach , these logs with "flash verify error" mean that we are indeed seeing some hardware dependent issue.
I can't explain why you are not seeing these with the other machine, my only guess is that you are copying Esp.cpp file into one core installation, but the IDE is using some other one.

I have updated the test: https://gist.github.com/igrr/964b0d139ce63fcee07c8c1f1facb24d and [Esp.zip](https://github.com/esp8266/Arduino/files/1609994/Esp.zip), please give it another try. If the log ends up being long, could you please paste it into gist.github.com and post a link here?
Suggest doing this on the machine where you got `Flash verify error`s in the previous test.

ondabeach, 2018-01-08:
@igrr No longer than the others I think.

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 524288
Testing flash write bits...
Mask=0xffffffff
Mask=0xfffffffe
Mask=0xfffffffd
Mask=0xfffffffb
Mask=0xfffffff7
Mask=0xffffffef
Mask=0xffffffdf
Mask=0xffffffbf
Mask=0xffffff7f
Mask=0xfffffeff
Mask=0xfffffdff
Mask=0xfffffbff
Mask=0xfffff7ff
Mask=0xffffefff
Mask=0xffffdfff
Mask=0xffffbfff
Mask=0xffff7fff
Mask=0xfffeffff
Mask=0xfffdffff
Mask=0xfffbffff
Mask=0xfff7ffff
Mask=0xffefffff
Mask=0xffdfffff
Mask=0xffbfffff
Mask=0xff7fffff
Mask=0xfeffffff
Mask=0xfdffffff
Mask=0xfbffffff
Mask=0xf7ffffff
Mask=0xefffffff
Mask=0xdfffffff
Mask=0xbfffffff
Mask=0x7fffffff
Testing flash random write...
Seed=0 Start=107 Count=16
Seed=1 Start=107 Count=16
Seed=2 Start=107 Count=16
Seed=3 Start=107 Count=16
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Flash write verify error addr=0x6b0fc, orig=0x0000ffff, data=0xffff0429, read=0xfff70429
Flash write verify error addr=0x6c0fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x6d0fc, orig=0x0000ffff, data=0xffff0429, read=0xffdf0429
Flash write verify error addr=0x6e0fc, orig=0x0000ffff, data=0xffff0429, read=0xffef0429
Flash write verify error addr=0x6f0fc, orig=0x0000ffff, data=0xffff0429, read=0xefff0429
Flash write verify error addr=0x700fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x710fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x720fc, orig=0x0000ffff, data=0xffff0429, read=0x7fff0429
Flash write verify error addr=0x730fc, orig=0x0000ffff, data=0xffff0429, read=0x7fff0429
Flash write verify error addr=0x740fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x750fc, orig=0x0000ffff, data=0xffff0429, read=0xefdf0429
Flash write verify error addr=0x760fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x770fc, orig=0x0000ffff, data=0xffff0429, read=0xff770429
Flash write verify error addr=0x780fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x790fc, orig=0x0000ffff, data=0xffff0429, read=0xffd70429
Flash write verify error addr=0x7a0fc, orig=0x0000ffff, data=0xffff0429, read=0xbfff0429
Mounting FS...
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
Flash read verify (#0) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#1) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#2) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#0) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#1) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#2) error addr=0x710fc, data=0xffbf0429, check=0xffff0429
Flash read verify (#0) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
Flash read verify (#1) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
Flash read verify (#2) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
Flash read verify (#0) error addr=0x790fc, data=0xffd70429, check=0xffdf0429
Flash read verify (#2) error addr=0x790fc, data=0xffd70429, check=0xffdf0429
Flash read verify (#0) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
Flash read verify (#1) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
Flash read verify (#2) error addr=0x720fc, data=0x7eff0429, check=0x7fff0429
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
37: Failed to open file
appendFile
File size=0
f.write returned 7
Flash write verify error addr=0x6b000, orig=0xffff8001, data=0x0001ffff, read=0x0001f57f
Flash write verify error addr=0x6b204, orig=0xfffffffc, data=0x6f6f66ff, read=0x6f6f66ff
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

igrr, 2018-01-08:
Okay, so reads are the issue, not writes. Let's try to reduce flash frequency to 20MHz and see if that helps. In boards.txt, please add the following lines:
```
generic.menu.FlashFreq.20=20MHz
generic.menu.FlashFreq.20.build.flash_freq=20
```
next to
```
generic.menu.FlashFreq.40=40MHz
generic.menu.FlashFreq.40.build.flash_freq=40
```
Restart the IDE, select 20MHz, flash and run the same test.

ondabeach, 2018-01-08:
@igrr I'll do that now but my sketch can read but it can't write. If I manually create an end user's settings file and upload it to flash with the Sketch Data Upload, then the sketch can read the settings file but it can't write new settings. Output from 20mhz upload coming shortly.

ondabeach, 2018-01-08:
@igrr Ok, here's the output after flashing at 20Mhz

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   00146085
Flash size, calculated by ID: 1048576

Flash size, as set in IDE: 524288
Testing flash write bits...
Mask=0xffffffff
Mask=0xfffffffe
Mask=0xfffffffd
Mask=0xfffffffb
Mask=0xfffffff7
Mask=0xffffffef
Mask=0xffffffdf
Mask=0xffffffbf
Mask=0xffffff7f
Mask=0xfffffeff
Mask=0xfffffdff
Mask=0xfffffbff
Mask=0xfffff7ff
Mask=0xffffefff
Mask=0xffffdfff
Mask=0xffffbfff
Mask=0xffff7fff
Mask=0xfffeffff
Mask=0xfffdffff
Mask=0xfffbffff
Mask=0xfff7ffff
Mask=0xffefffff
Mask=0xffdfffff
Mask=0xffbfffff
Mask=0xff7fffff
Mask=0xfeffffff
Mask=0xfdffffff
Mask=0xfbffffff
Mask=0xf7ffffff
Mask=0xefffffff
Mask=0xdfffffff
Mask=0xbfffffff
Mask=0x7fffffff
Testing flash random write...
Seed=0 Start=107 Count=16
Seed=1 Start=107 Count=16
Seed=2 Start=107 Count=16
Seed=3 Start=107 Count=16
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Flash write verify error addr=0x6b0fc, orig=0x0000ffff, data=0xffff0429, read=0xffb70429
Flash write verify error addr=0x6c0fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x6d0fc, orig=0x0000ffff, data=0xffff0429, read=0xffdf0429
Flash write verify error addr=0x6e0fc, orig=0x0000ffff, data=0xffff0429, read=0xffef0429
Flash write verify error addr=0x6f0fc, orig=0x0000ffff, data=0xffff0429, read=0xefff0429
Flash write verify error addr=0x700fc, orig=0x0000ffff, data=0xffff0429, read=0xfffe0429
Flash write verify error addr=0x710fc, orig=0x0000ffff, data=0xffff0429, read=0xfbbf0429
Flash write verify error addr=0x720fc, orig=0x0000ffff, data=0xffff0429, read=0x7fff0429
Flash write verify error addr=0x730fc, orig=0x0000ffff, data=0xffff0429, read=0x7fff0429
Flash write verify error addr=0x740fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x750fc, orig=0x0000ffff, data=0xffff0429, read=0xefdf0429
Flash write verify error addr=0x760fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x770fc, orig=0x0000ffff, data=0xffff0429, read=0xff770429
Flash write verify error addr=0x780fc, orig=0x0000ffff, data=0xffff0429, read=0xffff0429
Flash write verify error addr=0x790fc, orig=0x0000ffff, data=0xffff0429, read=0xffdf0429
Flash write verify error addr=0x7a0fc, orig=0x0000ffff, data=0xffff0429, read=0xbfff0429
Mounting FS...
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
Flash read verify (#0) error addr=0x6b0fc, data=0xfff70429, check=0xffff0429
Flash read verify (#0) error addr=0x6b0fc, data=0xffff0429, check=0xfff70429
Flash read verify (#1) error addr=0x6b0fc, data=0xffff0429, check=0xffb70429
Flash read verify (#2) error addr=0x6b0fc, data=0xffff0429, check=0xfff70429
Flash read verify (#0) error addr=0x6c0fc, data=0xffff0429, check=0xdfff0429
Flash read verify (#1) error addr=0x6c0fc, data=0xffff0429, check=0xdfff0429
Flash read verify (#1) error addr=0x6c0fc, data=0xffff0429, check=0xdfff0429
Flash read verify (#0) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#2) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#0) error addr=0x700fc, data=0xfffe0429, check=0xffff0429
Flash read verify (#2) error addr=0x700fc, data=0xfffe0429, check=0xffff0429
Flash read verify (#0) error addr=0x710fc, data=0xfbbf0429, check=0xfbff0429
Flash read verify (#2) error addr=0x710fc, data=0xfbbf0429, check=0xfbff0429
Flash read verify (#0) error addr=0x710fc, data=0xfbbf0429, check=0xfbff0429
Flash read verify (#2) error addr=0x710fc, data=0xfbbf0429, check=0xfbff0429
Flash read verify (#0) error addr=0x790fc, data=0xffd70429, check=0xffdf0429
Flash read verify (#2) error addr=0x790fc, data=0xffd70429, check=0xffdf0429
Flash read verify (#1) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#2) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#0) error addr=0x710fc, data=0xfabf0429, check=0xfbbf0429
Flash read verify (#2) error addr=0x730fc, data=0xffff0429, check=0x7fff0429
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
Flash read verify (#1) error addr=0x730fc, data=0xffff0429, check=0x7fff0429
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
37: Failed to open file
appendFile
Flash read verify (#0) error addr=0x700fc, data=0xfffe0429, check=0xffff0429
Flash read verify (#0) error addr=0x710fc, data=0xfbbf0429, check=0xfabf0429
Flash read verify (#1) error addr=0x710fc, data=0xfbbf0429, check=0xfabf0429
Flash read verify (#2) error addr=0x710fc, data=0xfbbf0429, check=0xfabf0429
Flash read verify (#1) error addr=0x6c0fc, data=0xffff0429, check=0xdfff0429
Flash read verify (#2) error addr=0x6c0fc, data=0xffff0429, check=0xdfff0429
Flash read verify (#0) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#1) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#2) error addr=0x700fc, data=0xffff0429, check=0xfffe0429
Flash read verify (#0) error addr=0x710fc, data=0xfabf0429, check=0xfbbf0429
Flash read verify (#1) error addr=0x710fc, data=0xfabf0429, check=0xfbbf0429
Flash read verify (#0) error addr=0x730fc, data=0xffff0429, check=0x7fff0429
File size=0
f.write returned 7
Flash write verify error addr=0x6b000, orig=0xffff8001, data=0x0001ffff, read=0x0001e77f
Flash write verify error addr=0x6b204, orig=0xfffffffc, data=0x6f6f66ff, read=0x6f6f66ff
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
SPIFFS_read rc=-10012
File contents:

---
SPIFFS_close: fd=2

ondabeach, 2018-01-08:
@igrr I just received some 4Mbit flash chips and replaced the PUYA chip. One odd thing is the flash size of 2 based on chip ID but it looks like it's working. So that's positive proof that it's the PUYA chip that 8266 SPIFFS doesn't like. I'll load up my sketch on the other machine and let you know the result.

The chip I soldered on is this one from RS Components:
http://docs-asia.electrocomponents.com/webdocs/1328/0900766b8132801f.pdf

Here is the output:

phys_addr=438272
phys_size=65536
phys_page=256
phys_block=4096
Flash ID:   0001841F
Flash size, calculated by ID: 2

Flash size, as set in IDE: 524288
Testing flash write bits...
Mask=0xffffffff
Mask=0xfffffffe
Mask=0xfffffffd
Mask=0xfffffffb
Mask=0xfffffff7
Mask=0xffffffef
Mask=0xffffffdf
Mask=0xffffffbf
Mask=0xffffff7f
Mask=0xfffffeff
Mask=0xfffffdff
Mask=0xfffffbff
Mask=0xfffff7ff
Mask=0xffffefff
Mask=0xffffdfff
Mask=0xffffbfff
Mask=0xffff7fff
Mask=0xfffeffff
Mask=0xfffdffff
Mask=0xfffbffff
Mask=0xfff7ffff
Mask=0xffefffff
Mask=0xffdfffff
Mask=0xffbfffff
Mask=0xff7fffff
Mask=0xfeffffff
Mask=0xfdffffff
Mask=0xfbffffff
Mask=0xf7ffffff
Mask=0xefffffff
Mask=0xdfffffff
Mask=0xbfffffff
Mask=0x7fffffff
Testing flash random write...
Seed=0 Start=107 Count=16
Seed=1 Start=107 Count=16
Seed=2 Start=107 Count=16
Seed=3 Start=107 Count=16
Formatting FS...
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
Mounting FS...
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Testing read/write...
readFile
SPIFFSImpl::open: fd=-10002 path=`f.txt` openMode=0 accessMode=1 err=-10002
37: Failed to open file
appendFile
File size=0
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=7
File contents:
foobar

---
SPIFFS_close: fd=1
appendFile
File size=7
f.write returned 7
SPIFFS_close: fd=1
readFile
File size=14
File contents:
foobar
foobar

---
SPIFFS_close: fd=1

igrr, 2018-01-08:
Well, yes, it's quite certain that there's something interesting going on between the ESP and PUYA flash chip. "Flash read verify error" messages show that every now and then, one or two bits are read incorrectly (e.g. instead of 0xffff0429 it reads 0xdfff0429), which causes SPIFFS to fail during write operation.
Glad that you have a workaround for now. We have ordered a few of these PUYA chips, my colleagues will investigate this as soon as the chips arrive.

ondabeach, 2018-01-08:
@igrr Ok, the same module with the 4Mbit flash chip soldered on uploads and can write and read SPIFFS. It runs my sketch and the module functions perfectly.
I also just received the last batch of 20 modules I had on order and they all carry the PUYA chip. I now have about 95 esp-01's with this chip on-board. At least now I can replace the chips to keep filling orders, but I sure hope I don't have to swap them all out.
Thanks for trying Ivan, at least you've laid the groundwork for when your guys get some PUYA chips in-house. I'm sure then it will be much easier to sort out.
I look forward to some good news in the coming days :)

Cheers, Steve.

TD-er, 2018-01-15:
I was just looking into your testing/debugging and I was wondering if it is just a matter of writing 4 bytes to the flash to fix it?
(I hope I just did not miss the post where you already tried something like this....)

Just look at these code snippets from the ESP library:
https://github.com/esp8266/Arduino/blob/3b0c3958377d868c6374699c0e45903c559b152e/cores/esp8266/Esp.cpp#L202-L216

https://github.com/esp8266/Arduino/blob/3b0c3958377d868c6374699c0e45903c559b152e/cores/esp8266/Esp.cpp#L241-L260

If the 3rd byte is a 0, the flash size will be estimated as 512 k
But if you write the proper value to the 3rd byte, based on the result of `ESP.getFlashChipRealSize()` then it might start to work?
I think it is worth a try, but I don't have these chips here.

ondabeach, 2018-01-18:
@igrr have your colleagues had any luck with the PUYA chip?

ondabeach, 2018-01-22:
@igrr I know you're always busy working on multiple issues but can I get a little feedback on this issue please. As I warned, it's getting more widespread and it's only going to spread faster and faster and I fear that it will soon spread to other modules.

igrr, 2018-01-22:
Hi @ondabeach, sorry for the late reply. We've got some samples of this flash chip, and sadly, we don't have a solution yet. From what we see, the chip does not correctly read data in some cases, and does not correctly write in other cases. This is not specific to the ESP8266 — same issue can be reproduced with ESP32. My colleagues are still looking into this. I will update the issue when we find something.

igrr, 2018-01-22:
@TD-er We know that the pattern used by the flash tools does not cause problems:

1. erase sector
2. write whole sector

However, read-modify-write patterns used in SPIFFS do cause problems, sometimes.
For example, the following code snippet illustrates one of the failure modes. The code is for ESP32, but similar test can fail on the ESP8266:

```c++
    ESP_ERROR_CHECK(spi_flash_erase_range(0x90000, 0x1000));
    uint32_t flash_addr = 0x900fc;
    for (int bit = 0; bit < 32; ++bit) {
        uint32_t val_rd = 0, val_rd_2 = 0;
        uint32_t val_wr = ~BIT(bit);
        ESP_ERROR_CHECK(spi_flash_read(flash_addr, &val_rd, 4));
        printf("bit=%d val_rd=0x%08x val_wr=0x%08x\n", bit, val_rd, val_wr);
        ESP_ERROR_CHECK(spi_flash_write(flash_addr, &val_wr, 4));
        ESP_ERROR_CHECK(spi_flash_read(flash_addr, &val_rd_2, 4));
        assert(val_rd_2 == val_rd & val_wr);
    }
```

In this code snippet, we erase bits of a 32-bit word one by one. With this flash chips, one of the read-modify-write iteration fails with an assert: a bit which should be cleared is not cleared, or a bit which should not be cleared is.

Another (possibly related) failure mode is that inconsistent values are returned when reading same address multiple times. In some values, some of the bits which should be set are reported as cleared.

Regarding chip model number: it is P25Q80H, as reported by @ondabeach.

Regarding the datasheet: it used to be available from puyasemi.com, until the whole website went 404 somewhere during last week. Just one note about the datasheet: as one of my colleagues has noticed, it has unusually good English for a small Chinese chip company, and some sentences  can be found verbatim in Cypress S25FS512S datasheet... That alone gives a few hints.

Our next step would be to make the failing test 100% repeatable and check the waveforms on the scope to see if this is a timing issue of some sorts (e.g. the flash chip outputs the right bit too late for the ESP to sample it correctly) or if the flash chip just doesn't output correct signal level at all.

TD-er, 2018-01-22:
OK, that sounds quite serious indeed.

Since last week at ESPeasy, a PR was created to detect CRC errors in settings: https://github.com/letscontrolit/ESPEasy/pull/720
It was done by someone who had experienced flash errors apparently often enough to dive into this.
So perhaps timing was already quite critical, also for other flash chips?

Can you share the datasheets, if they are no longer accessible from the manufacturer's site?
Webarchive doesn't seem to have it and Google only has a cached version. (in html, not PDF)

ondabeach, 2018-01-22:
@igrr, @TD-er, here is the datasheet for the PUYA chip, I downloaded a couple of weeks ago when I first started digging into the problem.
[P25Q80H_FLASH CHIP.pdf](https://github.com/esp8266/Arduino/files/1651858/P25Q80H_FLASH.CHIP.pdf)

This is the datasheet for the chip I am using to replace the PUYA chip. It's only 512Kbyte, but that's all my sketch needs. FYI this chip is reported as having a size of zero bytes by ESP.getFlashChipRealSize() but it works just fine including writing and reading files in flash.
[AT25SF041.pdf]
(https://github.com/esp8266/Arduino/files/1651969/AT25SF041.pdf)

So, it looks like the whole PUYA company has gone down the toilet. I suspect they dumped these chips onto the market at a bargain price to try and get a bit of cash together as the ship sank. Who knows how many thousands of them are now floating around and how many module manufacturers have bought them. One thing I do know is that they will keep selling the affected modules even if they know they have a problem. So will the resellers that we buy them from. This means that we're going to be stuck with this issue for quite some time.

@igrr, I find it strange that you are getting random bit states when reading the same memory locations. When I put my user settings file into flash using the 'ESP8266 Sketch Data Upload', my sketch reads the data just fine.

I will check the reliability of reading from flash by doing the following:

I will upload my sketch in the usual way with the IDE 1.8.5 8266 core 2.3. 
Next I will upload the settings file using 'Sketch Data Upload'.

The first thing my sketch does in setup is SPIFFS.begin then it reads the data in the settings file into variables. To test, I will try continually rebooting the module and see if the variables are read reliably every time. I'll let you know the result.

igrr, 2018-01-22:
@ondabeach the tests you have done above (with modified Esp.cpp) have also indicated data read errors — these were logged as `Flash read verify (#0) error addr=0x6b0fc, data=0xfff70429, check=0xffff0429`. In this example, the address first reads `0xfff70429` and then `0xffff0429`, i.e. a single bit error.

igrr, 2018-01-22:
Read errors set aside for a while, the odd thing about this flash chip is that it behaves like some byte-erasable EEPROMs do, when we write data to it.

```
addr=0x00090000 bit=1 val_rd=0xfffffffe val_wr=0xfffffffd expected=0xfffffffc result=0xfffffffd
addr=0x00090000 bit=2 val_rd=0xfffffffd val_wr=0xfffffffb expected=0xfffffff9 result=0xfffffffb
addr=0x00090000 bit=3 val_rd=0xfffffffb val_wr=0xfffffff7 expected=0xfffffff3 result=0xfffffff7
addr=0x00090000 bit=4 val_rd=0xfffffff7 val_wr=0xffffffef expected=0xffffffe7 result=0xffffffef
addr=0x00090000 bit=5 val_rd=0xffffffef val_wr=0xffffffdf expected=0xffffffcf result=0xffffffdf
addr=0x00090000 bit=6 val_rd=0xffffffdf val_wr=0xffffffbf expected=0xffffff9f result=0xffffffbf
addr=0x00090000 bit=7 val_rd=0xffffffbf val_wr=0xffffff7f expected=0xffffff3f result=0xffffff7f
addr=0x00090000 bit=8 val_rd=0xffffff7f val_wr=0xfffffeff expected=0xfffffe7f result=0xfffffeff
addr=0x00090000 bit=9 val_rd=0xfffffeff val_wr=0xfffffdff expected=0xfffffcff result=0xfffffdff
addr=0x00090000 bit=10 val_rd=0xfffffdff val_wr=0xfffffbff expected=0xfffff9ff result=0xfffffbff
addr=0x00090000 bit=11 val_rd=0xfffffbff val_wr=0xfffff7ff expected=0xfffff3ff result=0xfffff7ff
addr=0x00090000 bit=12 val_rd=0xfffff7ff val_wr=0xffffefff expected=0xffffe7ff result=0xffffefff
addr=0x00090000 bit=13 val_rd=0xffffefff val_wr=0xffffdfff expected=0xffffcfff result=0xffffdfff
addr=0x00090000 bit=14 val_rd=0xffffdfff val_wr=0xffffbfff expected=0xffff9fff result=0xffffbfff
addr=0x00090000 bit=15 val_rd=0xffffbfff val_wr=0xffff7fff expected=0xffff3fff result=0xffff7fff
addr=0x00090000 bit=16 val_rd=0xffff7fff val_wr=0xfffeffff expected=0xfffe7fff result=0xfffeffff
addr=0x00090000 bit=17 val_rd=0xfffeffff val_wr=0xfffdffff expected=0xfffcffff result=0xfffdffff
addr=0x00090000 bit=18 val_rd=0xfffdffff val_wr=0xfffbffff expected=0xfff9ffff result=0xfffbffff
addr=0x00090000 bit=19 val_rd=0xfffbffff val_wr=0xfff7ffff expected=0xfff3ffff result=0xfff7ffff
addr=0x00090000 bit=20 val_rd=0xfff7ffff val_wr=0xffefffff expected=0xffe7ffff result=0xffefffff
addr=0x00090000 bit=21 val_rd=0xffefffff val_wr=0xffdfffff expected=0xffcfffff result=0xffdfffff
addr=0x00090000 bit=22 val_rd=0xffdfffff val_wr=0xffbfffff expected=0xff9fffff result=0xffbfffff
addr=0x00090000 bit=23 val_rd=0xffbfffff val_wr=0xff7fffff expected=0xff3fffff result=0xff7fffff
addr=0x00090000 bit=24 val_rd=0xff7fffff val_wr=0xfeffffff expected=0xfe7fffff result=0xfeffffff
addr=0x00090000 bit=25 val_rd=0xfeffffff val_wr=0xfdffffff expected=0xfcffffff result=0xfdffffff
addr=0x00090000 bit=26 val_rd=0xfdffffff val_wr=0xfbffffff expected=0xf9ffffff result=0xfbffffff
addr=0x00090000 bit=27 val_rd=0xfbffffff val_wr=0xf7ffffff expected=0xf3ffffff result=0xf7ffffff
addr=0x00090000 bit=28 val_rd=0xf7ffffff val_wr=0xefffffff expected=0xe7ffffff result=0xefffffff
addr=0x00090000 bit=29 val_rd=0xefffffff val_wr=0xdfffffff expected=0xcfffffff result=0xdfffffff
addr=0x00090000 bit=30 val_rd=0xdfffffff val_wr=0xbfffffff expected=0x9fffffff result=0xbfffffff
addr=0x00090000 bit=31 val_rd=0xbfffffff val_wr=0x7fffffff expected=0x3fffffff result=0x7fffffff
```

in this test, val_rd is the word read from flash before writing, val_wr is the word being written, expected is what you would normally expect to see in this word after writing with other flash chips (=val_rd & val_wr), and result is the word actually read back from flash.

As you see, it somehow sets the previously programmed bits back to 1, using just the program operation — without an erase. Which is really odd. We could work around this by reading data first, doing logical AND with the data we want to write, and writing the result back. That is, *if* we can get the reads to work reliably.

earlephilhower, 2018-01-22:
Is there an exact model number of the PUYA chip available?  I don't see it in this thread, other than possibly "P25Q80".  And, of course, http://puyasemi.com is dead.  

Google has a cached copy of the P25Q80H chip in HTML (so bye bye timing charts):
http://webcache.googleusercontent.com/search?q=cache:FUgOwHRjnqUJ:www.puyasemi.com/attached/file/20171025/20171025133034_11952.pdf+&cd=1&hl=en&ct=clnk&gl=us&client=ubuntu

The Page Program section may be of interest here, but I'm not familiar with SPIFFs internals enough to comment.

> 10.24 Page Program (PP)
> The Page Program (PP) instruction is for programming the memory to be "0". A Write Enable (WREN) instruction must execute to set the Write Enable Latch (WEL) bit before sending the Page Program (PP).  The device programs only the last 256 data bytes sent to the device. If the entire 256 data bytes are going to be programmed, A7-A0 (The eight least significant address bits) should be set to 0. If the eight least significant address bits (A7-A0) are not all 0, all transmitted data going beyond the end of the current page are programmed from the start address of the same page (from the address A7-A0 are all 0). If more than 256 bytes are sent to the device, the data of the last 256-byte is programmed at the request page and previous data will be disregarded. If less than 256 bytes are sent to the device, the data is programmed at the requested address of the page without effect on other address of the same page.

>The sequence of issuing PP instruction is: CS# goes low→ sending PP instruction code→ 3-byte address on SI→ at least 1-byte on data on SI→ CS# goes hi

ondabeach, 2018-01-22:
earlephilhower, look back at my next to last post. There is a link to the datasheet of the PUYA chip in question.

earlephilhower, 2018-01-22:
Sorry, my bad.  No reading these issues before my first coffee of the day!

s0170071, 2018-01-22:
@igrr is right, this behaviour is more like an EEProm, but with a memory size of a flash chip. It would appear they execute a full page erase/rewrite on every write access. That would also explain the early failure and essentially make that chip a WOM. 
https://en.wikipedia.org/wiki/Write-only_memory_(joke)

TD-er, 2018-01-22:
A page is like 256 bytes?
That could work, but then the reading and writing scheme should be changed.

For example:
Just write to the first available free page until it is full. Then erase the first few pages needed to store all variable data, copy them over to the first usable page and erase the rest.
Reading is then just looking at the (presumable) last used page +1 is that's erased, it was the correct location, else continue searching.

Drawback is that you must always have at least the largest data chunk in size available as free space.

devyte, 2018-01-22:
Is [this](https://github.com/pellepl/spiffs/issues/172) related?

TD-er, 2018-01-22:
@devyte That sounds like a plausible explanation.

devyte, 2018-01-22:
To test, someone with the chip would have to pull a newer version of spiffs, and then enable SPIFFS_NO_BLIND_WRITES in spiffs_config.h, then rebuild.

ondabeach, 2018-01-22:
@devyte I have almost 80 esp-01 modules left with the PUYA chip on-board, since I probably have the most to gain from a fix I'll put my hand up to do whatever it takes.

Just let me know what you want me to do and where to get the files.

ondabeach, 2018-01-23:
@igrr @devyte @TD-er 

Ok, the problem is definitely how SPIFFS writes to flash.

This is the contents of the settings file my sketch stores in flash:
YakkerFlash
12345678
YourNetwork
YourPassword
19200
2000
23
1
0
2

If I upload the file using Sketch Data Upload my sketch reads the file perfectly EVERY time. I just rebooted the esp-01 about 30 times and the data was read accurately every single time.

BUT, as soon as I try to change a line in the file with the end user html interface,which rewrites the whole file one line at a time using:
  File f = SPIFFS.open("/login.txt", "w");
  f.println(APssid);
  f.println(APpwd);
  f.println(STAssid);
  f.println(STApwd)........

  When it reboots the file can not be read using:
  File f = SPIFFS.open("/login.txt", "r");
  if (!f) { 

Don't know if this needs to be taken into account, it's from the PUYA datasheet:

DP bit. The Dual Page (DP) bit is a non-volatile Read/Write bit in the Configure Register that allows Dual Page operation. When the DP bit is set to 0 (Default) the page size is 256bytes. When the DP pin is set to 1, the page size is 512bytes.   
This bit controls the page programming buffer address wrap point. Legacy SPI devices generally have used a 256 Byte page programming buffer and defined that if data is loaded into the buffer beyond the 255 Byte location, the address at which additional bytes are loaded would be wrapped to address zero of the buffer. The P25Q80H provides a 512 Byte page programming buffer that can increase programming performance. For legacy software compatibility, this configuration bit provides the option to continue the wrapping behavior at the 256 Byte boundary or to enable full use of the available 512 Byte buffer by not wrapping the load address at the 256 Byte boundary.  
When the DP pin is set to 1, the page erase instruction (81h) will erase the data of the chosen Dual Page to be "1"

igrr, 2018-01-23:
My tentative plan for working around read and write issues was to introduce 'quirk' flags, which would be set at startup depending on flash vendor/model id. Then at run time, customize flash read and write operations based on these quirks. For write, we would do read-mask-write instead of just write, and for read we would read multiple times until we get the same value multiple times. (That is, if we don't figure out which timing adjustment we need to make to fix erroneous reads).

In the meantime, SPIFFS_NO_BLIND_WRITES is worth trying. It only fixes the issue for SPIFFS, but that might be sufficient, as the first step.

ondabeach, 2018-01-23:
@igrr I'll keep my fingers crossed for you, the 'quirk' flag approach will be a pain in the butt for you guys, but what can you do when manufacturers can't standardize.

I'll happily give the SPIFFS_NO_BLIND_WRITES a try which I can do immediately, but I will need instructions on where to download and how to rebuild (or a link to a how to) as I've never done anything like that before. I've never gone deeper than the IDE when it comes to anything Arduino.

devyte, 2018-01-23:
@igrr That's sort of what I was getting at. The solution in the SPIFFS repo provides a fix for SPIFFS, which fixes the most urgent issue, but it's not a complete solution. The issue likely affects the entire flash IO layer, so a complete solution for us would have to be implemented at that level. What the fix in the SPIFFS repo does, if it pans out, is show the way.
If the plan is to set a flag based on vendor/model id, then I suggest a solution that integrates both this R/W mode flag as well as the flash size. I don't mean necessarily a single table for everything, but just put everything in one place.

TD-er, 2018-01-23:
I think I could live with a compile-time flag to switch functionality. For ESPeasy it would mean a separate build task to deliver a separate artifact for download.
Trial-and-error attempts to detect the flash type sounds like something tricky and hard to maintain.
Also a list of brands/models is probably impossible to maintain and are those values guaranteed correct? (fake chips)
Only if we can detect/query the supported flash features of the chip, a runtime switch would be the preferred solution. 

All based on the idea this is actually the solution, but it sounds all very plausible. So let's assume you guys already found the solution ;)

s0170071, 2018-01-23:
@ondabeach, can you try to give the chip some time between the writes ? And check the operating voltage at the memory using a scope ? 
Those flash chips have an internal charge pump that provides the programming voltage. If it is too weak that may explain why it works during upload and fails on other occasions.  
If a write is exexuted with a low programming voltage, it actually is a bad write.  The written byte may have a bad read margin and thus fail from time to time or even degrade. 

Giving a write some more time (more that in the spec and more than is signalled by SPI) may support that charge pump.

ondabeach, 2018-01-23:
Do you mean simply use something like this:

File f = SPIFFS.open("/login.txt", "w");
Delay(50);
f.println(APssid);
Delay(50);
f.println(APpwd);
Delay(50);
f.println(STAssid);
Delay(50);
f.println(STApwd)........

I'll give it a try and let you know. 
Ive only got a cheap little pocket sized digital scope but I'll see if is good enough to get a decent reading.

s0170071, 2018-01-23:
Exactly, thats what I mean. If that does not work maybe even on a byte-by-byte basis.
If your scope does not read any voltage glitches, you can also try and just add a small capacitor (e.g. ceramic, 100nF) across the voltage supply pins of the flash.

TD-er, 2018-01-23:
Keep in mind, delay() may trigger other actions than expected.
It is also an indication to do some background tasks.

ondabeach, 2018-01-23:
Yes, I tried both delay() and delayMicroseconds(), with delays ranging from 200us to 200ms and it made no difference.

It's way past my bedtime so I'll dig the scope out in the morning and check the volts during writes.

[94 more replies]

closed 2018-12-14 by devyte