esp8266/Arduino #4063: twi_status clockCount  wrong usage ??

miky2k, 2018-01-01:
while (SDA_READ()==0 && clockCount>0){ 

should be

while (SDA_READ()==0 && clockCount-->0){ 

i feel that clockCount should be 10 not 20 (eight bit + ack + stop)

devyte, 2018-01-08:
@miky2k When you opened the issue, you were presented with an issue template requesting specific information. You have completely ignored that. Please edit your post and add the required info.
Also, can you please explain better what you mean?

miky2k, 2018-01-08:
clockCount is initialized , tested , but never changed (so why test it?),
I'm sure that, if this is true , the board,the compiler and other stuff are irrelevant, it's only c/c++ programming issue.

twi_status is helpful to unlock i2c after random reset if stolen by slave for what i understand.

5chufti, 2018-01-09:
maybe devs would be more willing to looking into this if you'd provide more reference like e.g. filename and line number ?

devyte, 2018-01-10:
After investigating the universe, I managed to descipher that:
- this issue refers to I2C [here](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/core_esp8266_si2c.c#L205) 
- the comment does seem to make sense: clockCount seems to be some kind of timeout, and in that case decrementing it is likely correct
- There is a relevant PR: #4070 

The question that remains is whether clock_count should start at 20 like now (the PR doesn't change this value), or at 10 as proposed here.

PR #5292 (2018-10-29): Add missing decrement operator to I2C clockCount
Fixes #4063 .
I am not changing the clockCount init value of 20 to 10 because I don't know the impact. Given that it's essentially a timeout, it shouldn't be relevant.
closed 2018-10-29 by devyte