esp8266/Arduino #4067: Wifi AP SSID Doc Issue

timpur, 2018-01-01:
In the wifi AP doc there seems to be an error.

https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html#softap

> ssid - character string containing network SSID (max. 63 characters)

Isnt the mac 32 char ? Think this is meant for the password?

gicho, 2018-01-01:
> esp8266wifi/soft-access-point-class.html#softap

SSID is different to MAC address - here ssid variable means pointer to
string (text) that is the network name. This name (ssid) is broadcasted by
the AP and you see it as network name on your device (smartphone, PC,
etc.). The MAC address is not designed to be randomly defined by users - it
is allocated by HW manufacturers (and sometimes overriden by developers).
But certified devices need to have a constant, pre-defined one.
For example:
const char ssid[] = "MyNewHotspot";

timpur, 2018-01-01:
@gicho I didn't mention mac?

gicho, 2018-01-01:
"Isnt the mac 32 char ? Think this is meant for the password?"

timpur, 2018-01-01:
My bad meant to be "max", sorry.

earlephilhower, 2018-11-08:
Fixed via #5325

closed 2018-11-08 by earlephilhower