esp8266/Arduino #4071: ESP8266 doesnt wake up after put to deepsleep.

grek22, 2018-01-02:
Hardware: wemos d1 r1
SDK: v2.1.0-10-g509eae8

.ino:

#include <OneWire.h>
#include <DallasTemperature.h>
#include <ESP8266WiFi.h>

const char *ssid = "";
const char *password = "";
WiFiClient *client = new WiFiClient();
IPAddress clientIP = IPAddress(9xxx);

OneWire oneWire(5); 
DallasTemperature sensors(&oneWire); 
float temp1;
unsigned czas;
void work();

void setup(void) {
	Serial.begin(115200);
	while (!Serial)
		delay(10);
	czas = millis();
	WiFi.config(IPAddress(192, 168, 8, 110), IPAddress(192, 168, 8, 1),IPAddress(255,255,255,0));
	WiFi.begin(ssid, password);
	sensors.setResolution(9);
	sensors.begin(); 
	while (WiFi.status() != WL_CONNECTED) {
		if((millis()-czas)>10000)
			ESP.deepSleep(60e6, WAKE_RF_DEFAULT);
		delay(100);
	}
	work();
}

void loop(void) {
	delay(10);
}

void work() {
	do {
	sensors.requestTemperatures();
	temperatura1 = sensors.getTempCByIndex(0);
	} while (temperatura1 == 85 || temperatura1 == -127);
	sensors.requestTemperatures();
	delay(500);
	temp1 = sensors.getTempCByIndex(0);
	String url = "example"; 
	client->connect(clientIP, 80);
	client->println(url)
	client->flush();
	client->stop();
	delete client;
	ESP.deepSleep(10e6, WAKE_RF_DEFAULT);
}

Problem: 
After a random time ESP doesnt wake up. [zombie mode on GPIO-0]

Conection: 
EN -> GPIO16

5chufti, 2018-01-28:
you're sure it is zombie mode and not wdt because it got stuck in 
"while (temperatura1 == 85 || temperatura1 == -127);" because of flakey sensor?

devyte, 2020-03-31:
Does this:
https://github.com/esp8266/Arduino/issues/6007#issuecomment-606414184
help at all? (47K pullup on MISO)

sjirka, 2020-05-04:
47K pullup on MISO solved this issue for me on ESP12F

kebwi, 2020-05-19:
I've been dealing with the zombie deep sleep wake up on my D1 mini clones. I went through that other thread in detail. I don't have any 47k resistors, but I put a 10K on MISO (pin D6 on the D1 -- this terminology is horrendous), but I still get the same failed wake up.

devyte, 2020-10-09:
Does this work to wake up the ESP:
https://github.com/esp8266/Arduino/issues/6007#issuecomment-542192213
?