esp8266/Arduino #4073: Soft WDT reset NodeMCU with HX7111.h and pubsubclient

SKRahulSK, 2018-01-02:
### Basic Infos

#### Hardware
Hardware:			NodeMCU Lolin V3 ESP-12E

### Description

I am getting "Soft WDT reset" when I used both HX711.h and pubsubclient.h
My code is working when I use these libraries separately. That is I can measure weight using HX711 and connect to MQTT broker using two different sketches. 
But getting following error when I use them both.

Connecting to TP-LINK_FFAA
.
WiFi connected
IP address: 
192.168.0.103
Attempting MQTT connection...
Soft WDT reset

ctx: cont 
sp: 3ffefe70 end: 3fff00f0 offset: 01b0

>>>stack>>>
3fff0020:  3ffe897c 00000f48 3ffeed68 40202ceb  
3fff0030:  00000000 3ffeed64 3ffeed68 402038f0  
3fff0040:  514d0400 3f045454 3fff00ac 4020423f  
3fff0050:  00000000 3fff10cc 00000000 4020428b  
3fff0060:  3ffef074 00000154 00000154 3ffef0c0  
3fff0070:  3fffdad0 3ffef094 3ffeed64 40203974  
3fff0080:  00000000 00000000 3ffeed64 402041e4  
3fff0090:  3ffe84e4 3ffef094 3ffeed64 4020232b  
3fff00a0:  00000000 00000000 00000000 3fff10cc  
3fff00b0:  0000001f 00000012 3ffeee38 40204824  
3fff00c0:  4020217c 3ffeed64 3ffef0b9 40202472  
3fff00d0:  3fffdad0 00000000 3ffef0b9 402044fc  
3fff00e0:  feefeffe feefeffe 3ffef0d0 40100710  
<<<stack<<<
H!⸮⸮?⸮F

### Settings in IDE

Module:  			NodeMCU1.0(ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

#include <HX711.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define calibration_factor1 -409200 //This value is obtained using the Calibration sketch
#define calibration_factor2 -442400 //This value is obtained using the Calibration sketch

#define DOUT1  D1
#define CLK1  D2

#define DOUT2 D5
#define CLK2 D6

float weight1, weight2, total_weight;

HX711 scale1(DOUT1,CLK1);
HX711 scale2(DOUT2,CLK2);

#define BUILTIN_LED D4

const char* ssid = ".....";
const char* password = ".....";
const char* mqtt_server = ".....";

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    delay(50);
    Serial.print("Attempting MQTT connection...");
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str())) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic_Try/", "hello world");
      // ... and resubscribe
      client.subscribe("InTopic_Try/");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  //pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(9600);
  delay(50);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  scale1.set_scale(calibration_factor1); //This value is obtained by using the Calibration sketch
  scale1.tare();  //Assuming there is no weight on the scale at start up, reset the scale to 0

  scale2.set_scale(calibration_factor2);
  scale2.tare();

}

void loop() {
  
  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  weight1=scale1.get_units(5);
  weight2=scale2.get_units(5);
  total_weight=weight1+weight2;
    if (now - lastMsg > 5000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "Total weight #%ld", total_weight);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic_Try/", msg);
  }
}

### Debug Messages

```
messages here
Stack Error Decoded:
Decoding 12 results
0x40202ceb: WiFiClient::available() at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 329
0x402038f0: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at C:\Users\Rahul\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x4020423f: String::changeBuffer(unsigned int) at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 46
0x4020428b: String::reserve(unsigned int) at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 46
0x40203974: PubSubClient::connect(char const*) at C:\Users\Rahul\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x402041e4: String::~String() at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 46
0x4020232b: reconnect() at C:\Users\Rahul\Documents\Arduino\mqtt_weigh_scale/mqtt_weigh_scale.ino line 85
0x40204824: PubSubClient::connected() at C:\Users\Rahul\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x4020217c: callback(char*, unsigned char*, unsigned int) at C:\Users\Rahul\Documents\Arduino\mqtt_weigh_scale/mqtt_weigh_scale.ino line 56
0x40202472: loop at C:\Users\Rahul\Documents\Arduino\mqtt_weigh_scale/mqtt_weigh_scale.ino line 122
0x402044fc: loop_wrapper at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\Rahul\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

closed 2018-01-03 by SKRahulSK

sniperlucian, 2018-11-04:
came across the same problem. 

you found any solution for this?

neironstudio, 2019-05-20:
I'm localize this problem - problem in client.setCallback(), if not connect to mqtt  broker callback call wdt ....

roncoa, 2019-11-13:
I have the same problem, esp crashes after client.setCallback(callback);
Have you solved the problem?