esp8266/Arduino #4074: arduino-esp8266.upload.resetmethod=ck missing in boards.txt

jandrassy, 2018-01-02:
2.4 release
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to {upload.resetmethod}
error: unknown board: {upload.resetmethod}
known boards are: none ck wifio nodemcu

d-a-v, 2018-01-02:
True. 
This is also true for gen4iod oak wifinfo.
Fix is coming.
For the record current boards.txt is generated from #3722.

d-a-v, 2018-01-03:
@jandrassy please take and check the file from #3722 (original post, link "latest") until a solution is found to fix current release.

jandrassy, 2018-01-03:
of course I added the line into my installation and I modified the [instruction in Uno WiFi Wiki](https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki/Programming-ESP8266) yesterday.

I googled the 160 MHz and I see it is set by software. If I select 160 MHz in IDE will the arduino 8266 core set the frequency?

d-a-v, 2018-01-03:
> If I select 160 MHz in IDE will the arduino 8266 core set the frequency?

yes

d-a-v, 2018-01-08:
Fixed by #3722.

closed 2018-01-08 by d-a-v