esp8266/Arduino #4077: Cannot Build with Wire.h on Arduino

eggsactly, 2018-01-03:
### Basic Infos

#### Hardware
Hardware:			?Arduino?
Core Version:      	?2.4.0?

### Description

I cannot build a sketch that includes both the Wire.h and EP8266WiFi.h libraries. I am trying to build for an Arduino Uno Wifi. Using Arduino 1.8.5 on Debian 9.3. 

### Settings in IDE

Module:  			?Arduino?
Flash Size: 		?4MB/(1MB SPIFFS)?
CPU Frequency:		?80Mhz?
Flash Mode:			?N/A?
Flash Frequency:	?115200?
Upload Using:		?SERIAL?
Reset Method:		?N/A?

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <Wire.h>

void setup() {
  // Do Nothing
}
```

### Debug Messages

```
/home/$USER/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/Wire/Wire.cpp:48:30: error: 'SDA' was not declared in this scope
 static int default_sda_pin = SDA;
                              ^
/home/$USER/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/Wire/Wire.cpp:49:30: error: 'SCL' was not declared in this scope
 static int default_scl_pin = SCL;
                              ^
exit status 1
Error compiling for board Arduino.
```

pieman64, 2018-01-03:
@eggsactly you are using an incomplete sketch, see http://arduino-esp8266.readthedocs.io/en/2.4.0/libraries.html#i2c-wire-library

igrr, 2018-01-03:
Which board do you have selected?

eggsactly, 2018-01-04:
@pieman64, I'm assuming you mean that I should call Wire.begin(0, 2). Regardless, adding it produces the same result. I don't believe it is necessary to add that begin call because the SDA and SCL lines should default to pins 4 and 5 respectively. 

Modifying  esp8266/2.4.0/libraries/Wire/Wire.cpp to use 
```
static int default_sda_pin = 4;
static int default_scl_pin = 5;
```

Instead causes this code to compile (I modified the following code below from my code in the first post because it won't work on the generic esp8266 board, but this will because it has the loop() function). 

```
#include <ESP8266WiFi.h>
#include <Wire.h>

void setup() {
  // Do nothing 
}

void loop(){
  //Do Nothing
}
```
It appears these values are defined in esp8266/2.4.0/variants/generic/pins_arduino.h, but I don't know enough about how your project is configured to try to include it. 

@igrr To try to answer your question I have selected the Arduino under ESP8266 Modules, and selected Uno WiFi as the model.

igrr, 2018-01-04:
Thanks @eggsactly, this looks like an issue with Uno WiFi board variant.

Other board variant header files define SDA and SCL constants for the default I2C pin names, e.g.:

https://github.com/esp8266/Arduino/blob/d9ef6b5f183d34ab2d471ee8573de1bf0b5a70a6/variants/generic/pins_arduino.h#L31-L35

These definitions are missing from the variant header file for Uno WiFi:

https://github.com/esp8266/Arduino/blob/master/variants/arduino_uart/pins_arduino.h

WereCatf, 2018-01-05:
@igrr I would assume that the reason those pins aren't defined is because all the GPIO-pins except GPIO5 are already in use for something else on the board, at least according to [https://eu.mouser.com/pdfdocs/Arduino-UNO-WiFi-V4_AF1.pdf](https://eu.mouser.com/pdfdocs/Arduino-UNO-WiFi-V4_AF1.pdf) Given that they're already in use, I don't think it's a good idea to define any I2C-pins in the first place.

WereCatf, 2018-01-05:
Couldn't find a schematic for Star OTTO, so no idea if the pins are in use there or not.

igrr, 2018-01-05:
Alternatively, we could have something like this in Wire.cpp:

```c++
#if !defined(PIN_WIRE_SDA) || !defined(PIN_WIRE_SCL)
#error Wire library is not supported on this board
#endif
```

Not sure whether this is what @eggsactly wants...

jandrassy, 2018-01-07:
@eggsactly, on [Uno WiFi](https://github.com/jandrassy/UnoWiFiDevEdSerial1/wiki) the esp8266 has no pin headers. why do you try to use the Wire library?

jandrassy, 2018-01-07:
@WereCatf, not all pins are used. GPIO0 is connected to enable pulling low for bootloader mode, GPIO2 is connected, only to a soldering point, 4 controls the direct serial connection for Atmega flashing from esp, 5 is not connected, 12 is to Atmega reset pin, 13 is not connected, 14 is LED, 15 is not connected only pulled down, ,

eggsactly, 2018-01-08:
@WereCatf and @jandrassy perhaps I'm not understanding something, but it appears that on the Uno and Uno Wifi, the I2C pins are located on the same pins, Pins 27 and 28 on ZU4 on both devices. 
[Uno Rev 3 Diagram](https://www.arduino.cc/en/uploads/Main/Arduino_Uno_Rev3-schematic.pdf)
[Uno Wifi Diagram](https://eu.mouser.com/pdfdocs/Arduino-UNO-WiFi-V4_AF1.pdf)

I believe I should be able to attach an I2C device to the A4 and A5 pins. I was able to compile the code below using the vanilla Arduino AVR Board "Arduino Uno WiFi", but the same code does not compile with the ESP8266 Module "Arduino". This code is different from above because it does not include the ESP8266WiFi.h library so it should compile regardless of which device I have selected. 
```
#include <Wire.h>

void setup() {
  // Do nothing 
}

void loop(){
  //Do Nothing
}
```

My perception as a user is this code should compile for both the Arduino AVR Board "Arduino Uno WiFi" and the ESP8266 Module "Arduino". My perception is based on the assumption that the normal Arduino library for Wire is the same as the library for ESP8266. 

Perhaps I made a mistake in assuming that pins 4 and 5 correspond to A4 and A5, perhaps these pins are different, I am unsure because the Arduino's API for Wire does not allow you to choose output pins, you're glued on A4 and A5.

@jandrassy to answer your question about why I would like to use the Wire library, I would like to hook this shield https://www.adafruit.com/product/772 up to the Arduino Wifi, while also being able to get wifi BSSIDs without being connected to a network, demonstrated on [this project](https://github.com/reynico/arduino-wardriving). It appears that the Arduino WiFi library does not have this capability but ESP8266WiFi does.

WereCatf, 2018-01-08:
@eggsactly Yes, A4 and A5 are brought out, but A4 and A5 are the Atmega328P-pins, not ESP8266-pins.  The ESP8266 and the Atmega328P are separate microcontrollers, you can't program the Atmega's I2C-pins with this ESP8266-core.

Also, the pins on the ESP8266 seemingly haven't been brought out on any pin-headers, so you couldn't use them anyways, unless you actually went and modified the board itself.

jandrassy, 2018-01-08:
@eggsactly, you must write two sketches and program the communication between them. one for Armega and one for esp8266. if you need guidance with it, write a post on[ Arduino forum](https://forum.arduino.cc/)

jandrassy, 2018-01-08:
@igrr, this is not a bug and not a 'good first', but the the #define check would help

devyte, 2018-01-30:
Closed via #4261

closed 2018-01-30 by devyte

Bagunda, 2018-08-18:
I change place 'AppData' and etc.
Then i have error:
\libraries\Wire\Wire.cpp:37:2: error: #error Wire library is not supported on this board
 #error Wire library is not supported on this board

I remove Arduino IDE. Then install Arduoino IDE from Microsoft Store.
Ver 1.8.5.
Error stil appears.

Sketch:
```
#include <Wire.h>
#include <ESP8266WiFi.h>

#define pin_sda 5 // default sda pin
#define pin_scl 4 // default scl pin

void setup() {
  //
}

void loop(){
  //
}
```

How to fix?

d-a-v, 2018-08-18:
This looks like an installation problem.
Move every arduino installation directories in a backup directory and reinstall everything from scratch.
Do not use closed issues, please open a new one with your data in the issue template.

jandrassy, 2018-08-18:
@Bagunda, esp8266 core version you use?