esp8266/Arduino #4080: 2.4.0 lwip compile error

s0170071, 2018-01-03:
### Basic Infos
I read in the 2.4.0 release notes that lwip 2 is now the state of the art lwip. so I gave it a try and compiled. Originally I compiled the ESPEasy project, but it failed when including the "lwip/tcp_impl.h" See sketch below.

#### Hardware
Hardware:			Wemos D1 mini
Core Version:      	2.4.0

### Description

Problem description
I use the Arduino IDE 1.8.5 as portable install. 
Boards manager URL is 
```
https://github.com/esp8266/Arduino/releases/download/2.4.0/package_esp8266com_index.json
```

When I select the lwip version from the drop down menu, only 1.4 prebulit compiles. 

### Settings in IDE

Module:  			wemos d1 mini
Flash Size: 		4MB
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include "lwip/tcp_impl.h"
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### v2 prebuilt mss636 and mss 1460
```
C:\ArduinoPortable\sketchbooks\_includelwip_tcp_implH\_includelwip_tcp_implH.ino:1:27: fatal error: lwip/tcp_impl.h: No such file or directory

 #include "lwip/tcp_impl.h"

                           ^

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

### 1.4 prebuilt 
(compiles ok)

### lwip1.4 (open source)
```
Compiling sketch...
make -C "C:\ArduinoPortable\arduino-1.8.5 - 2.4.0\portable\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/src" install TOOLS_PATH="C:\ArduinoPortable\arduino-1.8.5 - 2.4.0\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-"
exec: "make": executable file not found in %PATH%
Error compiling for board WeMos D1 R2 & mini.
```

d-a-v, 2018-01-03:
I'm afraid tcp-impl.h not the state of the art lwip.
It is part of lwip1.x only.
I'll have a look to this project.
What you can do is comment the #include, recompile and grep for the missing declaration in tools/sdk/lwip2/

s0170071, 2018-01-03:
tried that, no luck so far. For the moment I am searching a replacement for 

> tcp_tw_pcbs

The only occurence is in "sdk\lwip2\include\lwip\priv\tcp_priv.h" but swapping tcp-impl.h for this .h yields a couple of redefinition errors. So this seems not to be a quick fix.

s0170071, 2018-01-03:
just out of interest... what does @igrr mean by "Use lwip v2 by default" in the release notes.

devyte, 2018-01-03:
@s0170071 as of v2.4.0, building a sketch links against lwip2 by default. Up to v2.3.0 linking was against lwip1.4.
The newer lwip solves more issues than it creates, so the decision was made to change the default to lwip2 now, even though it is technically still experimental.
This issue likely stems from ESPEasy using lwip internals for something. They will have to update their implementation to account for the new lwip2.
Closing as the change of lwip is intentional, and therefore this is an issue in the 3rd party lib. I suggest opening an issue against ESPEasy requesting they update.

closed 2018-01-03 by devyte

d-a-v, 2018-01-03:
Here is a what should be done for this first error.
Feel free to address your lwip2 specific questions to the lwip2 repo (check tools/sdk/lwip2/README).
Like @devyte said, ESPEasy is not using pure esp8266/arduino library. About other compilation problems, an ESPEasy developper will be able to solve them.

```
diff --git a/src/ESPEasy.ino b/src/ESPEasy.ino
index 175de19..27842c1 100644
--- a/src/ESPEasy.ino
+++ b/src/ESPEasy.ino
@@ -314,7 +314,15 @@
   #define FILE_SECURITY     "security.dat"
   #define FILE_NOTIFICATION "notification.dat"
   #define FILE_RULES        "rules1.dat"
-  #include "lwip/tcp_impl.h"
+  #include <lwip/init.h>
+  #ifndef LWIP_VERSION_MAJOR
+  #error
+  #endif
+  #if LWIP_VERSION_MAJOR == 2
+  #include <lwip/priv/tcp_priv.h>
+  #else
+  #include <lwip/tcp_impl.h>
+  #endif
   #include <ESP8266WiFi.h>
   #include <ESP8266WebServer.h>
   ESP8266WebServer WebServer(80);
```

Krish402, 2021-07-01:
i'm also getting this exec: "make": executable file not found in %PATH% error in windows