esp8266/Arduino #4081: ArduinoOTA not working any more

pkishino, 2018-01-03:
### Basic Infos

#### Hardware
Hardware:			nodemcu 1.0
Core Version:      	

### Description

Problem description
Previously I was able to successfully run the OTA but today I was reworking my board and putting in proper cabling etc for the project as the prototype had worked as wanted.
First I had some issues with the nodemcu not working as expected so I did some troubleshooting and finally got it back to OK status, but when I run the OTA it goes all the way to ArduinoOTA.onEnd and I print Restarting on my screen..after this it doesnt restart as it used to. When I check serial log I can see the following reset crashing over and over until I hit reset button:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
@cp:0
ld

Soft WDT reset

ctx: cont 
sp: 3ffefd20 end: 3ffeffa0 offset: 01b0

>>>stack>>>
Decoding 9 results
0x401064a9: __udivmoddi4 at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/src/gcc-4.8.2/libgcc/libgcc2.c line 965
:  (inlined by) __divdi3 at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/src/gcc-4.8.2/libgcc/libgcc2.c line 1131
0x402062c0: ets_vsprintf at ?? line ?
0x40202fc2: system_upgrade_userbin_check at ?? line ?
0x40202fb7: system_upgrade_userbin_check at ?? line ?
0x40206435: ets_snprintf at ?? line ?
0x40206529: eagle_lwip_if_alloc at ?? line ?
0x4020b3f0: tcp_connect at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp.c line 696
0x40202a5c: system_get_os_print at ?? line ?
0x40203c1c: wifi_set_broadcast_if at ?? line ? 
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

Soft WDT reset

ctx: cont 
sp: 3ffefd30 end: 3ffeffa0 offset: 01b0

>>>stack>>>
Decoding 8 results
0x402062aa: ets_vsnprintf at ?? line ?
0x40202fc2: system_upgrade_userbin_check at ?? line ?
0x40202fb7: system_upgrade_userbin_check at ?? line ?
0x40206435: ets_snprintf at ?? line ?
0x40206529: eagle_lwip_if_alloc at ?? line ?
0x4020b3f0: tcp_connect at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp.c line 696
0x40202a5c: system_get_os_print at ?? line ?
0x40203c1c: wifi_set_broadcast_if at ?? line ?
<<<stack<<<
```

pkishino, 2018-01-03:
I have nothing else connected to the board currently

devyte, 2018-01-03:
@pkishino I assume you are doing a serial upload, and then an ArduinoOTA. After the serial upload, are you doing a physical reset by pressing the button or doing a power cycle?
Which core version? Please retest with latest git or 2.4.0, there was a fix merged just before 2.4.0 release.

pkishino, 2018-01-03:
No, not doing a serial upload first. I did see the thread about that and
tried that to but same result

pkishino, 2018-01-03:
I will check if I have latest code

pkishino, 2018-01-03:
this is the code which was working fine last week:
```
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include "MFRC522.h"
#include <Wire.h>
#include "SSD1306.h"

#define RST_PIN 15                // RST-PIN for RC522
#define SS_PIN 2                  // SDA-PIN for RC522
MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance

//Pins screen
SSD1306 display(0x3c, D1, D2);

String server = "XXXX";
int port = 3000;
const char *ssid = "XXXX";
const char *password = "XXXX";
byte nuidPICC[4];
void setup()
{
  Serial.begin(115200);
  SPI.begin();        // Init SPI bus
  mfrc522.PCD_Init(); // Init MFRC522
  delay(10);

  display.init();
  display.flipScreenVertically();
  display.setContrast(255);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    display.drawString(5, 0, "Connection Failed! Rebooting...");
    display.display();
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname("myesp8266");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    display.clear();
    display.setFont(ArialMT_Plain_10);
    display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
    display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2 - 10, "OTA Update");
    display.display();
  });
  ArduinoOTA.onEnd([]() {
    display.clear();
    display.setFont(ArialMT_Plain_10);
    display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
    display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2, "Restart");
    display.display();
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {

    display.drawProgressBar(4, 32, 120, 8, progress / (total / 100));
    display.display();
  });
  ArduinoOTA.onError([](ota_error_t error) {
    display.clear();
    display.setFont(ArialMT_Plain_10);
    display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
    display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2 - 10, "OTA Error :");
    if (error == OTA_AUTH_ERROR)
      display.drawString(DISPLAY_WIDTH / 2, 32, "Auth Failed");
    else if (error == OTA_BEGIN_ERROR)
      display.drawString(DISPLAY_WIDTH / 2, 32, "Begin Failed");
    else if (error == OTA_CONNECT_ERROR)
      display.drawString(DISPLAY_WIDTH / 2, 32, "Connect Failed");
    else if (error == OTA_RECEIVE_ERROR)
      display.drawString(DISPLAY_WIDTH / 2, 32, "Receive Failed");
    else if (error == OTA_END_ERROR)
      display.drawString(DISPLAY_WIDTH / 2, 32, "End Failed");
    display.display();
    delay(5000);
  });
  ArduinoOTA.begin();

  display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
  display.setFont(ArialMT_Plain_10);
  display.drawString(DISPLAY_WIDTH / 2, 20, "Ready for OTA:\n" + WiFi.localIP().toString());
  display.drawString(DISPLAY_WIDTH / 2, 50, "TAG server:\n" + server);
  display.display();
  Serial.println("Ready for OTA:" + WiFi.localIP().toString());
  Serial.println("TAG server:" + server);
  delay(5000);
  display.clear();
  display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
  display.setFont(ArialMT_Plain_10);
  display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2, "Starting to read tags....");
  display.display();
  Serial.println("Starting to read tags...");
  delay(3000);
}

void loop()
{
  ArduinoOTA.handle();

  if (!mfrc522.PICC_IsNewCardPresent())
  {
    display.clear();
    display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
    display.setFont(ArialMT_Plain_10);
    display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2, "No tag read, waiting....");
    display.display();
    Serial.println("No tag read, waiting....");
    delay(50);
    return;
  }
  if (!mfrc522.PICC_ReadCardSerial())
  {
    display.clear();
    display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
    display.setFont(ArialMT_Plain_10);
    display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2, "Could not read UID, trying again....");
    display.display();
    Serial.println("Could not read UID, trying again....");
    delay(50);
    return;
  }
  String uid = readUID(mfrc522.uid.uidByte, mfrc522.uid.size);
  displayTag(uid);
  mfrc522.PICC_HaltA();
  mfrc522.PCD_StopCrypto1();

  HTTPClient http;
  http.begin(server + ":" + port + "/tags/" + uid);
  int httpCode = http.sendRequest("PUT");
  String reply = "";
  if (httpCode > 0)
  {
    reply = http.getString();
  }
  else
  {
    reply = "Error:" + httpCode;
  }
  http.end();
  display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
  display.setFont(ArialMT_Plain_10);
  display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2 + 20, "Http:" + reply);
  display.display();
  Serial.println("Http:" + reply);
}

void displayTag(String uid)
{
  display.clear();
  display.setTextAlignment(TEXT_ALIGN_CENTER_BOTH);
  display.setFont(ArialMT_Plain_10);
  display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2, "UID:" + uid);
  MFRC522::PICC_Type piccType = mfrc522.PICC_GetType(mfrc522.uid.sak);
  display.drawString(DISPLAY_WIDTH / 2, DISPLAY_HEIGHT / 2 + 20, mfrc522.PICC_GetTypeName(piccType));
  display.display();
  Serial.println("UID:" + uid);
  Serial.println(mfrc522.PICC_GetTypeName(piccType));
}

String readUID(byte *buffer, int buffersize)
{
  String uid = "";
  for (byte i = 0; i < buffersize; i++)
  {
    uid += buffer[i] < 0x10 ? "0" : "";
    uid += buffer[i], DEC;
  }
  return uid;
}

```

igrr, 2018-01-04:
@pkishino If you are using latest git, this may be a regression related to https://github.com/esp8266/Arduino/pull/4086.

pkishino, 2018-01-04:
Ah, I'll try again with the latest from right now

yoursunny, 2018-01-04:
Please don’t blame me. @pkishino ‘s sketch was posted 6 hours earlier than #4086 was merged.
Is there a protocol documentation on how OTA is supposed to work (the packet format)?
Also, it would be helpful if @pkishino posts a tcpdump on what packets are sent to the ESP8266 when OTA is working and when it’s not working. The tcpdump can be captured on a home router, for example. I usually use Windows’s “wlan hostednetwork” feature for this.

igrr, 2018-01-04:
@yoursunny no blame cast at all, i just thought that it might be related, because saw some OTA related changes being merged.

I have updated to latest git version and can confirm that I'm seeing a crash when using OTA, albeit a different one. Rolling back to 2.4.0, no issue is observed. Also no issue is observed if i make the change mentioned by @Juppit in #4086.

devyte, 2018-01-04:
A bug was introduced yesterday related to OTA as part of a OTA stability fix.
@yoursunny no blame :) your fix was needed, and I also missed the unneeded index increments when reviewing.

pkishino, 2018-01-04:
Hmm, I have now tried the code on several other nodemcu boards and using a new project and it is working fine.. I think based on this and a few other issues that the first board might have somehow become defective..

closed 2018-01-04 by pkishino