esp8266/Arduino #4082: v 2.4.0 and Soft WDT reset while WiFi.forceSleepBegin(); execution

essenemari, 2018-01-03:
### Basic Infos
After board upgrade: "esp8266 by Community" from 2.3.0 to 2.4.0 observed Soft WDT reset on WiFi.forceSleepBegin(); execution. 
#### Hardware
Hardware:			ESP-12E, ESP-07
Core Version:      	2.4.0

### Description

Hi Team,
**thanks a lot for your awesome work on ESP8266! Appreciate!** 

I am using WiFi.forceSleepBegin(); as critical command to turn off WiFi module for 10 minutes then turnning  on WiFi module for 15 seconds to execute commands then turn WiFi off again. On 2.3.0 it does work perfect (currently ~10 ESP devices).
Noticed recently, that there is new 2.4.0 ESP8266 board software version, so upgraded and found out that my ESPs are reseting. If you'll take a look for loop it crashes exactly after `Serial.println("Before WiFi.forceSleepBegin()");` and usually on 2nd or 3rd loop.
When downgraded to esp v2.3.0 - back to normal.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		521K
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
#include "JsonStreamingParser.h"
#include "JsonListener.h"
#include "ExampleParser.h"
#include "Base64.h"
#include <ESP8266WiFi.h>
long int cykl_counter_start = 600000;

void setup() {
}

void loop() {
  for (cykl_counter = cykl_counter_start; cykl_counter > 0; cykl_counter--) {
    delay(1);
    if (cykl_counter == cykl_counter_delay) {
      delay(100);
      Serial.println("Before WiFi.forceSleepBegin()");
      WiFi.forceSleepBegin();
      delay(100);
      Serial.println("Sleeping");
    }
    if (cykl_counter == 15000) {
      Serial.println("WiFi.forceSleepWake()");
      WiFi.forceSleepWake();
      Serial.println("Wakeing");
      delay(100);
    }
    if (cykl_counter == 10000) {
      Serial.println("Tasks");
      delay(100);
    }
  }
}

### Debug Messages

```
Soft WDT reset

ctx: cont 
sp: 3fff0770 end: 3fff0970 offset: 01b0

>>>stack>>>
3fff0920:  0001360d 00000600 3ffef638 4022608c  
3fff0930:  0fffffff 4020533a 00000000 40205325  
3fff0940:  00000000 3ffef53c 3ffef540 40203a16  
3fff0950:  3fffdad0 00000000 3ffef93c 40207090  
3fff0960:  feefeffe feefeffe 3ffef950 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,1)

 ets Jan  8 2013,rst cause:4, boot mode:(1,1)

wdt reset

```

trentbrown13, 2018-01-04:
perhaps related to
https://github.com/esp8266/Arduino/issues/3662

igrr, 2018-01-04:
@essenemari could you please add the definitions of `cykl_counter` and `cykl_counter_delay`?

igrr, 2018-01-04:
@essenemari Could you please try with "lwIP Variant" set to "v1.4 Prebuilt" in Tools menu? I'm not getting my variant of the crash with lwIP 1.4.

essenemari, 2018-01-04:
Hello igrr,
thanks a lot for looking into that. I have changed "lwIP Variant" set to "v1.4 Pre Built" but unfortunately it did not help. Because issue occurs  around WiFi, let me put full Sketch and somekind of debugger.

### Sketch

#include "Base64.h"
#include <ESP8266WiFi.h>
#define EIOT_USERNAME         ""
#define EIOT_PASSWORD         ""
#define EIOT_IP_ADDRESS       ""
#define EIOT_PORT             81
#define USER_PWD_LEN          40
WiFiClient rPiClient;
char unameenc[USER_PWD_LEN];
int cykl_counter_start = 20000;
int cykl_counter = 0;
int cykl_counter_delay = cykl_counter_start - 100;
int WifiConnectionAttempt = 0;
const char* ssid = "";
const char* password = "";
void setup() {
  Serial.begin(115200);
  //initial Wifi and  server connection
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println();
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  delay(8000);
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("Initial WiFi connection OK");
    Serial.println(WiFi.localIP());
  } else
    Serial.println("Initial WiFi connection FAILED");
  rPiClient.connect(EIOT_IP_ADDRESS, EIOT_PORT);
  delay(10);
  if (rPiClient.connect(EIOT_IP_ADDRESS, EIOT_PORT))
    Serial.println("Initial connection to server OK");
  else
    Serial.println("Initial connection to server FAILED");
  char uname[USER_PWD_LEN];
  String str = String(EIOT_USERNAME) + ":" + String(EIOT_PASSWORD);
  str.toCharArray(uname, USER_PWD_LEN);
  memset(unameenc, 0, sizeof(unameenc));
  base64_encode(unameenc, uname, strlen(uname));
}
void loop() {
  for (cykl_counter = cykl_counter_start; cykl_counter > 0; cykl_counter--) {
    delay(1);
    if (cykl_counter == cykl_counter_delay) {
      delay(100);
      Serial.println("Before WiFi.forceSleepBegin()");
      WiFi.forceSleepBegin();
      delay(100);
      Serial.println("Sleeping");
    }
    if (cykl_counter == 15000) {
      Serial.println("WiFi.forceSleepWake()");
      WiFi.forceSleepWake();
      Serial.println("Wakeing");
      delay(100);
    }
    if (cykl_counter == 10000) {
      Serial.println("Tasks");
      delay(100);
    }
  }
}
void connectivity() {
  if (WiFi.status() != WL_CONNECTED || !rPiClient.connected()) {
    while (WifiConnectionAttempt < 3) {
      delay(1000);
      WiFi.begin(ssid, password);
      delay(10);
      rPiClient.connect(EIOT_IP_ADDRESS, EIOT_PORT);
      delay(10);
      WifiConnectionAttempt++;
      //Serial.println(WifiConnectionAttempt);
      if (WifiConnectionAttempt == 3)
        Serial.println("Not connected to WiFi or to  Server. Skipping connection to next loop.");
    }
    if (WiFi.status() == WL_CONNECTED) {
      Serial.println("WiFi connected");
      Serial.println(WiFi.localIP());
    }
    if (rPiClient.connect(EIOT_IP_ADDRESS, EIOT_PORT)) {
      Serial.println("Connected to  Server");
    }
  }
}

### Debbug
Initial WiFi connection OK
192.168.0.114
Initial connection to server OK
Before WiFi.forceSleepBegin()
Sleeping
WiFi.forceSleepWake()
Wakeing
Tasks
Before WiFi.forceSleepBegin()

Soft WDT reset

ctx: cont 
sp: 3ffefc40 end: 3ffefe50 offset: 01b0

>>>stack>>>
3ffefdf0:  0001360d 00000600 3ffeebbc 40219dac  
3ffefe00:  0fffffff 402023f6 00000000 402023e1  
3ffefe10:  00000000 3ffeedf0 3ffeeb78 4020209a  
3ffefe20:  00000000 00000000 00000001 3ffeee1c  
3ffefe30:  3fffdad0 00000000 3ffeee14 40203854  
3ffefe40:  feefeffe feefeffe 3ffeee30 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

Thelmos, 2018-01-19:
Same problem here...

Switched to 2.4.0-rc2 and all works again.

Exception was:
```
Soft WDT reset

ctx: cont 
sp: 3fff0e20 end: 3fff1090 offset: 01b0

>>>stack>>>
3fff0fd0:  0001360d 00000600 3ffefd28 40234330  
3fff0fe0:  0fffffff 40203822 00000000 4020380d  
3fff0ff0:  3ffefd1c 3ffefd24 3ffefd28 402025d1  
3fff1000:  3ffefd1c 3ffefd24 3ffeffa0 40203252  
3fff1010:  00000000 00000000 00000000 00000000  
3fff1020:  00000000 00000000 00000000 00000000  
3fff1030:  00000000 00000000 00000000 00000000  
3fff1040:  00000001 00000000 00000000 40106a05  
3fff1050:  00000000 3ffefd18 3fff0059 3fff0060  
3fff1060:  3fffdad0 00000000 3fff0059 4020328d  
3fff1070:  3fffdad0 00000000 3fff0059 402094f4  
3fff1080:  feefeffe feefeffe 3fff0070 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
Decoded stack trace:
```
Decoding 9 results
0x40234330: dhcp_renew at core/dhcp.c line 1083
0x40203822: WiFiEventHandlerOpaque::operator()(_esp_event*) at c:\users\thelmos\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
:  (inlined by) ESP8266WiFiGenericClass::_eventCallback(void*) at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp line 233
0x4020380d: std::list  , std::allocator   > >::_M_erase(std::_List_iterator   >) at c:\users\thelmos\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
:  (inlined by) std::list  , std::allocator   > >::erase(std::_List_iterator   >) at c:\users\thelmos\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/list.tcc line 112
:  (inlined by) ESP8266WiFiGenericClass::_eventCallback(void*) at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp line 230
0x402025d1: printConnectionType(int) at F:\Proyectos\Arduino\Alarma\Alarma/espWiFi2eeprom.ino line 135
0x40203252: ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress) at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiAP.cpp line 294
0x40106a05: timer0_isr_handler at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_timer.c line 80
0x4020328d: ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress) at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFiAP.cpp line 294
0x402094f4: std::_Sp_counted_base(__gnu_cxx::_Lock_policy)0>::_M_release() at c:\users\thelmos\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr_base.h line 157
:  (inlined by) std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at c:\users\thelmos\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr_base.h line 546
0x40100710: cont_norm at C:\Users\Thelmos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 113
```

d-a-v, 2018-01-25:
Hi, I also reproduced the bug.
It is also reproducable with WiFiOn/WiFiOff imported from at least #2330.
I slightly modified WiFiOff then I get it no more.
The added waiting loop has been running 66 times over 320 main loops (20 seconds each)
The test seems to be working but is not very statisfactory since it relies on dhcp state.
If it is proven working, I will try to find a lwip1/lwip2 compatible way to check the netif UP state instead.

I have no explanation why it was working with 2.4.0rc2.
What I can say is that it crashes when ```rm 0``` is coming too late hence the loop to wait for it (and fortunately ```wifi_station_get_connect_status``` helps with that).

Normal loop:
```
---disconnect
state: 5 -> 0 (0)
rm 0
pm close 7
---disconnected
---off...
```

With waiting loop otherwise crash happen:
```
---disconnect
---disconnected
dhcp not stopped?
state: 5 -> 0 (0)
rm 0
pm close 7
---off...
```
Updated full sketch:
```

#include <ESP8266WiFi.h>
#include <user_interface.h>

void WiFiOn()
{
  wifi_fpm_do_wakeup();
  wifi_fpm_close();
  wifi_set_opmode(STATION_MODE);
  wifi_station_connect();
}

void WiFiOff()
{
  Serial.println("---disconnect");
  wifi_station_disconnect();
  Serial.println("---disconnected");
  bool stopped;
  do
  {
    stopped = wifi_station_get_connect_status() == DHCP_STOPPED;
    if (!stopped)
    {
      Serial.println("dhcp not stopped?");
      delay(100);
    }
  } while (!stopped);
  Serial.println("---off...");
  wifi_set_opmode(NULL_MODE);
  wifi_set_sleep_type(MODEM_SLEEP_T);
  wifi_fpm_open();
  wifi_fpm_do_sleep(0xFFFFFFF);
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);   // try uncommenting if OTA does not work
  WiFi.persistent(false);
  WiFi.begin("open", "");
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(200);
  }
  Serial.println("ok");
}

void loop() {
  for (long int cykl_counter = 20000; cykl_counter > 0; cykl_counter--) {
    delay(1);
    if (cykl_counter == 18000) {
      delay(100);
      Serial.println("\n------------\nBEFORE OFF");
      Serial.println(system_get_free_heap_size());
      WiFiOff(); //WiFi.forceSleepBegin();
      delay(100);
      Serial.println("OFF");
    }
    if (cykl_counter == 15000) {
      Serial.println("\n------------\nBEFORE ON");
      Serial.println(system_get_free_heap_size());
      WiFiOn(); //WiFi.forceSleepWake();
      Serial.println("ON");
      delay(100);
    }
    if (cykl_counter == 10000) {
      Serial.println("Tasks");
      delay(100);
    }
  }
}
```

d-a-v, 2018-01-26:
@Thelmos @essenemari can you try this fix ?

essenemari, 2018-01-26:
@d-a-v 
thanks a lot that you've looked into that.

Just started the tests off your fix.
First, I am going check several hundred 45 second loop
Then, I am going to check dozens 10 minutes loop.

For now, the 45 second loop works fine.  

Thanks!

Teddyz, 2018-02-18:
I tested the WiFiOff(), WiFiOn() -functions above with 2.4.0 code and lwIP v1.4 prebuilt. Their functionality was put into my project involving conencting to an MQTT-broker via "256dpi/arduino-mqtt". My goal was to run ESP on less power by having no WiFi when not needed.
WiFiOff() turned off Wifi properly, but the current consumption rarely/randomly went down to the lower ,20-ish mA, power consumption from the normal ~70 mA. My solution was to add WiFi.forceSleepBegin() to the end of that function. Now it seems reliable. 
Also, it sometimes gets stuck in the "dhcp not stopped?"-loop. I have not yet managed to capture the output before this happens.

ardyesp, 2018-08-12:
I ran into same issue with 2.4.2, getting SoftWDT when invoking forceSleepBegin().
The issue does not appear when forceSleepBegin() is called within few short seconds of WiFi.forceSleepWake(), which might explain why some never experience it.

My workaround is to add a delay between disconnect and sleep. This code snippet works:
```
WiFi.disconnect(); 
delay(1000);
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(100);
```

jindrichsirucek, 2018-08-20:
I had the same problem.. @ardyesp's workaround seems working so far.. thx a lot for sharing

robynjayqueerie, 2018-10-25:
Similar problem here, I upgraded to IDE 2.4.2 primarily trying to solve the flash problem introduced by manufacturers using the PUYA flash chips - see issue #4061
Finally got that issue resolved and then compiled firmware which had worked fine in older IDE, not sure what version it was (may be back as far as 1). All went well till I went into light sleep and as soon as that happened I get the boot reset problem, similar to that here. This is the extract from my code, does not disable wifi, just goes into light sleep and is woken by a low level on GPIO4

		gpio_pin_wakeup_enable(GPIO_ID_PIN(4), GPIO_PIN_INTR_LOLEVEL); // 
		wifi_fpm_open(); // Enables force sleep
		wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time

I tried all the 2.4.* versions and went back to 2.3.0 and suddenly the issue disappeared. (I had to patch in the fix for the PUYA chips each time)
Can anyone advise what the possible problem is, seems as though an issue is introduced in ver 2.4.0 which creates a reset when you go into light sleep.
Thanks

devyte, 2018-10-25:
The issue is not in the core releases, but a change in sleep behavior in newer sdk versions. Among the behavior changes, e.g. you can't do "sleep for max int" anymore. 
Latest git integrates sdk pre-3, where the sdk also has some new api to better control sleep, which is used by the core. I suggest trying latest git.

robynjayqueerie, 2018-11-02:
Thanks @devyte However I am a little confused here. (Sorry happens a lot) I have upgraded to version 2.2.4 using the boards manager, I would think that was the latest version as per the esp8266/Arduino git. However when I chase through the documentation I see https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html?highlight=sleep  section WiFi power management, DTIM I see a reference to esp8266-arduino core v2.5.0 using the last V2 revision of nonos-sdk before V3). Is this the version you are referring to and where do I find it?
Also, looking at that documentation it does not use the libraries I have been using to date ie
	wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // set sleep type
	wifi_fpm_open(); // Enables force sleep
	wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time
Are you saying these libraries are now redundant and I should go with that shown in the documentation above?
Thanks for your help

devyte, 2018-11-02:
As I said above, the issue is due to changes in newer sdk versions. Latest git, aka master branch, integrates sdk pre-3, which provides an api to set sleep type. Our core was modified to make use of it.That is not released yet, you have to uninstall whatever you installed with board manager and install latest git.

robynjayqueerie, 2018-11-03:
farther to the last posting, the only way I have been able to get my sketch to upload was to install ver 2.4.2 in the IDE then copy all the files from the clone of the git over the files in .../Arduino15/.../2.4.2/*.*
This meant the IDE at least allowed the selection of ESP8266 Generic board and uploaded the sketch. The heading of the ESP boards selection was ESP8266 (2.5.0) so I guess it had the right cores and libraries installed.
However, getting back to the sleep issue where this all started I have had no luck with 
if(WiFi.setSleepMode (WIFI_LIGHT_SLEEP, 0))Serial.println("light sleep mode is set OK");
.
.
.
WiFi.forceSleepBegin (10000000);

As soon as the forceSleepBegin is executed the whole thing goes into reset. There is virtually no documentation on the new sleep functions as far as I can find so I am not sure I am doing it right. Any comments there?
thanks again

devyte, 2018-11-03:
Try ESP.deepSleepInstant() instead

robynjayqueerie, 2018-11-04:
@devyte But I don't want deep sleep, the process requires light sleep, not deep sleep, reset etc

robynjayqueerie, 2018-11-05:
@devyte Thanks for your help. I really appreciated everything you can provide. This product is semi commercial and we have been stalled, not able to go back (PUYA chip problem) and not able to go forwared (Light sleep problem)
I notice you did say earlier that the old 2.4.2 version had to be uninstalled, sorry I missed that.
Can you confirm that the correct installation of the git clone is in the Programs directory and not in the AppData directory please.
Thank you

robynjayqueerie, 2018-11-05:
Farther to the last post @devyte , what is the solution to the inability to upload a sketch to the device because of the esptool.exe issue I raised above. When I delete 2.4.2 and install the git clone the esptool does not work, see the print out 2 days ago

robynjayqueerie, 2018-11-06:
Has anyone got an example of light sleep happening successfully with 2.4.2 or the git clone or any other mods, basically hitting my head against a wall here and not getting anywhere.
Does anyone even have an explanation of WiFi.setSleepMode (WIFI_LIGHT_SLEEP) and what it does? It seems to me that calling this in 2.4.2 immediately sends the unit to sleep without calling WiFi.forceSleepBegin (). I really need to solve this one, customers are shouting at me. I had it all working nicely in an older version of the ESP board but every version available now seems to have holes, either in the sleep modes or wifi not reconnecting unless there is a power cycle etc.
Thanks

robynjayqueerie, 2018-11-07:
@devyte , @igrr Guys, I would really appreciate some help here, I know that you are busy.
@devyte commented above that 
    wifi_fpm_open(); // Enables force sleep
    wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time
does not work anymore, which is certainly my finding but when I do get into the libraries, looking at ESP8266Generic.cpp I find that the forceSleepBegin() function uses them anyway, ie
/**
 * Disable WiFi for x us when value is not 0
 * @param sleep_time_in_us
 * @return ok
 */
bool ESP8266WiFiGenericClass::forceSleepBegin(uint32 sleepUs) {
    _forceSleepLastMode = getMode();
    if(!mode(WIFI_OFF)) {
        return false;
    }

    if(sleepUs == 0) {
        sleepUs = 0xFFFFFFF;
    }

    wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
    wifi_fpm_open();
    return (wifi_fpm_do_sleep(sleepUs) == 0);
}
A couple of things here, firstly the code ignores any sleep type that may have been set up previously because it has hard coded wifi_fpm_set_sleep_type(MODEM_SLEEP_T); when this should be
wifi_fpm_set_sleep_type(_forceSleepLastMode);???????
Secondly, the code just does what I had always done in the past and uses wifi_fpm_open(); and wifi_fpm_do_sleep(sleepUs) where sleepUs can be set to the same maximum value 0xFFFFFFF  which is Sleep for longest possible time.
So can someone explain what is happening here, obviously the function never works in Light Sleep because of the hard coding and have the wifi_fpm functions been changed in some way to prevent them working. Any possibility of this latest git version working seems slim
Thanks in advance, anyone who can enlighten me

tve, 2018-11-14:
I can confirm that using the latest git version (master) as of nov 14 has light sleep working. All you need to do is call `WiFi.setSleepMode(WIFI_LIGHT_SLEEP, 5)`, pick your favorite DTIM value. This way the chip remains connected to the wifi access point but turns the radio off. *However* I'm finding that the resulting power level is still much higher than advertised. See https://bbs.espressif.com/viewtopic.php?f=7&t=24393 (also has some scope screen shots).

I have not tried `WiFi.forceSleepBegin` because that only goes into modem_sleep. In any case, you need to use master to get the latest important fixes, specifically https://github.com/esp8266/Arduino/pull/5210

Overall, I'd says that the esp8266 still sucks at low power. It seems Espressif has fixed almost nothing in the past two years. Really disappointing.

DIRR70, 2019-03-31:
Hello everybody,

"wifi_station_get_connect_status" (see suggestion from @d-a-v - thank's for that!) won't work for me.
It prints "dhcp not stopped?" several times and then the watchdog is triggered.

With 2.5.0 (lwIP v1.4 or v2) I use tickers to check, whether the connection is established or not (every 5secs) and also a ticker to force sleep after disconnect (3secs). I also use "setSleepMode" with highest listenInterval. That gives me less than **20mA** most of the time. If a connection is tried to be established or is currently used the current is 70mA and if the connection is lost (router's WiFi turned off) 70mA (with bursts up to 250mA) for max. 5secs.

Here's the full sketch with a simple WebServer as example:

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <Ticker.h>

const char* ssid = "...";
const char* password = "...";

ESP8266WebServer webServer (80);

Ticker staTicker;
Ticker offTicker;

bool staOn = true; // true, if WiFi is on

byte requests; // just to let the webServer return something

void setup(void) {
  delay(1000); // power up

  pinMode (2, OUTPUT); // LED near antenna is on while WiFi is on (just an example)

  WiFi.persistent (false); // I just hate the persistent nonsense
  WiFi.setSleepMode (WIFI_MODEM_SLEEP, 10); // Save power if connection is established but not used (delay in loop required!)

  staTurnOn (); // Try to establish connection

  webServer.on("/", []() {
    webServer.send (200, "text/plain", "Hello #" + String (++requests));
  });
  webServer.begin (); // Needs to be started only once even if connection gets lost for a while
}

void staTurnOn () {
  WiFi.mode (WIFI_STA); // Set always, because set to WIFI_OFF in offTick
  WiFi.begin (ssid, password); // GO!

  staTicker.detach (); // Because staTurnOn is called from staTick
  staTicker.attach (5.0, staTick); // Check in 5 seconds if we are connected
}

void staTurnOff () {
  WiFi.disconnect (); // STOP!
  offTicker.attach (3.0, offTick); // After 3 seconds force sleep

  staTicker.detach (); // Because staTurnOn is called from staTick
  staTicker.attach (180.0, staTick); // Check again in 3 minutes
}

void staTick () {
  if (staOn) {
    if (WiFi.status () == WL_CONNECTED) return;
    staTurnOff ();
    staOn = false;
  } else {
    WiFi.forceSleepWake ();
    staTurnOn ();
    staOn = true;
  }
}

void offTick () {
  offTicker.detach (); // only once!

  WiFi.mode (WIFI_OFF); // Required to make forceSleepBegin work
  WiFi.forceSleepBegin (); // Turns current down permanently to <20mA (needs delay in loop, too)
}

void loop(void) {
  if (staOn) { // We don't need to handle if WiFi is off
    // Other stuff like DNS can go here, too...
    webServer.handleClient();
  }

  // Your code goes here...
  digitalWrite (2, (staOn) ? LOW : HIGH);

  delay (10); // Together with setSleepMode in the setup it saves power
}
```

trentbrown13, 2019-04-18:
Just to confirm, I still had the same issue with WiFi.forceSleepBegin() after upgrading from 2.3 to 2.5. The workaround posted by ardyesp above worked for me, my sketch has been up for over two days down, with a force sleep every 60 seconds.
 This seems to be same as the issue https://github.com/esp8266/Arduino/issues/3662 that is also still open - I'll post there also.

As an aside, I also had a mqtt publish statement directly before the WiFi.forceSleepBegin() statement that worked in 2.3. With 2.5 the message was not published until I added another small sleep(1000) just before WiForceSleepBegin().

It now looks like this
{code}
publishSleepState(1);
    delay(500);  // added delay as above does not work with core 3.5
    WiFi.disconnect();
    //************* HACK!
    delay(1000);
    WiFi.mode(WIFI_OFF);

    Serial.print(F("************ forceing modem sleep "));
    WiFi.forceSleepBegin();
{code}

earlephilhower, 2019-07-25:
Closing as we've got an active discussion of this in #6172 where folks are making some progress, I believe.

closed 2019-07-25 by earlephilhower