esp8266/Arduino #4084: Having issue with ultrasonic sensor

tjahmed, 2018-01-04:
Hello Sir! i am working on a project such as to send the ultrasonic sensor data to my http server with the help of ESP8266 Wifi-Module, I successfully upload the code. but on serial monitor it sends Distance =0 values and i also checked  to my server, having same issue. My esp8266 is working fine checked by sending raw data manually to server.Now i have a confusion regarding ESP8266 is that, when we use ESP8266 module then we select the board to generic ESP8266 for that all the operation works related to module. but the Question is that when we interface sensor with module, than how esp8266 calculate the sensor values,cause we deselect the board  arduino Uno  that calculate the sensor values with the help of AT-mega controller  but for ESP is it able to calculate the sensor values cause esp-module is not a controller itself.

Here is the Pin-configuration for my project.

Arduino |  sensor  |   ESP
5V      |    5v    |  GIving 3.3 volt  by input 5v from arduino using 3.3 voltage regulator-ic
        |          |       both(ch-vcc);
        |          |
ground  |   ground |   ground

12      |    tring          

13      |    echo 
ground ----------------->  GPIO0
                               GPIO2(not connected)  
Tx----------------------------> Tx
RX-----------------------------> RX

Note: I successfully send the DHT-temprature sensor data to Thingspeakserver. but i don't know  why it's not working when i want to send sensor values to my own server. please help me it's been week that I am trying but didn't get success.

devyte, 2018-01-05:
This repo implements an Arduino code layer that allows a user to implement code directly on the ESP8266 in a similar manner as can be done on an Arduino board. Therefore, if you need to retrieve sensor data, then you don't need the Arduino anymore. You develop your code, build it, upload it to the ESP, and then you communicate with the ESP.
If you must have an Arduino board involved, then you would have one application for the Arduino board, and a different application for the ESP board.
This is not the right place to request help for your particular project, or to ask general "how do I..." type questions. This is an issue tracker, meant to track issues in the ESP core libs. For further assistance, please refer to a community forum like esp8266.com.
Closing per issue policy doc.

closed 2018-01-05 by devyte

tjahmed, 2018-01-06:
ok i will give it a try..