esp8266/Arduino #4085: v2.4.0  - ESP8266WebServer missing

bill-orange, 2018-01-04:
Hardware:		Wemos R2
Core Version:      	unknown

Description: (see below)

My sketch worked fine with v2.3.0.  compiling with 2.4.0 I get:

use of deleted function 'ESP8266WebServer::ESP8266WebServer(const ESP8266WebServer&)

compiling with: 2.4.0 vs 2.3.0

Problem description:  Can not compile

### Settings in IDE

Module:  			ESPDurino D2 mini
Flash Size: 		4 MB
CPU Frequency:	80Mhz

igrr, 2018-01-04:
Please post the (preferably minimal) sketch which has this problem.

bill-orange, 2018-01-04:
Unfortunately the sketch is large.  Perhaps an excerpt will be sufficient.  If not here is a link to the whole sketch.

Here is a link to the whole sketch:
https://1drv.ms/u/s!AmXKqAwyCrbxh-BLqssNGUJ4Sdh92Q

excerpt:
`
#include <ESP8266WebServer.h>

ESP8266WebServer server = ESP8266WebServer(port);// create a web server on port 8021

/*------------------------------startServerFUNCTIONS----------------------------------*/

void startServer() { // Start a HTTP server with a file read handler and an upload handler
  server.on("/edit.html",  HTTP_POST, []() {  // If a POST request is sent to the /edit.html address,
    server.send(200, "text/plain", "hello from esp8266!");
  }, handleFileUpload);                       // go to 'handleFileUpload'

  server.onNotFound(handleNotFound); // if someone requests any other file or page, go to function 'handleNotFound'
  // and check if the file exists

  server.begin();                    // start the HTTP server
  Serial.println("HTTP server started.");
}

void setup() {
  startServer();
  }
void loop() {
  server.handleClient();
  }
`

igrr, 2018-01-04:
Indeed, the issue is with `ESP8266WebServer server = ESP8266WebServer(port);`

Is there some specific reason why you need to use assignment here, rather than just `ESP8266WebServer server(port);`?

bill-orange, 2018-01-04:
Probably not. I will test it with this change tomorrow. There are many sample sketches around with the syntax I used.

bill-orange, 2018-01-04:
Worked fine when using v2.4.0 and` ESP8266WebServer server(port);`

Thanks

devyte, 2018-01-05:
@igrr The compiler seems to think that the copy constructor is deleted, but I don't see that explicitly done in the class. AFAIK, the default copy constructor for a class T is implicitly deleted if one of these conditions is met:
1. T has non-static data members that cannot be copied (have deleted, inaccessible, or ambiguous copy constructors);
2. T has direct or virtual base class that cannot be copied (has deleted, inaccessible, or ambiguous copy constructors);
3. T has direct or virtual base class with a deleted or inaccessible destructor;
4. T has a user-defined move constructor or move assignment operator;
5. T is a union-like class and has a variant member with non-trivial copy constructor;
6. T has a data member of rvalue reference type.

Ref: [cppreference](https://en.cppreference.com/w/cpp/language/copy_constructor#Deleted_implicitly-declared_copy_constructor)

WiFi has multiple inheritance, but I don't see 2, 3, 4, 5, or 6 anywhere, so my guess is 1.: some member was added that the compiler thinks has a non-trivial constructor.

igrr, 2018-01-05:
That's ESP8266WebServer, not WiFi class. It has a `unique_ptr` member which can not be copied, case 1). In general, copying ESP8266WebServer is not a well defined operation. Should we expect that two web server instances on the same port somehow continue to work after copying? Or only one of them works (in which case this is more like a move operation)?

Possible solutions are:

a) Replace unique_ptr with shared_ptr, make copying possible. This will allow the OPs use case to work, but would allow silent copying of ESP8266WebServer object. Since this object contains some pointers which are managed manually, in general copying it is not a safe thing to do.

b) Implement move constructor — i think this will also allow the OPs use case to work (not sure, need to check).

c) Don't allow copying at all. This will require the existing sketches to be updated as mentioned above.

devyte, 2018-01-05:
I read ESP8266WebServer but my brain registered ESP8266WiFi. Ok, enough for today, getting some sleep now :p
IMT(ired)O, copying should be allowed, but in a well defined manner. 
In contrast, ESP8266WiFi should not allow copy.

SciLor, 2018-01-07:
Do you have any workaround I may use here:

https://github.com/SciLor/ESP8266_Hyperion_LED-Controller/blob/master/HyperionRGB/WrapperWebconfig.h

bill-orange, 2018-01-07:
I think if you change the line ` ESP8266WebServer _server = ESP8266WebServer(80);`  to
`ESP8266WebServer _server(80);`  it should work,

SciLor, 2018-01-07:
Won't work.

WrapperWebconfig.h:83: error: expected identifier before numeric constant
     ESP8266WebServer _server(80);// = ESP8266WebServer(80);

mdhiggins, 2018-01-07:
I too am having issues with this change
With my project users used to have option to load a custom port from WiFiManager settings after they were read from a JSON file and I would reinitialize the EPS8266WebServer as follows:

```
  if (port != defaultport) {
    Serial.println("Default port changed");
    server = ESP8266WebServer(port);
  }
```

A way to change the port would be ideal, but this workaround doesn't work anymore with this change

bill-orange, 2018-01-07:
This is not really a work-around but have you tries compiling with v2.4.0 using earlier versions of lwIP?
[image]

mdhiggins, 2018-01-07:
Found a workaround

Had to change 
```
ESP8266WebServer server(port);
...
  if (port != 80) {
    Serial.println("Default port changed");
    server = ESP8266WebServer(port);
  }
```

to 

```
ESP8266WebServer *server;
...
  server = new ESP8266WebServer(port);
```

Had to change all syntax from `server.*` to `server->*` as well with this formatting but seems to be compiling

devyte, 2018-01-07:
@mdhiggins you have to delete the pointer before doing another new, or it will leak.
There is a PR with a proposal for adding a method that changes the port.
About copy construction (and assignment), it currently doesn't work correctly, so for the time being it's better that it doesn't compile.

mdhiggins, 2018-01-07:
@devyte good point, I just never initialized the first 'new' and removed the conditional to check if the defaultport is set. Do I still need to delete?

devyte, 2018-01-07:
Init the ptr to nullptr, then check it right before you new it if !=nullptr. If so, delete it.

chun48, 2018-01-07:
Hello, I am not an expert but I had the same problem with a NodeMCU V1.0 and I solved it by installing the version of esp8266 2.4.0-rc2
[image]

devyte, 2018-01-07:
@chun48 rc2 was a release candidate, and is now superseded by 2.4.0. 
To be blunt: any code that copies the webserver object is wrong, as the class was not designed for it.
If what is needed here is to change the port, then there is a PR proposing an implementation. Please  test that. If it works, it can be merged.

devyte, 2018-01-07:
Referenced pr is #2716 , adds a begin(port) method.

igrr, 2018-01-08:
@devyte That PR is for WiFiServer, not ESP8266WebServer... However adding an equivalent method for ESP8266WebServer should not be hard once #2716 is merged.

devyte, 2018-01-18:
Ok, created PR #4148 for the port method.

FahadHussain777, 2018-01-27:
@bill-orange ESP8266WebServer _server(80) is working in my case

devyte, 2018-02-07:
Closed via #4148 . Don't use.copy constructor anymore, it's still disallowed.

closed 2018-02-07 by devyte

Kermit66, 2018-05-10:
Same problem with version 2.4.1
Ok with version 2.4.0
I use Wemos D1 R2, gizwits module, Adafruit HUZZAH ESP8266 Breakout
The server don't start with version 2.4.1

d-a-v, 2018-05-10:
@Kermit66 have you tried the erase flash options ?

Kermit66, 2018-05-20:
Yes always delete the memory