esp8266/Arduino #4088: Update WiFiClient example

igrr, 2018-01-04:
WiFiClient example shows how to send an HTTP request to Sparkfun Data service. However, this service has been discontinued. Some users attempt to use WiFiClient example as a base when implementing HTTP client functionality, which is suboptimal because there are better libraries for that purpose (ESP8266HTTPClient).

Need to develop a new WiFiClient example which will not use a discontinued service, and would not demonstrate something that is not considered best practice.

d-a-v, 2018-01-07:
Why not just making a [Quine](https://en.wikipedia.org/wiki/Quine_(computing)) by reading and printing a small page that we know will live not shorter than the sketch ? This one: 

http://raw.githubusercontent.com/esp8266/Arduino/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino

...The bad news is github serving https only :disappointed:

igrr, 2018-01-07:
That would be a nice example for HTTPClient library.

dragondaud, 2018-01-07:
http://www.iana.org/ will always be there... they're the root of DNS. :)

yoursunny, 2018-01-07:
Without asking the user to setup a TCP server (it's hard in Windows), there are limited choices for non-HTTP TCP servers.

* `time.nist.gov:13`: [daytime protocol](https://www.nist.gov/pml/time-and-frequency-division/services/internet-time-service-its)
* `djxmmx.net:17`: [quote of the day](https://gkbrk.com/wiki/qotd_protocol/)

A drawback is that the ESP8266 cannot send anything, but is receive-only.

d-a-v, 2018-01-08:
Then what about a WiFiClientBasic example:
* sending "hello"
* reading back data
* using by default quote of the day (only once, we want the service to stay alive :)
* having some included comments explaining how to use ```nc``` for the sent "hello" to be displayed, for windows and unix

tablatronix, 2018-01-30:
Some kind of loopback might be nice then you can see the http in the same code file as you expect to receive from client.

PR #5197 (2018-10-11): WiFi: improve WiFiClient(Basic) examples
WiFiClient no longer depends on now-defunct data.sparkfun.com
service, but uses a TCP "quote of the day" service instead.

fixes #4088
commit d17ffc28 (2018-10-11): WiFi: improve WiFiClient(Basic) examples (#5197)

WiFiClient no longer depends on now-defunct data.sparkfun.com
service, but uses a TCP "quote of the day" service instead.

fixes #4088
closed 2018-10-11 by devyte