esp8266/Arduino #4091: wifi.Status returns 6 while busy connecting to AP, not 0.

errolt, 2018-01-04:
### Basic Infos

#### Hardware
Hardware:		NodeMCU 0.9
Core Version:      	2.4.0

### Description

WiFi.Status() returns 6 while attempting to connect to AP. According to documentation, 6 means that the module is not set up for Station mode. Should it not return 0 while connecting to AP?

### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
#include <ESP8266WiFi.h>
void setup() {
  // put your setup code here, to run once:
  WiFi.begin("xyz", "xyz"); // connect to the network
  Serial.begin(115200);
  Serial.println("Reset Reason: " + String(ESP.getResetReason()));
  Serial.print("Auto Connect:"); 
  Serial.println(WiFi.getAutoConnect()?"True":"False");
  Serial.println("Core Version:" + String(ESP.getCoreVersion()));
  Serial.println("SDK Version:" + String(ESP.getSdkVersion()));
  

  while(WiFi.status() != WL_CONNECTED)
  {
  Serial.println("WiFi.Status():" + String(WiFi.status()));
    delay(100);
  }
  Serial.println("WiFi.Status():" + String(WiFi.status()));
  Serial.println("WiFi connected");

}

void loop() {
  // put your main code here, to run repeatedly:
    ESP.deepSleep(1000000 * 10, WAKE_RF_DEFAULT);
}

### Debug Messages
Auto Connect:True
Core Version:2_4_0
SDK Version:2.1.0(deb1901)
WiFi.Status():6
wifi evt: 2
WiFi.Status():6
WiFi.Status():6
WiFi.Status():6
WiFi.Status():6
WiFi.Status():6
WiFi.Status():6
WiFi.Status():6
wifi evt: 0
WiFi.Status():6
wifi evt: 3
WiFi.Status():3
WiFi connected
state: 5 -> 0 (0)
rm 0
del if0
usl
wifi evt: 1
STA disconnect: 8
enter deep sleep

kapyaar, 2018-01-08:
I have had a similar issue. In my case it was mostly in the second or third time (as in program, run, turn off,  and then when turned back on) that I would see this issue. After many digs, Something that seems to improve the situation was as follows.

Wifi.disconnect();
delay(1);
WiFi.begin("xyz", "xyz"); // connect to the network

Give it a shot and see if this works.

errolt, 2018-01-10:
Hi kapyaar,

This is for a battery powered device, so it must connect as fast as possible, I can't afford to kill the connection that is in progress, wait 1ms, then restart the connection process.

Thank you,
Errol

kapyaar, 2018-01-10:
True, and I am not even sure if this will fix your issue, but if not having wifi.disconnect() will keep the module waiting for many more milliseconds and there is no other obvious fix, this might be worth a try. :)

neu-rah, 2019-03-26:
I'm facing a similar issue, wifi state seems to get updated only when there is activity, servers can sit behind a lost connection (reporting state 3) while the connection was long gone. My assembly is primarly a server so it sits there waiting.

Periodic pings to the gateway made `onStationModeDisconnected` fire again when a connection is lost... the first attempt with a ping utility revealed inadequate has it was a blocking thing, so I came up with this short tick. One non-blocking ping request and a tolerance check.

```c++
extern "C" {
  #include <ping.h>
}

void tick_back(void *opt, void *resp) {
  constexpr int ticks_tolerance=3;//how many fails in a row will we tolerate (or sort off)
  static volatile int ticks_ok=0;
  ping_resp* ping_resp = reinterpret_cast<struct ping_resp*>(resp);
  if (ticks_ok>0&&ping_resp->ping_err==-1) ticks_ok--;
  else if (ticks_ok<ticks_tolerance) ticks_ok++;
  wifiConnected&=ticks_ok>0;//update my own state
}

//network tick, not using delay stuff and sending only one packet
void nw_tick(IPAddress dest) {
  static ping_option tick_options;
  memset(&tick_options, 0, sizeof(struct ping_option));
  tick_options.count = 1;
  tick_options.coarse_time = 0;
  tick_options.ip = dest;
  tick_options.sent_function = NULL;
  tick_options.recv_function = reinterpret_cast<ping_recv_function>(&tick_back);
  ping_start(&tick_options);
}

...

nw_tick(WiFi.gatewayIP());//network tick, not realy a ping, keep things alive
```

it might shed some light on what the problem is

GitArunsh, 2020-03-07:
trick by niteshjindalxb worked for me.

Westcott1, 2021-06-18:
Thanks, that worked for me as well!
An old ESP8266 would no longer connect, but an ESP32 did on the same code

alexandre-pansan, 2024-12-28:
I Know tha this issue seems really old.. BUT..
 
 Im trying to build a boat controller, which needs to be  battery powered.. and only on the Batery it does that error..  tru the usb cale it doesn't.. and btw. i'm using a esp32 c3 super mini.  althought, that solution worked for me as well.. thanks a lot