esp8266/Arduino #4094: Strange char array behaviour

SupraJames, 2018-01-04:
### Basic Infos

#### Hardware
Hardware:	       ESP8266MOD ? NodeMCU / Amica
Core Version:      	2.3.0

### Description

All I am trying to do is copy an array of chars to a new array and stick a '+' on the end, without resorting to using Strings.

So if controlTopic is "james" then subTopic needs to be "james+".

The sketch runs and in the first loop the second array (subTopic) contains the source data PLUS garbage characters, making the string longer than it should be, proven by strlen().

Strangely, second and subsequent runs round the loop are fine. Please help me figure this out!

Output showing twice round the loop:

controlTopic: testString
subTopic: testString+?⸮ @⸮⸮⸮?
strlen controlTopic: 10
strlen subTopic: 20
controlTopic: testString
subTopic: testString+
strlen controlTopic: 10
strlen subTopic: 11

Same code works fine on an Arduino UNO, correct values are displayed first time every time.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512KB with 64K SPIFFS
CPU Frequency:       80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:	        nodemcu

### Sketch

```
const char* controlTopic = "testString";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {

  // Allocate a new buffer one char bigger than controlTopic
  char subTopic[strlen(controlTopic) + 1];
  // Copy the contents of controlTopic to it
  strncpy(subTopic,controlTopic,strlen(controlTopic));
  // Add a '+' as the final character
  subTopic[strlen(controlTopic)] = '+';

  //Diag output
  Serial.print("controlTopic: ");
  Serial.println(controlTopic);
  Serial.print("subTopic: ");
  Serial.println(subTopic);
  Serial.print("strlen controlTopic: ");
  Serial.println(strlen(controlTopic));
  Serial.print("strlen subTopic: ");
  Serial.println(strlen(subTopic));
  
  delay(5000);
}
```

mtiutiu, 2018-01-04:
As far as I know and correct me if I'm wrong but when you allocate the new buffer you need to make it +2 not +1 to accommodate the string terminator too. So this line should be char subTopic[strlen(controlTopic) + 2]; instead of char subTopic[strlen(controlTopic) + 1];
 I will also do a memset on the new buffer and fill it with zeros afterwards just to be on the safe side.

devyte, 2018-01-04:
Previous comment is correct.
Closing as user error.

closed 2018-01-04 by devyte

SupraJames, 2018-01-04:
Hmm, yes good point. I tried that during my troubleshooting but over complicated things by trying to manually set the null at the end of the string. I’ll give it another go with the one change only. Good call about the memset. Guesss I’m just confused as to why it sorta corrects itself the second loop around!

mtiutiu, 2018-01-04:
In don't know how safe are variable length arrays or VLA's introduced by the C99 standard. I don't use that nor dynamic allocation as I'm afraid of memory fragmentation. I prefer to use fixed size arrays whenever possible. In C/C++ ( Arduino is a mixture of both) you get strange/undefined results (some are funny though) when not taking care of string termination character and at some point the program crashes if not immediately as far as I know.