esp8266/Arduino #4098: [Problem] ets Jan 8 2013,rst cause:1, boot mode:(3,4) - HELP

PunisherX99, 2018-01-05:
### Basic Infos

Hello to everyone! I'm new on github and the reason is that i prefer to resolve my "granades" myself.
But i'm actually disperate! I've set-up an arduino with an esp8266 12-e module on it , but i get this weird response when i give current to it :

 ets Jan  8 2013,rst cause:1, boot mode:(3,4)

load 0x40100000, len 816, room 16 
tail 0
chksum 0x8d
load 0x3ffe8000, len 788, room 8 
tail 12
m
ioofl 2lm
fde  e 
M 
 Sbmnrra e 

#### Hardware
Hardware:			Esp8266 12e

### Description

Well as i said i get that weird responde AFTER flashing it to AT 1.5.1 ( downloaded from the official site ) , i decide to upgrade it since WiFiEsp told me : "Unsupported Firmware" and i think that was the problem of failing connection with my WiFi - WPA2_PSK network ( BEFORE flashing it i was able to scan the network propely , so the module works ).
First i used "esp8266_flasher" and the "v0.9.2.2 AT Firmware" , but it doesn't seems to change anything! , Then i used ESPFlashDownloadTool ( v3.6.2.2 ) with "AT 1.5.1 Firmware" and when i run the code below i got this response ( and of course the AT command doens't work ).
I've seen that "rst cause:1" is an normal boot anomaly or something like that , but i'm still confused.

[image]

The pins are : 

TX->6
RX ->7
CH_PD+VCC->3.3v
GND->GND ( with GPIO0 when i'm flashing it )

Somebody can help me ? I'm simply frustrating ( ask me for more information ( if i know what we are talking about ) if you need! )

### Settings in IDE

Module:  	Esp8266 12e

### Sketch

```cpp

#include <SoftwareSerial.h>
SoftwareSerial sw (6,7);

void setup() {

  sw.begin(74880);
  Serial.begin(9600);

}
void loop() {

  if(Serial.available()) 
  {  
      sw.write(Serial.read());
       
  }
  
  if(sw.available()) 
  {
    Serial.write(sw.read());
  }
  
}
```

### Debug Messages

 ets Jan  8 2013,rst cause:1, boot mode:(3,4)

load 0x40100000, len 816, room 16 
tail 0
chksum 0x8d
load 0x3ffe8000, len 788, room 8 
tail 12
m
ioofl 2lm
fde  e 
M 
 Sbmnrra e

dragondaud, 2018-01-06:
The issue board is not a general help forum, it is for reporting specific reproducible errors with the project. I would suggest closing your issue, and posting over at: http://www.esp8266.com/viewforum.php?f=25

devyte, 2018-01-06:
@PunisherX99 I'm a bit confused. Your description talks about the AT cmd firmware, which is completely unrelated to this repo, but then you post an example sketch.
Anyways, assuming that what you said about the AT cmd firmware is irrelevant, I think that you are using forbidden pins:
SoftwareSerial sw (6,7);
I believe that GPIO6 and 7 are used by the SPI.
In addition, it is recommended to erase the entire flash before uploading a new sketch (e.g.: with esptool).
Closing as user error.

closed 2018-01-06 by devyte

PunisherX99, 2018-01-06:
Where i was even talk about the AT cmd firmware ???

PunisherX99, 2018-01-06:
@devyte Where i was even talk about the AT cmd firmware ?

devyte, 2018-01-06:
@PunisherX99 said:

> i get that weird responde AFTER flashing it to AT 1.5.1

and

> First i used "esp8266_flasher" and the "v0.9.2.2 AT Firmware" , but it doesn't seems to change anything! , Then i used ESPFlashDownloadTool ( v3.6.2.2 ) with "AT 1.5.1 Firmware" 

Anyways, change your gpio pins.

PunisherX99, 2018-01-07:
@devyte I TRIED to change the rx/tx pins to 2/3 , nothing changes , that error continues to comes up.
The problem is that i didn't find ANYTHING on internet , everyone has "rst cause:2" instead of "rst cause:1".

Pablo2048, 2018-01-07:
@PunisherX99 well You definitely did something wrong (so Your ANYTHING is NONSENSE) - try https://espressif.com/sites/default/files/documentation/esp8266_reset_causes_and_common_fatal_exception_causes_en.pdf (4-th link from google for search "esp8266 rst cause")

PunisherX99, 2018-01-07:
I already saw that the"error" is a message that esp sends to me when i manually power it.
The problem is that after that message the module don't respond at all when i send any AT commands.

Pablo2048, 2018-01-07:
You are a newguy with ESP right? If so please go to esp8266.com and ask your questions there (as @dragondaud already suggested).

PunisherX99, 2018-01-08:
I think the problem is that i tried to boot a > 512KB firmware when my flash size of my esp8266 ( 01 not 12e ) is 4Mb ( 512 KB ).
I flash to v1.5.0 and now it work well.
Thanks to everyone!

Pablo2048, 2018-01-08:
The main problem is that You are complete off-topic here. You have mixed two things together - Arduino and Core for ESP8266. This repository is for Arduino CORE running INSIDE ESP8266, but You have IMHO Arduino BOARD with AVR as a main CPU and ESP8266 module with AT firmware as a slave, connected via AVR D6 and D7 GPIO.

lrmoreno007, 2018-01-08:
[image]

This is the hardware you are flashing ---> Arduino Uno. 

**This framework is for flashing ESP8266 hardware with Arduino FIRMWARE (software) directly**, without any Arduino hardware.

If you want to flash the ESP8266 with a new AT Firmware ( Is this what you want?) and Arduino UNO like FTDI, you must search in google "Arduino Uno as FTDI ESP8266". But it is completely unrelated here, ask in esp8266.com.

Good luck.