esp8266/Arduino #4099: FSWebServerLib /  ESPAsyncTCP-master will not compile in v2.4.0

bill-orange, 2018-01-05:
Hardware:			Wemos D1
Core Version:      	2.4.0

Problem description:

The fairly popular FSWebServerLib which uses ESPAsyncTCP and ESPAsyncWebServer will not compile using v2.4.0.  There were no similar issues in v2.3.0.  I used the example UserConfigExample.ino as a test when my sketch would not compile.

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		unknown

Sketch: UserConfigExample.ino

Compiler Message:

```
netif* interface = ip_route(&addr);

                                    ^
C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]

```

There were many other messages but the first one appears to be key.

A workaround would be fine for now.

rick-bp, 2018-01-06:
It is caused by lwip 2.0 upgrade.
Up to 2.4 RC2 it is working with lwip 1.4.
I think ESPAsyncTCP must be changed to work with new lwip standards.

devyte, 2018-01-06:
This is odd, my app makes use of ESPAsyncTCP and ESPAsyncUDP (both latest git), and it builds fine with latest git of this repo.
Then again, my install has been around for a while...

rick-bp, 2018-01-08:
@devyte 
To me, ESPAsyncTCP and ESPAsyncUDP, worked up to 2.4 RC2.
Now with the final release 2.4 doesn't work.

d-a-v, 2018-01-10:
I just tried with latest core's master with latest ESPAsyncTCP's master with its SyncClient example.
There is not a single warning during compilation and it works as expected.

bill-orange, 2018-01-10:
@d-a-v ,

I re-test and got the same negative result, using the SyncClient example.  I am running 1.8.4 for the IDE with v2.4.0 voard manager.  On this test, I used the ESP8266 generic board settings.  The errors start with this:

> C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:113:36: error: macro "ip_route" requires 2 arguments, but only 1 given
> 
>    netif* interface = ip_route(&addr);
>                                     ^
> C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
> 
> C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
> 
>      tcp_recv(_pcb, &_s_recv);
>                             ^
> In file included from C:\Users\William\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:
> 

There must be some difference in our environment.  Perhaps something in:

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

```
Your thoughts?

devyte, 2018-01-10:
It's possible that my build env is similar to @d-a-v 's, because I'm building a custom lwip from the sources.

d-a-v, 2018-01-10:
Your line 113 is line 115 in the repository ([link](https://github.com/me-no-dev/ESPAsyncTCP/blob/master/src/ESPAsyncTCP.cpp#L115))
> ```netif* interface = ip_route(&addr);```

If you don't have the ```#if LWIP_VERSION_MAJOR == 1``` just above this line then your version is not up to date (but will compile with lwip v1.4 though).

I was wrong saying I have not a single warning, in fact there are some. It works as expected anyway.

d-a-v, 2018-01-10:
@bill-orange I did not see you state it is neither working with v1.4. So that is strange.

As suggested by @devyte I tried without any custom modification (empty ```git status``` result) and it still works with both v1.4 and v2.

d-a-v, 2018-01-10:
I will try with an installation via the board manager.

bill-orange, 2018-01-10:
@d-a-v 

You are correct! I updated ESPAsyncTCP manually (copy and paste) from the May 2017 version in my libraries folder to the Nov 2017 version in github and now I compile the SyncClient example fine.  I presumed, apparently incorrectly that the library manager would keep these libraries up to date by offering me the updates. Apparently not.

Thanks

devyte, 2018-01-10:
@bill-orange The library manager provides updates based on releases, not gihub. That's why I specified:

> ESPAsyncTCP and ESPAsyncUDP (both latest git)

I'm glad this is resolved.
Closing.

closed 2018-01-10 by devyte