esp8266/Arduino #4100: ESP8266 after restart or power on/off.  does not worked

karenYerevan, 2018-01-05:
hi there
i have ESP8266-12, it is have other power
I flashing esp8266 from arduino uno
Flashing programm (XTCOM_UTIL)
aruduino uno -- esp8266
RX -> RXD
TX -> TXD
GND -> GND,GPIO0

After flashing all AT commands worck good.
But when esp8266 is restart (AT+RST) , AT commands stopping works
and show this text

> AT
> OK
> 
> AT+RST
> OK
> 
> ets Jan 8 2013,rst cause:4, boot mode:(1,7)
> wdt reset

[image]

devyte, 2018-01-06:
@karenYerevan you were already advised in #4087 that this is not the AT command firmware repo. The code here has nothing to do with the AT command set.
Closing.

closed 2018-01-06 by devyte

karenYerevan, 2018-01-06:
what's the difference, any sketch I fill is the same, well I'll open it yet but there will not be written about at. and i hope you will not close my issue

devyte, 2018-01-06:
@karenYerevan the difference is that this is an issue tracker for issues in our Arduino core libs. We're not a general support forum, and this is not the right place to request help for your particular project. Please read our issue POLICY doc.