esp8266/Arduino #4101: WiFi not reconnecting after connection lost in lwip 2

EricMc1289, 2018-01-05:
**Basic Infos**
Hello everyone. I have a very strange problem since 2.4.0.
Hardware
Hardware:	Esp8266 12f and Esp8266 -01

**Description**
 If the ESP looses the connection to the router (just disable your network for 30 seconds and reenable again at the router) the ESP won't connect to the wifi again.  
It is a very strange behaviour, because the function wifi_station_get_connect_status() return only 0 then when the WiFi.begin was called 1 and after sometime 0 again. And now it gets even more funny when the ESP stucks at 1 my Router a Fritzbox 7490 lists the device as connected and you can access the httpserver on the ESP but only until it goes back to 0 again. 

**Settings in IDE**
lwip v2 prebuilt MSS=536

Sketch
/*
 *  This sketch trys to Connect to the best AP based on a given list
 *
 */

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() {
    Serial.begin(115200);
    delay(10);
	
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

	Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop() {
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        Serial.println(wifi_station_get_connect_status());
        delay(1000);
    }
}

jp112sdl, 2018-01-06:
I've got a similar behavior.

The ESP **does** reconnect to the WiFi network - I can successful ping the device, after the AP has gone and came back again.

But the state of WiFi.status() isn't getting to WL_CONNECTED again, so that my HTTP GET Requests always fail.

devyte, 2018-01-06:
@EricMc1289 @jp112sdl Please test the referenced PR.

jp112sdl, 2018-01-06:
@devyte Result commented in [#4060](https://github.com/esp8266/Arduino/issues/4060#issuecomment-355761883)

EricMc1289, 2018-01-06:
@devyte For everyone else with this issue. The solution above works so far, so I will close this issue. I will test this software the next couple of days and report any problems in this thread

closed 2018-01-06 by EricMc1289

d-a-v, 2018-01-06:
Thanks for testing