esp8266/Arduino #4103: connectino refused  Basic http client

riker65, 2018-01-06:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.3.0?

### Description
using the example basichttpclient.ino
I always get a connection refused, why? 

Problem description

### Settings in IDE

Module:  			basic http client example
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

#include <wifi_tr.h>

void setup() {

    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFi.mode(WIFI_STA);
    WiFiMulti.addAP(ssid, password);
Serial.setDebugOutput(true);
}

void loop() {
    // wait for WiFi connection
    Serial.println(WiFi.localIP());
    WiFi.printDiag(Serial);
    
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        http.begin("http://192.168.0.168/test.html"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}

### Debug Messages
how do I get debug mesages?
```
messages here
```

Auto connect: 1
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
192.168.0.168
Mode: STA
PHY mode: N
Channel: 6
AP id: 0
Status: 5
Auto connect: 1
SSID (6): TR7272
Passphrase (16): xxx
BSSID set: 1

yoursunny, 2018-01-11:
Do you have an HTTP server running at `192.168.0.168`?
Can you execute `curl -v http://192.168.0.168/test.html` on your computer and see the results?

riker65, 2018-01-11:
Hi
you were right.
thanks for your hint
The example does not have a running webserver ....

closed 2018-01-11 by riker65

manish-hello-gupta, 2020-03-09:
Hi
i am encountering the same problem 
[HTTP] GET... failed, error: connection refused
and my "curl -v http://localhost/testcode/show1.php" is running  in the cmd

d-a-v, 2020-03-09:
`curl http://localhost/...` cannot be a request to your esp.
Try with `curl http://<ip address of the esp>/...`.

This is a 2 years old closed issue. Please open a new one with your details.

manish-hello-gupta, 2020-03-10:
sorry and thanks to guide me.