esp8266/Arduino #4106: Sketch using Timezone-Lib will not compile in v2.4.0

digi-thomas2003, 2018-01-06:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0
Core Version:      	2.4.0

### Description

Any sketch using the Timezone-library from JChristensen will not compile with 2.4.0
There are lots of "... is not declared in this scope" errors referencing to the Timezone lib

Going back to 2.3.0 the compilations run fine without any error.

Can this please be checked? Thanks a lot.

Best regards

5chufti, 2018-01-06:
most likely a problem with your includes (sequence of).
without your sketch not reproducable
BUT: why use third party TZlib when newlibc allready supports TZ stuff?

digi-thomas2003, 2018-01-06:
Hello Schufti (and d-a-v),

is there a kind of documentation about the newlibc and the supported time, TZ and DST functions?
Sorry, if this may be a silly question.

Thank you.

5chufti, 2018-01-07:
one can find the newlib documentation here
https://sourceware.org/newlib/libc.html

so since your - not core related - problem is solved, may you please close this issue?

digi-thomas2003, 2018-01-07:
as desired

closed 2018-01-07 by digi-thomas2003

digi-thomas2003, 2018-01-10:
#4122