esp8266/Arduino #4109: v2.4.0 - ESP8266WebServer  , Wish option that we  can change port  in setup()

TridentTD, 2018-01-07:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos
Wish option that we  can change port  in setup().

#### Hardware
Hardware:			Wemos D1 Mini
Core Version:      	2.1.0-rc2

### Description

I try to create  ESP8266WebServer and use  config structure.
In my config structure  has  web_port for config  ESP8266WebServer's port.

How I can  change port  of ESP8266WebServer in setup() ?

### Settings in IDE

Module:  			Wemos D1 Mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer webserver(80);

struct  __config {
   uint8_t ssid[32];
   uint8_t passwod[64];
   uint16_t  web_port;
} config;

void setup() {
        sprintf( (char*) config.ssid , "%s",  "-----SSID-----");
        sprintf( (char*) config.password, "%s",  "-----PASSWORD-----");
        web_port = (uint16_t) 8080;

        Serial.begin(115200);
        WiFi.begin((char*) config.ssid, (char*) config.password);
        while(!WiFi.isConnected()) delay(400);
        Serial.printf("WiFi connected , IP : %s\r\n", WiFi.localIP().toString().c_str());

        //  At here I wish to change webserver 's port  
        // by config after declare ESP8266WebServer's object.
        //webserver( config.web_port);   

        webserver.on("/", [](){  webserver.send(200, "text/html", "Hello Webserver"); });
        webserver.begin();
}

void loop() {
        webserver.handleClient();
}
```

### Debug Messages

```

```

caverna, 2018-01-07:
Man, does this line gives you any clue? ;-)
ESP8266WebServer webserver(**_80_**);

igrr, 2018-01-08:
@caverna 

> here I wish to change webserver 's port by config **after declare** ESP8266WebServer's object.

(emphasis mine)

I think this is a valid request, perhaps we can fix the issue with assignment between ESP8266WebServer objects to support it.

igrr, 2018-01-08:
Closing as same basic issue as #4085.
@TridentTD please follow to the linked issue and click "Subscribe" to receive notifications.

closed 2018-01-08 by igrr

TridentTD, 2018-01-09:
Thank you @igrr