esp8266/Arduino #4111: SPIFFS fails to work

weswitt, 2018-01-07:
Using the current releases of everything I cannot get a simple sketch that reads from SPIFFS to work.

Using the IDE I simply compile & upload the sketch and then using the IDE I upload the SPIFFS image. When the sketch runs it finds NO file in the file system. This seems very straight forward but I cannot make this work. I have verified that a sketch can successfully write and read to the fs so it seems that the SPIFFS upload is broken or for some reason the core implementation cannot read from an uploaded SPIFFS. Any help would be appreciated.

Sketch:
#include "FS.h"

void setup()
{
    Serial.begin(115200);
    delay(5000);
    SPIFFS.begin();
    String str = "";
    Serial.println("FILES:");
    Dir dir = SPIFFS.openDir("/");
    while (dir.next())
    {
        str += dir.fileName();
        str += " / ";
        str += dir.fileSize();
        str += "\r\n";
    }
    Serial.print(str);
    Serial.println("==>***************");
}
void loop()
{
}

Arduino build log:
D:\dev\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware D:\dev\arduino-1.8.5\hardware -hardware C:\Users\wesw\AppData\Local\Arduino15\packages -tools D:\dev\arduino-1.8.5\tools-builder -tools D:\dev\arduino-1.8.5\hardware\tools\avr -tools C:\Users\wesw\AppData\Local\Arduino15\packages -built-in-libraries D:\dev\arduino-1.8.5\libraries -libraries C:\Users\wesw\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=1M64,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\wesw\AppData\Local\Temp\arduino_build_490613 -warnings=none -build-cache C:\Users\wesw\AppData\Local\Temp\arduino_cache_124339 -prefs=build.warn_data_percentage=75 -verbose C:\temp\test\test.ino
D:\dev\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware D:\dev\arduino-1.8.5\hardware -hardware C:\Users\wesw\AppData\Local\Arduino15\packages -tools D:\dev\arduino-1.8.5\tools-builder -tools D:\dev\arduino-1.8.5\hardware\tools\avr -tools C:\Users\wesw\AppData\Local\Arduino15\packages -built-in-libraries D:\dev\arduino-1.8.5\libraries -libraries C:\Users\wesw\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=1M64,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\wesw\AppData\Local\Temp\arduino_build_490613 -warnings=none -build-cache C:\Users\wesw\AppData\Local\Temp\arduino_cache_124339 -prefs=build.warn_data_percentage=75 -verbose C:\temp\test\test.ino
Using board 'generic' from platform in folder: D:\dev\arduino-1.8.5\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: D:\dev\arduino-1.8.5\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\Users\wesw\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\wesw\AppData\Local\Temp\arduino_build_490613/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\variants\generic" "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\sketch\test.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\wesw\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\wesw\AppData\Local\Temp\arduino_build_490613/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\variants\generic" "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\sketch\test.ino.cpp" -o "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\dev\arduino-1.8.5\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\wesw\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\wesw\AppData\Local\Temp\arduino_build_490613/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\variants\generic" "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\sketch\test.ino.cpp" -o "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\sketch\test.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Users\wesw\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\wesw\AppData\Local\Temp\arduino_build_490613/core" -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\variants\generic" "D:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\cont.S" -o "C:\Users\wesw\AppData\Local\Temp\arduino_build_490613\core\cont.S.o"
"C:\Users\wesw\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/lwip2/include" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\wesw\AppData\Local\Temp\arduino_build_490613/core" -c -w -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266" "-ID:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\variants\generic" "D:\dev\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266\cont_util.c" -o "C:\Users\wesw\AppData\Local\Temp\arduino_bu [... cut ...]
[SPIFFS] data   : C:\temp\test\data
[SPIFFS] size   : 64
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/homie/config.json

[SPIFFS] upload : C:\Users\wesw\AppData\Local\Temp\arduino_build_490613/test.spiffs.bin
[SPIFFS] address: 0xEB000
[SPIFFS] reset  : ck
[SPIFFS] port   : COM3
[SPIFFS] speed  : 115200

Uploading 65536 bytes from C:\Users\wesw\AppData\Local\Temp\arduino_build_490613/test.spiffs.bin to flash at 0x000EB000

................................................................                 [ 100% ]

weswitt, 2018-01-07:
Further analysis shows that the core runtime may be corrupting the FS. If I upload the FS bin and then immediately (not allowing the sketch to run) download the SPIFFS using read_flash I can see that the FS is fine. What is uploaded looks like what was downloaded. If I then let the sketch run and again download the FS I can see that it is completely different. All the files are gone.

devyte, 2018-01-07:
@weswitt this is starting to sound like a format() gone wrong

weswitt, 2018-01-07:
@devyte that's what I thought too, but I rebuilt everything with SPIFFS_READ_ONLY enabled so that format can never happen. If I run a simple sketch that never calls SPIFFS.begin then the flash area for the SPIFFS remains as it was when uploaded. If the sketch just calls SPIFFS.begin then the SPIFFS region gets hosed.

I also added some debug prints in spiffs_api to notify when format happens and those never appear so it doesn't seem like a format issue.

**Debug Output:**
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=-10025
ip:192.168.1.133,mask:255.255.255.0,gw:192.168.1.1
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0

weswitt, 2018-01-07:
After enabling more diagnostics in SPIFFS is see the output below. What is happening is that spiffs_obj_lu_scan is returning SPIFFS_ERR_NOT_A_FS when called from SPIFFS_mount. Looks like SPIFFS does not like the data that is laid down in the flash region. I suppose this point to the tool creating the SPIFFS image. 

And the reason that I was seeing a changed FS is because the mount path will format the FS when the mount fails. In my view this is really bad API semantics. The mount API should simply return an error code and allow the caller to deal with it. Performing a destructive operation like this is really bad. I cannot think of ANY operating system that does this or has done this. Linux, NO. Windows, NO. MacOS, NO.

SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFS_mount  sz:0 logpgsz:256 logblksz:0 perasz:0 addr:00000000 fdsz:240 cachesz:1400
SPIFFSImpl: mount rc=-10025
SPIFFS_format: rc=-10032, err=-10025

weswitt, 2018-01-07:
OK, I rebuilt SPIFFS with SPIFFS_USE_MAGIC set to zero to disable the FS integrity checks. Sure enough the FS now mounts. It seems that MKSPIFFS and the Arduino IDE both are creating filesystems with too many unerased blocks -- in my case there are 16 of them.

Anyone have any insight on this?

**Debug Output from Mount:**
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFS_mount  sz:0 logpgsz:256 logblksz:0 perasz:0 addr:00000000 fdsz:240 cachesz:1400
CACHE_ALLO: allocated cache page 0 for pix 0000
CACHE_ALLO: allocated cache page 1 for pix 0010
CACHE_ALLO: allocated cache page 2 for pix 0020
CACHE_ALLO: allocated cache page 3 for pix 0030
CACHE_ALLO: allocated cache page 4 for pix 0040
CACHE_FREE: free cache page 0 pix 0000
CACHE_ALLO: allocated cache page 0 for pix 0050
CACHE_FREE: free cache page 1 pix 0010
CACHE_ALLO: allocated cache page ip:192.168.1.133,mask:255.255.255.0,gw:192.168.1.1
1 for pix 0060
CACHE_FREE: free cache page 2 pix 0020
CACHE_ALLO: allocated cache page 2 for pix 0070
CACHE_FREE: free cache page 3 pix 0030
CACHE_ALLO: allocated cache page 3 for pix 0080
CACHE_FREE: free cache page 4 pix 0040
CACHE_ALLO: allocated cache page 4 for pix 0090
CACHE_FREE: free cache page 0 pix 0050
CACHE_ALLO: allocated cache page 0 for pix 00a0
CACHE_FREE: free cache p0
CACHE_FREE: free cache page 1 pix 0010
CACHE_ALLO: allocated cache page ip:192.168.1.133,mask:255.255.255.0,gw:192.168.1.1
1 for pix 0060
CACHE_FREE: free cache page 2 pix 0020
CACHE_ALLO: allocated cache page 2 for pix 0070
CACHE_FREE: free cache page 3 pix 0030
CACHE_ALLO: allocated cache page 3 for pix 0080
CACHE_FREE: free cache page 4 pix 0040
CACHE_ALLO: allocated cache page 4 for pix 0090
CACHE_FREE: free cache page 0 pix 0050
CACHE_ALLO: allocated cache page 0 for pix 00a0
CACHE_FREE: free cache page 1 pix 0060
CACHE_ALLO: allocated cache page 1 for pix 00b0
CACHE_FREE: free cache page 2 pix 0070
CACHE_ALLO: allocated cache page 2 for pix 00c0
CACHE_FREE: free cache page 3 pix 0080
CACHE_ALLO: allocated cache page 3 for pix 00d0
CACHE_FREE: free cache page 4 pix 0090
CACHE_ALLO: allocated cache page 4 for pix 00e0
CACHE_FREE: free cache page 0 pix 00a0
CACHE_ALLO: allocated cache page 0 for pix 00f0
page index byte len:         256
object lookup pages:         1
page pages per block:        16
page age 1 pix 0060
CACHE_ALLO: allocated cache page 1 for pix 00b0
CACHE_FREE: free cache page 2 pix 0070
CACHE_ALLO: allocated cache page 2 for pix 00c0
CACHE_FREE: free cache page 3 pix 0080
CACHE_ALLO: allocated cache page 3 for pix 00d0
CACHE_FREE: free cache page 4 pix 0090
CACHE_ALLO: allocated cache page 4 for pix 00e0
CACHE_FREE: free cache page 0 pix 00a0
CACHE_ALLO: allocated cache page 0 for pix 00f0
page index byte len:         256
object lookup pages:         1
page pages per block:        16
page header length:          5
object header index entries: 105
object index entries:        124
available file descriptors:  5
free blocks:                 15
SPIFFSImpl: mount rc=0
[... 53 lines cut ...]
CACHE_ALLO: allocated cache page 1 for pix 00a0
CACHE_FREE: free cache page 2 pix 0060
CACHE_ALLO: allocated cache page 2 for pix 00b0
CACHE_FREE: free cache page 3 pix 0070
CACHE_ALLO: allocated cache page 3 for pix 00c0
CACHE_FREE: free cache page 4 pix 0080
CACHE_ALLO: allocated cache page 4 for pix 00d0
CACHE_FREE: free cache page 0 pix 0090
CACHE_ALLO: allocated cache page 0 for pix 00e0E: free cache page 0 pix 0040
CACHE_ALLO: allocated cache page 0 for pix 0090
CACHE_FREE: free cache page 1 pix 0050
CACHE_ALLO: allocated cache page 1 for pix 00a0
CACHE_FREE: free cache page 2 pix 0060
CACHE_ALLO: allocated cache page 2 for pix 00b0
CACHE_FREE: free cache page 3 pix 0070
CACHE_ALLO: allocated cache page 3 for pix 00c0
CACHE_FREE: free cache page 4 pix 0080
CACHE_ALLO: allocated cache page 4 for pix 00d0
CACHE_FREE: free cache page 0 pix 0090
CACHE_ALLO: allocated cache page 0 for pix 00e0
CACHE_FREE: free cache page 1 pix 00a0
CACHE_ALLO: allocated cache page 1 for pix 00f0
/test.txt / 69
==>***************
SPIFFS_closedir

igrr, 2018-01-08:
@weswitt you have mentioned "current releases of everything". Just to clarify, does that mean that you are installing version 2.4.0 using boards manager? Or using git version of the core and installing tools (including mkspiffs) using get.py?

I'm asking because this issue looks like a mismatch between SPIFFS configuration in mkspiffs and in the sketch. This could be due to incompatible versions of the core and mkspiffs. However I have just tested FSBrowser example with the latest git version (which includes `mkspiffs-0.2.0-no_magic_length`), and it worked as expected.

weswitt, 2018-01-08:
I've built MKSPIFFS using the master repo so it is as current as possible. And I'm using 2.4.0 release of core, not the git master branch version.

weswitt, 2018-01-08:
MKSPIFFS and Arduino cores both use 0.3.7 of SPIFFS so there should not be any mismatch.

igrr, 2018-01-08:
2.4.0 is likely not compatible with latest mkspiffs, it uses mkspiffs-0.2.0-no_magic_length release. Could you please try building the filesystem using mkspiffs version which is installed together with the core?

If building mkspiffs manually, make sure you set build flags to match SPIFFS config in esp8266 core:

`make dist BUILD_CONFIG_NAME="-no_magic_length" CFLAGS="-DSPIFFS_USE_MAGIC_LENGTH=0 -DSPIFFS_ALIGNED_OBJECT_INDEX_TABLES=1"`

igrr, 2018-01-08:
Regarding mismatch: indeed, SPIFFS versions are likely the same. However SPIFFS configuration in [spiffs_config.h](https://github.com/igrr/mkspiffs/blob/master/include/spiffs_config.h) has recently been changed on mkspiffs side. At the moment it does not match [spiffs_config.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/spiffs/spiffs_config.h) in ESP8266 arduino core. Therefore the build script used to produce releases for ESP8266 modifies some of the flags from their default values (as shown in my above comment).

jamesarm97, 2018-01-08:
Would this have any bearing on an issue I am having trying to use the eeprom.read/write api? I used the CheckFlashConfig sample and added a print for all the spiffs.info data and it is returning an error and bad data for the totalBytes and usedBytes which is throwing off other functions:

SPIFFS_info: rc=-10024, err=-10024
TB: **4277137406**
UB: **4277137406**
BS: 8192
PS: 256
MO: 5
MP: 32
Flash real id: 001840C8
Flash real size: 16777216

Flash ide size: 16777216
Flash ide speed: 40000000
Flash ide mode: DIO
Flash Chip configuration ok.

snippet:
fs::FSInfo info;
SPIFFS.info(info);
Serial.printf("TB: %u\r\n",info.totalBytes);
Serial.printf("UB: %u\r\n",info.usedBytes);
Serial.printf("BS: %u\r\n",info.blockSize);
Serial.printf("PS: %u\r\n",info.pageSize);
Serial.printf("MO: %u\r\n",info.maxOpenFiles);
Serial.printf("MP: %u\r\n",info.maxPathLength);

jamesarm97, 2018-01-08:
never mind, wasn't calling SPIFFS.begin.
SPIFFS Opened
TB: 14900866
UB: 502
BS: 8192
PS: 256
MO: 5
MP: 32
Flash real id:   001840C8
Flash real size: 16777216

Flash ide  size: 16777216
Flash ide speed: 40000000

weswitt, 2018-01-10:
I resolved this. I built MKSPIFFS with SPIFFS_USE_MAGIC_LENGTH=0 and SPIFFS_ALIGNED_OBJECT_INDEX_TABLES=1 and it now works.

closed 2018-01-10 by weswitt

omerfraz, 2019-11-16:
Arduino IDE > Tools > Board Selected as ESP8266 (your version) > SPIFFS (4M 1M SPIFF) atleast