esp8266/Arduino #4114: PR #3860 broke DHCP client Wifi config

tve, 2018-01-07:
### Basic Infos

#### Hardware
Hardware:		ESP12
Core Version:      	master

### Description

As far as I can tell and experience, the change in #3860 broke DHCP. In order to get the dhcp client started one has to set the subnet to 0.0.0.0 in ESP8266WiFiSTAClass::config to pass this check: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L242
and the new code (https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L221-L235) makes it impossible to get subnet==0.

devyte, 2018-01-07:
@tve please provide a minimal example sketch of what doesn't work, and what does or used to.

tve, 2018-01-07:
I don't have a simple sketch, alas. I think it's pretty obvious from the code. The only place that _isStaticIp is set to false is line 243. That is guarded by the coditional at line 242:
```
if (local_ip == 0U && gateway == 0U && subnet == 0U) {
```
The new code's if statement body lines 222-235 is triggered if subnet is 0:
```
if(subnet[0] != 255)
    {
      //octet is not 255 => interpret as Arduino order

      if(dns1[0] == 0)
      {
        //arg order is arduino and 4th arg not given => assign it arduino default
        dns1 = IPAddress(255,255,255,0);
      }

      //current order is arduino:                   ip-dns-gway-subnet
      swap(gateway, subnet); //after this, order is ip-gway-dns-subnet
      swap(subnet, dns1);    //after this, order is ip-gway-subnet-dns (correct ESP order)
    } 
```
Exiting this if statement, it is impossible to get subnet==0, thus it is impossible to set _useStaticIp=false
This may not be noticed by most people because the default at reset is _useStaticIp=false, so it's only if config(0,0,0) is called that it doesn't work.

devyte, 2018-01-07:
@tve your explanation served instead of the sketch to understand the problem, thanks.
I'll catch the case when all are 0U and handle it separately.

devyte, 2018-01-12:
@tve please test PR #4145 .

devyte, 2018-01-17:
Fixed via #4145 .

closed 2018-01-17 by devyte