esp8266/Arduino #4115: HttpUpload crashes since version 2.4.0

drschlaumeier, 2018-01-07:
NodeMCU 1.0 (ESP12-E), 4M (1M SPIFFS)

Hi, I'm using ESP8266WebServer and http POST to update my firmware and my spiffs files. 
It took a while to find the problem: Since update to 2.4.0 its crashing in **Parsing.cpp** in **_parseForm** in **_currentUpload.reset(new HTTPUpload())**:

```
...
          } else {
	     _currentUpload.reset(new HTTPUpload());
            _currentUpload->status = UPLOAD_FILE_START;
            _currentUpload->name = argName;
            _currentUpload->filename = argFilename;
            _currentUpload->type = argType;
            _currentUpload->totalSize = 0;
            _currentUpload->currentSize = 0;
...
```
Obviously, the std::unique_ptr reset method is not working. I dont know why.
According to Parsing.cpp history and to #3721 the code has been massively changed for dynamic allocation. But it seems not to work for me.
All was working in 2.3.0.

Please help to fix.

Here the crash dump from Exception Decoder .. which was not very helpfull:

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 13 results
0x4022d5e6: memset at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S line 133
0x4021f160: ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/Parsing.cpp line 456
0x40224f9f: String::changeBuffer(unsigned int) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 545
0x4022501d: String::copy(char const*, unsigned int) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 545
0x40225120: String::operator=(String const&) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/WString.cpp line 545
0x4021fec1: ESP8266WebServer::_parseRequest(WiFiClient&) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/Parsing.cpp line 213
0x4021d8f0: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40225a2c: esp_yield at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 87
0x4021e723: ESP8266WebServer::handleClient() at C:\Users\DRS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214

```

drschlaumeier, 2018-01-09:
Hey girls/guys,
thanks for leaving me alone with that issue. It took me some more intensive research but I believe that the problem is the heap size which is only "Free heap: 2992 bytes" before executing the update loop. See also here #3031 . So I need to restructure my code to free some heap. Any other idea?

Rgds
DRS

devyte, 2018-01-09:
@drschlaumeier and those 3K left were likely fragmented.
You are right, you need to fix your application to improve heap. Instead of keeping  large structures in memory, keep what you can in a File on spiffs. Use .reserve() on Strings once before a series of concats,  and don't do long series of + ops. If you have const char * strings, put them in flash (use PROGMEM or F).
Closing as user error.

closed 2018-01-09 by devyte

d-a-v, 2018-01-09:
And there is a new debug option called OOM that may help in case of short memory.

drschlaumeier, 2018-01-09:
Great, many thanks both for the hints. 
I'm currently applying PROGMEM & F. 
The .reserve() is new for me and OOM as well. 
So, I will sit down now and get started :-)