esp8266/Arduino #4116: ESP is only reachable from one device

arget888, 2018-01-07:
Hi,
I have implemented a small webservice to controll a temperature and humidty sensor and a strip of ws2812b leds. I connect the ESP to my router. Now it starts to get strange. I receive an IP Adresse of the DHCP server but the DHCP server does not show the connected device (all other connected device are displayed like my iPad or my iPhone or my Mac). Over the Serial Port i can read the IP-Address. It is 10.0.0.48 with is in range of dhcp service and all settings like dns, subnet, gateway are also correct. On my iMac i can connect to my webservice but not with all other devices. 

This is how i start the WLAN connection and how i handle webservices
```arduino
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  int status = WiFi.waitForConnectResult();

  int status = WiFi.waitForConnectResult();
  if(status != WL_CONNECTED)
  {
      //do Something
  }

  server.on("/temperature", getTemperature);
  server.begin();
```

What can i do that the esp is reachable from other devices in network?

devyte, 2018-01-07:
@arget888 when you opened the issue, you were presented with an issue template requesting certain information. You have ignored most of it. Please edit your post and add what was requested, or close this issue and correctly open a new one.

devyte, 2018-01-07:
Also, make sure to include all relevant info, including a complete and minimal sketch, which the above is not.

arget888, 2018-01-07:
Updated, i hope all needed infos are given

d-a-v, 2018-01-08:
Can you try the included example sketch HelloServer (File>Examples>ESP8266WebServer>HelloServer) and see if it works better with your router ?
Can you try change your lines around WL_CONNECTED according to this example ?
The core version is showed where you got the "latest code version installed yesterday".
Are you sure you don't have two dhcp servers on your network ? Usually dhcp servers addresses end with .1 or .254, you have .138 which is perfectly legal but unusual.

arget888, 2018-01-09:
I flashed the HelloServer Example, the result is the same. I know that .128 is not ideal for a router but that was the standard setting from my providers router and i have no need to change that. I also disconnected all other devices from my router (all devices where added by cabel) but the result is the same. The mdns service is also not working, on my iMac i can call the HelloServer site by IP - Adress but not by the name ("esp8266.local").
The version of my library says 1.0.0!

All of my devices have same settings with ip-range, subnet, router and dns server so there should be just one dhcp server running, that would be very confusing if two server has exact the same (unusual) dhcp settings and i did not have a problem with it until now with the esp

d-a-v, 2018-01-09:
Have you tried with another mDNS or bonjour browser that could scan your network ?
(an android app for example)

devyte, 2018-01-09:
@d-a-v are you thinking of the compressed dns thing?

d-a-v, 2018-01-10:
@devyte I don't know, it seems the opposite (here the "dns server" would be the esp).

@arget888 
> I used the ESP earlier as AP and connected from Mac, can this be a reason?

So your router used to show the ESP before that ?

Can you try with the latest version of esp8266/arduino core by removing it and reinstalling with git ([link](https://github.com/esp8266/Arduino#using-git-version)) ?

Pablo2048, 2018-01-10:
IMHO this is same problem which we have solved here https://github.com/me-no-dev/ESPAsyncWebServer/issues/259 - it seems like @arget888 has his MAC connected via wire and his AP has set AP isolation. @arget888 please try to connect your MAC via Wifi and try to access ESP server. If this fails you need to disable AP isolation in your WiFi AP.

devyte, 2018-07-13:
Closing due to lack of feedback and age.

closed 2018-07-13 by devyte