esp8266/Arduino #4120: Arduino IDE won't compile programs onto Wemos D1 mini lite anymore

spencerkraisler, 2018-01-09:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

I got the Arduino IDE all set up to upload programs onto my Wemos D1 mini lite. It was originally working. A few weeks go by, and suddenly the IDE is returning "error compiling for WEMOS D1 mini lite". It works with my Arduino Uno however. I tried reinstalling the driver, the hardware packages, and the board support, yet that didn't do anything. 

This kept showing up in the error message: "undefined reference to _sbrk_r" Maybe thats useful?
 
I'm happy to provide more information. 

### Settings in IDE

Module:  			?WEMOS D1 mini lite?
Flash Size: 		?1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?The board is connected to a microUSB.?
Reset Method:		??

### Sketch
 void setup() {
 pinMode(D4, OUTPUT);
}

void loop() {
  digitalWrite(D4, HIGH); 
  delay(1000);                     
  digitalWrite(D4, LOW);    
  delay(1000);                                         
}

5chufti, 2018-01-09:
either your core version is wrong or please try with the current release 2.4.0

spencerkraisler, 2018-01-10:
@5chufti So I went into terminal and deleted my hardware directory, and remade everything. I entered git clone https://github.com/esp8266/Arduino.git esp8266 and python get.py into their respective directories, yet nothing changed.

5chufti, 2018-01-10:
Hi,
maybe if you post full error message ...
as unfortunately your example compiles without any problem on my machine and core 2.4.0 ...

spencerkraisler, 2018-01-12:
@5chufti Archiving built core (caching) in: /var/folders/sk/42dp9gr11bl50fh_hwpqhtxw0000gn/T/arduino_cache_9077/core/core_esp8266com_esp8266_d1_mini_lite_CpuFrequency_80,UploadSpeed_115200,FlashSize_1M512_b00842aadc87696670563c83aa065f95.a
arduino.ar(heap.c.o): In function `xPortGetFreeHeapSize':
/Users/spencerkraisler/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c:12: multiple definition of `_malloc_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o):mallocr.c:(.text+0x218): first defined here
arduino.ar(heap.c.o): In function `vPortFree':
/Users/spencerkraisler/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c:18: multiple definition of `_free_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x0): first defined here
arduino.ar(heap.c.o): In function `pvPortRealloc':
/Users/spencerkraisler/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c:24: multiple definition of `_realloc_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-reallocr.o):mallocr.c:(.text+0x0): first defined here
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o): In function `malloc_extend_top':
mallocr.c:(.text+0x5e): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x127): undefined reference to `_sbrk_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):(.literal+0x18): undefined reference to `_sbrk_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o): In function `_malloc_trim_r':
mallocr.c:(.text+0x366): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x38f): undefined reference to `_sbrk_r'
/Users/spencerkraisler/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x3a0): more undefined references to `_sbrk_r' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board WEMOS D1 mini Lite.

Update: I created a new user on my MacBook Pro, and it uploads to the board just fine on the new user I made. This is very confusing though since I downloaded literally the same software.

5chufti, 2018-01-12:
probably there is some "residue" in your old instalation, pulling in old stuff leading to unresolved symbols. Usually if I encountered such problem I uninstalled arduino and deleted all leftover arduino related directories.
I then switched to only use "portable" installations where everything ends up in one place/directory, so is much more controllable.

spencerkraisler, 2018-01-13:
@5chufti I see. I do remember cleaning my Mac earlier, and I may have deleted some important files. I'll do a mass deletion of anything Arduino and just redownload the software, I suppose. Thank you for your help.

5chufti, 2018-01-13:
do follow the "portable" install route and just use the regular boardmanager way of installing esp-core (even 2.4.0 as current stable (?) version, if you feel adventurous).

kghill, 2018-02-11:
I am having a similar problem with compiling after updating to esp8266 2.4.0. The main error message I am getting is: file does not exist.
I tried going back to esp8266 2.3.0 and still having a compile version.  Version 2.3.0 was working before. I also tried updating the IDE to 1.8.5 and still the same problem.

I have included my directory tree because it looks different then the error message. The bin directory appears to be in the wrong place.

C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\bin. The in the bin directory is titled: as

Here is the error message I am getting:
Arduino: 1.8.5 (Windows 10), Board: "WeMos D1 mini Pro, 80 MHz, 16M (15M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keng\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keng\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keng\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:CpuFrequency=80,FlashSize=16M15M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\keng\AppData\Local\Temp\arduino_build_867863 -warnings=none -build-cache C:\Users\keng\AppData\Local\Temp\arduino_cache_89680 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\keng\Documents\Arduino\0 Garage Door Thermostat\0 Garage Door Thermostat\Garage_Door_Thermostat3.ino\Garage_Door_Thermostat3.ino.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keng\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keng\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keng\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini_pro:CpuFrequency=80,FlashSize=16M15M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\keng\AppData\Local\Temp\arduino_build_867863 -warnings=none -build-cache C:\Users\keng\AppData\Local\Temp\arduino_cache_89680 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\keng\Documents\Arduino\0 Garage Door Thermostat\0 Garage Door Thermostat\Garage_Door_Thermostat3.ino\Garage_Door_Thermostat3.ino.ino
Using board 'd1_mini_pro' from platform in folder: C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Using core 'esp8266' from platform in folder: C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Detecting libraries used...
"C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\keng\AppData\Local\Temp\arduino_build_867863/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINIPRO -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINIPRO"  -DESP8266 "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\d1_mini" "C:\Users\keng\AppData\Local\Temp\arduino_build_867863\sketch\Garage_Door_Thermostat3.ino.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\keng\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\keng\AppData\Local\Temp\arduino_b [... cut ...] rs\keng\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\d1_mini" "C:\Users\keng\AppData\Local\Temp\arduino_build_867863\sketch\Garage_Door_Thermostat3.ino.ino.cpp" -o "C:\Users\keng\AppData\Local\Temp\arduino_build_867863\preproc\ctags_target_for_gcc_minus_e.cpp"
exec: "C:\\Users\\keng\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board WeMos D1 mini Pro.

kghill, 2018-02-11:
I am not having compile errors now. What I did different is I opened my sketch from the IDE rather than clicking on the file in my folder and having windows launch the IDE.

Hope this helps.

kghill, 2018-02-13:
Further follow up: It looks like the IDE I am using which does not have compilie issues is pointing to my preference directory under my Windows documents section. The IDE which comes up when I click on the file is pointing to my preference directory  under my app data. 

I am wondering why I have two different directories for Arduino?

The location of the preference directory is found under the menu File. Preferences, at the bottom under under the heading "More preferences can be edited directly in the file.

devyte, 2018-02-15:
This is typical of a bad install, or messed up update. Please do a clean install from scratch, that includes deleting all old dirs related to Arduino before reinstalling.
Closing as can't reproduce.

closed 2018-02-15 by devyte

floppyoffritz, 2019-05-14:
Thankyou sooooo much, had the same issue for ages now.

JoaquinHoffmann, 2020-06-26:
> Further follow up: It looks like the IDE I am using which does not have compilie issues is pointing to my preference directory under my Windows documents section. The IDE which comes up when I click [...]

Thank you! I was trying to upload an examplewith AJAX  and have this strange problem, your method worked for me