esp8266/Arduino #4121: Exception (28) problem

pryxilon, 2018-01-09:
I'm getting an exception thrown but I don't what to do or where to find something about this exception
```

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffefcd0 end: 3ffeff90 offset: 01a0

>>>stack>>>
3ffefe70:  e51988ff 3ffeff14 3ffefee0 4020190c  
3ffefe80:  00000000 6f742065 69687420 40202e98  
3ffefe90:  3ffefeb0 00000004 3ffefee0 40202940  
3ffefea0:  0000000d 00000001 3ffefee0 402029db  
3ffefeb0:  0000000d 3ffeff30 3ffefee0 3ffeedb0  
3ffefec0:  00000000 3252eb98 e51988ff 40202d18  
3ffefed0:  00000007 4b000000 64616b61 feef0075  
3ffefee0:  3fff07e4 3ffeedb0 00000032 00000000  
3ffefef0:  00124f80 5bfd29f0 00000000 3252eb98  
3ffeff00:  00000000 00000000 006f007f 0000000f  
3ffeff10:  00000000 2e373132 322e3238 322e3933  
3ffeff20:  00003134 0000000e 3ffeef30 40203bdd  
3ffeff30:  3ffeedb0 3ffeee30 3ffeef30 40203bdd  
3ffeff40:  00000000 3ffeef5c 3ffeef30 40203c08  
3ffeff50:  3fff00bc 40203134 3ffeef30 3ffeef5c  
3ffeff60:  3fffdad0 00000000 3ffeef54 40202927  
3ffeff70:  3fffdad0 00000000 3ffeef54 40203fe4  
3ffeff80:  feefeffe feefeffe 3ffeef70 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
Is there a list of exceptions?

Already found [that issue](https://github.com/esp8266/Arduino/issues/1275), but atm I'm not able to find the problem. And the first link is not working anymore :/

Could anybody help me?
Thanks :)

Pablo2048, 2018-01-09:
Next time please spent more time in searching for infos - documantation was moved to readthedocs, so actually is here https://arduino-esp8266.readthedocs.io/en/latest/exception_causes.html . To further diagnose your problem use the ESP exception decoder (STFW).

devyte, 2018-01-10:
@pryxilon when you opened the issue, you were presented with a template to fill out, requesting certain information. You have pretty much ignored it completely.
I see a double reset, so off the top of my head: after flashing over serial, did you do a physical reset by pressing the reset button?

Please read our issue policy, the repo documentation on readthedocs, and use the exception decoder.
This is an issue tracker, meant to track issues in the core libs, and not meant to provide general help.
Closing per issue policy.

closed 2018-01-10 by devyte

10xnorellana, 2019-07-29:
i started the connection to wifi again before to use the http.begin(). it worked for me