esp8266/Arduino #4124: Its possible 2 oled wemos in the same time?

udelunar, 2018-01-09:
### Basic Infos

#### Hardware
Hardware:		wemos d1 mini v2

### Description

Good morning, I am using the adafruit library to be able to use the wemos oled shield (v1.1.0) and it works perfectly. But I'd like to know if you can use 2 oleds at the same time.

I have searched the internet and studied the examples, but I am not able to. Is it possible what I want to do?
If I could be able to change the pins so that one oled would be on D1, D2 and the other oled on D3, D4, I think it would work.

Thank you very much

### Settings in IDE

Module:  	 Generic ESP8266 Module v2.0
OLED:       Wemos 1.1.0

### Sketch
I use all examples of Adafruit (OLED_Cube_with_SparkFun, OLED_Hello_with_SparkFun, etc)

5chufti, 2018-01-09:
Hi,
sorry but this is no support forum for adafruit nor wemos. 
This location is for solving issues with arduino esp8266 core.
Please inquire with the manufacturer of your parts or on some specialised ESP8266 board e.g.
http://www.esp8266.com

As your request is not in any way related to a malfunction of the arduino esp8266 core please close the issue.

devyte, 2018-01-10:
@udelunar Closing per current issue policy doc.

closed 2018-01-10 by devyte