esp8266/Arduino #4126: Exception (29): wifi_softap_get_station_info() causing fatal exception

cg101, 2018-01-10:
Hello team!
I need to trigger a function whenever my esp8266 assigns an IP Address to a connected station device. The function is going to simply print out the mac addresses of all the connected station devices.
I have the following code:
```
#include<ESP8266WiFi.h>
extern "C" {
#include "user_interface.h"
}
int dhcp_clients_num=0;
void setup() {
    Serial.begin(115200);
    delay(100);
    WiFi.begin();
    WiFi.softAP("esp8266","password");
}
void displayClients(){
  Serial.println("Number of clients currently connected:");
  Serial.println(dhcp_clients_num);  
  Serial.println("Displaying Clients");
  struct station_info *stat_info;
  stat_info = wifi_softap_get_station_info();  
  while(stat_info!=NULL){
    for(int i=0;i<6;i++){
      Serial.print(stat_info->bssid[i]);
      if(i!=5){
        Serial.print(":");
      }
    }
    stat_info = STAILQ_NEXT(stat_info,next);
  }
  Serial.println();
  Serial.println("Done displaying Clients!");
}
void clientCheck(){
    struct station_info *stat_info;
    stat_info = wifi_softap_get_station_info();  
    int temp = 0; //temp is for getting the number of station devices
    while(stat_info!=NULL){
      temp++;
      stat_info = STAILQ_NEXT(stat_info,next);
    }  
    if(dhcp_clients_num!=temp){
      dhcp_clients_num = temp;
      displayClients();
    }    
}
void loop() {
  clientCheck();
}
```
I got the following output when I connected to my ESP8266 using my laptop:

```
Number of clients currently connected:
1
Displaying Clients
88:251:132:144:54:247
Done displaying Clients!

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef290 end: 3ffef4b0 offset: 01a0

>>>stack>>>
3ffef430:  3fff017c 00000010 00000001 401004f4  
3ffef440:  40219de6 3fff0180 00000001 40106ff8  
3ffef450:  4021b441 3ffee450 00000005 4021b485  
3ffef460:  40201cdd 00000000 00000000 3fffdad0  
3ffef470:  3ffee47c 3ffee360 00000001 40201d00  
3ffef480:  00000000 00000000 3ffee474 40201d14  
3ffef490:  3fffdad0 00000000 3ffee474 402027d4  
3ffef4a0:  feefeffe feefeffe 3ffee490 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
I have decoded the message below

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 11 results
0x401004f4: calloc at C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x40219de6: wifi_get_opmode at ?? line ?
0x40106ff8: pvPortZalloc at C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4021b441: wifi_softap_get_station_info at ?? line ?
0x4021b485: wifi_softap_get_station_info at ?? line ?
0x40201cdd: clientCheck() at C:\Users\Dell\Documents\Arduino\cribgenietest/cribgenietest.ino line 34
0x40201d00: clientCheck() at C:\Users\Dell\Documents\Arduino\cribgenietest/cribgenietest.ino line 43
0x40201d14: loop at C:\Users\Dell\Documents\Arduino\cribgenietest/cribgenietest.ino line 46
0x402027d4: loop_wrapper at C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```
I don't see what's causing this. Could there be an issue with wifi_softap_get_station_info() function?

devyte, 2018-01-10:
@cg101 per Espressuf document 2c-esp8266_non_os_sdk_api_reference, you are not freeing the resulting station_info structure with wifi_softap_free_station_info(), hence you have a mem leak. See section 3.5.38 of Espressif's reference doc.
This is not the right place to ask questions about Espressif's SDK api usage. Please refer to Espressif's forum for that, or to a community forum.
Having said that, I wouldn't mind seeing this added to the ESP8266WiFiAp class interface, but with a clean robust implementation. Are you willing to discuss an implementation, implement it, and make a PR for review?

cg101, 2018-01-11:
@devyte  Adding wifi_softap_free_station_info() to my function solves the problem.
Sorry, I am only a beginner. I don't know how to write code for ESP8266WiFiAP class interface. But I have a few ideas.
I would love to see an implementation in ESP8266WiFiAP class which would allow developers to trigger a function whenever a station device is assigned an IP Address by the ESP8266. Many a time developers would like to get the MAC Addresses or IP Addresses of station devices connected to their ESP. Using wifi_softap_get_station_num() in a loop to check if a new device is connected to the ESP is unreliable. Because, the struct returned by wifi_softap_get_station_info() is not updated as quickly as wifi_softap_get_station_num(). Station device info is added in the struct only after the ESP assigns an IP Address to the station device. This is forcing developers to write code similar to mine, which is obviously not as robust as you would like. There needs to be function like  get_dchp_clients_num() in ESP8266WiFiAP class or an event listener for when a station device is assigned and IP Address by the ESP. I'll see what I can do.

devyte, 2018-01-11:
I think this is already done, at least for the station mac, see the WiFi event system. There's an event onSoftApStationConnects that triggers when a station connects, and the callback receives some info as argument.

devyte, 2018-01-11:
Closing as resolved.

closed 2018-01-11 by devyte