esp8266/Arduino #4130: Handle Espressiif's removal of mesh support

devyte, 2018-01-10:
### Basic Infos

#### Hardware
Hardware:			all
Core Version:      	        2.4+

### Description

Espressif has removed support for their mesh lib, which means that the lib won't exist starting from their upcoming v2.2 SDK release.
When we migrate to SDK v2.2, we'll need to either pull our ESP8266Mesh library, or freeze the current sdk libmesh, with the understanding that it may fail in the future.

### Sketch
HelloMesh example

yoursunny, 2018-01-11:
I don't see how *ESP8266WiFiMesh* library relates to `libmesh`.
*ESP8266WiFiMesh* is implemented on top of *ESP8266WiFi*: each node establishes an AP, and sends message to another node by connecting to its AP.
I searched for `espconn_mesh_create_packet`, a function provided by `libmesh`, and didn't find anything in this repository.

devyte, 2018-01-11:
@yoursunny updated description.

devyte, 2018-11-27:
The libmesh is already removed in one of the SDK updates. Closing.

closed 2018-11-27 by devyte