esp8266/Arduino #4137: UDP not recive and send correct large packet

asetyde, 2018-01-11:
With upgrade 2.4.0 our software not working properly, udp message arrive slowly and when it reply , it's sends many times same message 

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESPino
Core Version:      	2.4.0
### Description

Problem description

### Settings in IDE

Module:  			ThaiEasyElec
Flash Size: 		4mb
CPU Frequency:		80Mhz
Flash Mode:			qio

#include <SoftWire.h>
#include <HIH61xx.h>
#include <AsyncDelay.h>

HIH61xx hih;
AsyncDelay samplingInterval;

void setup(void)
{
  Serial.begin(9600);
  hih.initialise(4,5);
  samplingInterval.start(3000, AsyncDelay::MILLIS);
}

bool printed = true;
void loop(void)
{
  if (samplingInterval.isExpired() && !hih.isSampling()) {
    hih.start();
    printed = false;
    samplingInterval.repeat();
    Serial.println("Sampling started");
  }

  hih.process();
  
  if (hih.isFinished() && !printed) {
    printed = true;
    // Print saved values
    Serial.print("RH: ");
    Serial.print(hih.getRelHumidity() / 100.0);
    Serial.println(" %");
    Serial.print("Ambient: ");
    Serial.print(hih.getAmbientTemp() / 100.0);
    Serial.println(" deg C");
    Serial.print("Status: ");
    Serial.println(hih.getStatus());
  }
  
}

d-a-v, 2018-01-11:
Can you please be more accurate about your problem description ?
The best would be a minimal sketch that could allow us to reproduce.
You can also tell us if your software works by selecting lwip v1.4 in Tools menu.

asetyde, 2018-01-11:
lwip ?

d-a-v, 2018-01-11:
```lwIP Variant: "v1.4 Prebuilt"```

[image]

asetyde, 2018-01-12:
[image]
It's not available for me ... but this is one of few CE FCC certificated esp can be used

d-a-v, 2018-01-12:
Sorry I don't understand.
If you are using software release 2.4.0 or later git version then you should not have this menu.

asetyde, 2018-01-12:
ok now I test it, sorry how is it ?

d-a-v, 2018-01-12:
Here is the doc to install current source version of core: [link](https://github.com/esp8266/Arduino#using-git-version)

asetyde, 2018-01-12:
Using library Ticker version 1.0 in folder file:///C:/Users/Experimental%20alex/AppData/Local/arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/Ticker
"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Experimental alex\AppData\Local\Temp\VMBuilds\touch\esp8266_espinotee\Release/core" -c -Wall -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10801 -DARDUINO_ESP8266_ESP13 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_ESP13\" -DESP8266 -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\espinotee" -I"C:\Users\Experimental alex\Documents\Arduino\libraries\MySQL_Connector_Arduino-master\src" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\EEPROM" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ethernet\src" -I"C:\Users\Experimental alex\Documents\Arduino\libraries\OneWire" -I"C:\Users\Experimental alex\Documents\Arduino\libraries\DallasTemperature" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\SPI" -I"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\Extensions\pbd1usl1.2ll\Micro Platforms\visualmicro\ide\libraries" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries" -I"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\Extensions\pbd1usl1.2ll\Micro Platforms\default\debuggers" -I"C:\Users\Experimental alex\Documents\Arduino\libraries" "C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker\Ticker.cpp" -o "C:\Users\Experimental alex\AppData\Local\Temp\VMBuilds\touch\esp8266_espinotee\Release\Ticker\Ticker.cpp.o"
recipe.hooks.sketch.prebuild.1.pattern
make -C "C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/src" install TOOLS_PATH="C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-"
	Process: make -C "C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/src" install TOOLS_PATH="C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-"
Impossibile trovare il file specificato
 
Building project code ...
Error compiling project sources
Build failed for project 'touch'
"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip/include" "-IC:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Experimental alex\AppData\Local\Temp\VMBuilds\touch\esp8266_espinotee\Release/core" -c -Wall -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10801 -DARDUINO_ESP8266_ESP13 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_ESP13\" -DESP8266 -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\espinotee" -I"C:\Users\Experimental alex\Documents\Arduino\libraries\MySQL_Connector_Arduino-master\src" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\EEPROM" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker" -I"C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA"  [... cut ...] "C:\Users\Experimental alex\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries" -I"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\Extensions\pbd1usl1.2ll\Micro Platforms\default\debuggers" -I"C:\Users\Experimental alex\Documents\Arduino\libraries" "C:\Users\Experimental alex\AppData\Local\Temp\VMBuilds\touch\esp8266_espinotee\Release\touch.cpp" -o "C:\Users\Experimental alex\AppData\Local\Temp\VMBuilds\touch\esp8266_espinotee\Release\touch.cpp.o"

asetyde, 2018-01-12:
not compile with 1.4

d-a-v, 2018-01-12:
I guess you would have to restart the install process but before that carefully rename/remove the arduino installation directory **and** ```C:\Users\Experimental alex\AppData\Local\arduino15```.

asetyde, 2018-01-16:
ok I will try next days, thks a lot

devyte, 2020-05-14:
Closing due to age and lack of feedback.
If you think there is still a problem, please retest with latest release or latest git, then open a new issue and follow the template instructions.

closed 2020-05-14 by devyte