esp8266/Arduino #4144: Problem with string concatenation on 8266

sdybiec, 2018-01-12:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			LinkNode D1 powered by ESP-8266EX
Core Version:      	2.4.0

### Description

Problem description: Arduino crashes after 34667 executions of this loop. The sketch below can be used to reproduce the problem. The code runs without crashing on Uno, so there's something different about the ESP8266.  I've narrowed down the problem to this line of code: 

memoryMsg = "" + count;

If commented out, the program runs forever, as it should. 

### Settings in IDE

Module:  			Webmos D1 R2 & mini
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

void setup() {
  Serial.begin ( 115200 );
}

String memoryMsg;
long count = 0;

void loop ( void ) {
    memoryMsg = "" + count;
    Serial.println(count++);
}

### Debug Messages

The output looks like this at the end:

....
34663
34664
34665
34666
34667

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

devyte, 2018-01-12:
@hotchpotch of course it does. I'm not sure exactly what you're expecting, but this line is wrong:
`memoryMsg = "" + count;`
What that does is take a pointer to a const char * and add count to the pointer. The result is a pointer to some odd place in the ESP's memory area. This is basic C usage: neither C nor C++ do concatenation on const char *strings.
I think that your intent is to do this:
`memoryMsg = String("") + count;`

The String class implements operator+() as concatenation.

Closing due to user error.

closed 2018-01-12 by devyte

igrr, 2018-01-12:
Did you mean `memoryMsg = String("") + count` by any chance?

I.e. the way you have written means that you want to create a String from the contents of a character string which is starting 'count' bytes after the beginning of "" (empty string). That is, pointer arithmetic rules are used.

If you convert the empty literal string into a String object first, you will get string concatenation behavior.

sdybiec, 2018-01-12:
Thanks. Since this is Arduino Processing language, I thought the preprocessor took care of this. Sorry for the tangent.

https://www.arduino.cc/en/Tutorial/StringAdditionOperator

igrr, 2018-01-12:
@sdybiec The page you link to contains a paragraph on this exact issue:
```
Caution: You should be careful about concatenating multiple variable types on the same line, as you may get unexpected results. For example:

  int sensorValue = analogRead(A0); 
  String stringOne = "Sensor value: ";
  String stringThree = stringOne + sensorValue;
  Serial.println(stringThree);  

results in "Sensor Value: 402" or whatever the analogRead() result is, but

  int sensorValue = analogRead(A0); 
  String stringThree = "Sensor value: " + sensorValue;
  Serial.println(stringThree);  

gives unpredictable results because stringThree never got an initial value before you started concatenating different data types.
```